betaflight/betaflight #4735: Dynamic balancing

loeufdur, 2017-12-10:
Hi every one,
Why you do not use the FC to make a dynamic balancing system for propellers and engines.
You have got all you need on the FC, it misses only the softwar.

Loeufdur

jflyper, 2017-12-10:
@loeufdur Can you elaborate what "dynamic balancing of props and motors" is?

loeufdur, 2017-12-11:
I mean balancing of rotating masses.
I think we get al we need with esc and FC to make a little balacing machine for props and motors.

RipperDrone, 2017-12-11:
Good dynamic balancers indicate the exact position on the circumference of a rotating mass where additional mass has to be added / removed. I believe BL DC motors are not up to the task of giving a true rotary position. So only feature which seems realistic is to show unbalance in a graph which BFC can already do today (accels x/y/z). To assist your trimming of balance by trial/error regarding position to add/remove mass.

jflyper, 2017-12-11:
Doesn't that require an FC to know exact rotational position of a spindle?

fiveangle, 2017-12-12:
Only way I've been able to "dynamically balance" is by putting a mirror on an arm with a laser pointed at it so it hits a wall some distance away, then run the motor and tweak the prop with tape until the reflected beam movement is minimized.  You apply tape to the prop psuedo-randomly, since only info available is if it is balanced or not, but not where the balancing is off.

I long postulated that one could use the flight controller gyro to do the same thing without all this laser business, but never took the idea further.  I suspect that's what's being suggested.

-=dave

mikeller, 2017-12-12:
I've seen setups where a stroboscopic light was used to illuminate the prop in the moment when the movement on a certain axis was strongest. Combined with the use of a permanent marker to write numbers onto the blades, this approach could be used.

jflyper, 2017-12-12:
So, in conclusion, we don't have all we need?

mikeller, 2017-12-12:
Not sure. Using one of the on-board LEDs (or an external one with a driver) might work if there isn't too much ambient light.

jflyper, 2017-12-12:
Wo!

ledvinap, 2017-12-12:
@mikeller : I tried that with some random white led + driver:
- Light intensity was quite poor even in dim room (better led should do)
- Phase between maximum amplitude on gyro and propeller is frequency dependent (and that was with MPU board glued next to motor)
- There is lot of noise. On heavily unbalanced prop it is easy,  PLL-like algorithm was necessary when propeller was partially balanced.

Overall I got better/faster results using small reproductor as sensor + Visual Analyser (http://www.sillanumsoft.org/) with long FFT (few seconds time constant) to measure imbalance.

Gyro may work well if actual motor position is known and gyro/acc data are averaged over many revolutions / different speeds. Opto gate (or laser reflective sensor) may be used to detect blade. Starting from marked blade and accurately counting all passes will work for multiple blades.

ledvinap, 2017-12-12:
BTW: https://www.rcgroups.com/forums/showthread.php?2337523-Dynamic-Propeller-Balancer-Build

fiveangle, 2017-12-12:
It would be neat to try and make the balancing brain-dead by indicating the exact axial position of the imbalance with a strobe or whatnot, but honestly, the 80/20 solution of giving a FFT indicator of vibration with running peak indicator on the motor tab of BFC would do the trick (with the extra effort of having to test balance spots until it was right).

Using the FC to indicate qualitative vibration would be infinitely easier than the laser trick: so much so that many would then actually do it instead of procrastinating ;)