betaflight/betaflight #4736: Gyro overflow slew filter only works in 32KHz mode

etracer65, 2017-12-10:
@martinbudden @mikeller @gke 

TL;DR: The gyro overflow slew filter hasn't been actually doing anything for the majority of the users with the affected ICM206xx gyros.

The story:

I got out my "Drill of Doom" yaw spin rig to do some testing and thought I was losing my mind.  Testing with an ICM20608G gyro showed that the slew filter wasn't preventing the gyro inversion.  I knew it *was* working and even as recently as issue https://github.com/betaflight/betaflight/issues/4407 and PR4412 as this was the fix to add MPU6500 to the conditional compilation (ICM20608G is detected as MPU6500).  I tested the fix (I swear I did!) and everything was good.  So why wasn't it working now?

So I tested many versions back through earlier commits trying to find the problem.  All the way back to the original slew filter implementation in 3.2.0RC5 and still it wasn't working.

Through debugging and modifying the code I determined that the `gyroSlewLimiter()` function in gyro.c was in fact being called - it just wasn't catching the gyro overflow.  So I became suspicious of the +-16384 (1<<14) threshold for triggering.  After some spin runs where I captured raw gyro values I began to see that the raw gyro values were more "smoothly" moving between +32767 and -32768 with intermediate values in between.  The steps weren't enough to trigger the slew filter!

So why is this happening now?  Then I remembered that previously I did all the testing with 32KHz gyro mode enabled because we wanted to capture the most granular gyro data.  However these tests I was running with 32KHz off and the gyro/pid at 8/8K.  So I switched back to 32KHz mode and ran at 328K and the slew filter started  working again!  The raw gyro values were flipping from +32767 to -32768 is one sample like before.  Next I tested with 32KHz enabled but with the gyro/pid at 8/8K.  Again the slew filter worked as expected.

So looking deeper at the raw gyro data it appears when the gyro is not in 32K mode there is additional filtering or a lowpass stage in the gyro that is smoothing the data.  This can be seen in the log where rather than flipping from +32767 to -32768, it goes:
32768, 19595, 12524, 6220, -5250, -10255, -19344, -24054, -29021, -32768

This can be seen in the log: icm20608g-bf322-8k-raw.BBL from sample 28503 to 28516.

So looking at how the data is behaving I'm not sure how the current slew filter design can be reconciled to work.  The data appears to be too smoothed to allow effective detection by a magnitude change from one sample to the next.  Simply changing the detection threshold from 16384 to a smaller number doesn't look viable.  Too small a number and you get false detection.

**Perhaps we need to shift to comparing multiple samples or even a magnitude change *over a defined time period*.**  Like if the raw gyro value exceeds a magnitude change of over X in Y time then consider that invalid data.

Alternately, the ICM206xx gyros could be *forced* to run in 32KHz mode to disable the internal filtering (gyro/pid loop could still be allowed lower values as this works properly).  This would allow the slew filter to start working again.  However this would probably have noise/tuning ramifications.

The attached zip file has a bunch of logs with different settings.  Each "flight" has a slow ramp to overflow and a faster spin up for comparison.  Every one is an overflow situation.

`icm20608g-bf322-8k-notch.BBL     -- 32K off, 8/8K debug_mode = notch -- FAIL`
`icm20608g-bf322-8k-raw.BBL       -- 32K off, 8/8K debug_mode = raw   -- FAIL`
`icm20608g-bf322-32k-notch.BBL    -- 32K on, 32/8K debug_mode = notch`
`icm20608g-bf322-32k-raw.BBL      -- 32K on, 32/8K debug_mode = raw`
`icm20608g-bf322-32k@8k-notch.BBL -- 32K on, 8/8K debug_mode = notch`
`icm20608g-bf322-32k@8k-raw.BBL   -- 32K on, 8/8K debug_mode = raw`

You can probably ignore the "notch" versions as the debug data is after the slew filter.  So in cases where the slew filter is working it's not very informative.  But in cases where it's not you can see the intermediate gyro values in dps rather than the raw ADC values.

These logs are all with stable 3.2.2, target AIRBOTF4 and gyro ICM20608G.

[slew_filter_testing.zip](https://github.com/betaflight/betaflight/files/1545940/slew_filter_testing.zip)

mikeller, 2017-12-10:
Good finding, @etracer65! If the overflow is 'smoothed out' in the gyro if 32 kHz is not enabled then they are broken quite badly. :-(

etracer65, 2017-12-11:
Since the gyro outputs in an overflow event appear to be unpredictable, maybe an alternative strategy would be to look to the PID sum having no positive effect on the axis over a time period.  When the gyro overflows and inverts the PID sum on yaw would cause a positive reinforcement to the spin.  Detecting that no improvement vs. error over time has occurred might be a way to detect the problem.

Aileron8, 2017-12-11:
I’m not sure if this will help or not, probably not, but I believe the polarization events occur due to the way gyros measure coriolis forces and may be predictable in some respect. I’ve outlined the concept here in RCG:

https://www.rcgroups.com/forums/showpost.php?p=38784184&postcount=53098

ctzsnooze, 2017-12-11:
Also pid_sum_limit is not working, I think.  

See logs here from 3.3 yaw spins with reversals where pid_sum_limit regularly exceeds the default of 400 and goes to max:

https://github.com/betaflight/betaflight/files/1546369/YSTTM_Events.zip

from https://github.com/betaflight/betaflight/issues/4407#issuecomment-350610235

ctzsnooze, 2017-12-11:
If I only move the 'Enable 32k gyro mode' slider to the right, so it's on, while leaving gyro update rate and PID loop rate at 8kHz, my F4 CPU is 5%.

But if I both enable 32k gyro mode and set the gyro update loop rate to 32k, the CPU load is 26%.

If all I want is for the slew rate filter to be active on ICM206xx F4's, can I just enable 32k gyro sampling mode but leave gyro update rate at 8k?

martinbudden, 2017-12-14:
@etracer65 , good find. I've always been a bit sceptical about the slew filter - my analysis earlier indicated there were cases when it could be fooled. I was working on something that looked at the history of several gyro values, but it quickly became very complicated.

My original plan was to just disable the yaw PIDs when the yaw gyro values were above a threshold, but I was put off this by a pilot who said when doing acrobatics he got commanded yaw rates of over 1800dps.

I think the answer is to add the yaw PID limit as an option, default off. It can be turned on by racers who typically do not have high commanded yaws and only get high yaw rates when they clip a gate.

ctzsnooze, 2017-12-14:
Hi Martin, having experimented with this, the best solution for the pilot is to remove absolutely all PIDs when yaw exceeds a limit, so the motors just don't respond to anything.  I've flown this a lot now and its really good.  If you leave pitch and roll active, and if the quad is spinning fast, they just don't work, and in fact add to the problem.  You can't see what's going on anyway when it's spinning, no point pitching or rolling, no control.  

However it is no solution whatsoever to the gyro inversion issue, as logs of mine has shown, all that happens is that the quad gets 'stuck' between the zero PID point and the point when PIDs return, but when they return the push the wrong way and increase the yaw rate again until PIDs stop again, endlessly.

The only solution at present is the slew filter in 32k mode.  Or just don't race with these gyros

martinbudden, 2017-12-14:
I see your point about removing all PIDs, including roll and pitch.

As to the gyro inversion, one possible improvement is to add some hysteresis. That is, PIDs turn off at some high yaw value, say 1800, but then don't turn back on again until a low yaw value is reached, say 200. That should help with the "getting stuck" problem.

ctzsnooze, 2017-12-14:
Problem with inversion is that the data rolls off (say) the top at +2000 and immediately becomes -2000 even if say the real data is +3000.  Then sometimes the value comes back towards zero, but remains opposite in sign.  I don't think we really know the exact way the inversion resolves.  eTracer will have a good idea though.

etracer65, 2017-12-14:
I'm wary about simply chopping the PIDs based on some high gyro rate.  Doing that basically ensures that there will be no chance to recover from a gate/ground strike.  Not every impact will result in a gyro overflow on the affected gyros.  In fact, the overflow appears to happen somewhere significantly above a physical 2000dps rotation rate.  My drill rig ends up spinning at about 8500dps and accelerates pretty rapidly up to that speed.  The variable speed aspect is not very good as it has a discrete number of "steps" and with the trigger it's difficult to test various ranges.  But as can be seen by the graph below the gyro rate climbs to 2000dps and holds for a period before overflowing/inverting.  In reality the drill is still accelerating through this period but the gyro is dealing correctly with the overspeed - at least until some threshold (probably greater than 4000dps). I'm confident that if I could control it and set the speed say at 2500dps the gyro probably wouldn't overflow.

I think the key is detecting the overflow condition reliably.  If we can do that then I'm fine with killing the PID's as recovery is not an option otherwise.  It's easy to detect the overflow event in 32KHz mode and that's what the slew filter is doing - looking for unreasonable changes from one sample to the next.  The problem is that when not in 32KHz mode the gyro hardware appears to implement a lowpass filter and smooths the output (second screeenshot below) - making the slew filter unreliable.

I would propose instead of the slew filter we look at a system that looks at the gyro change over time to detect unreasonable events as a trigger for overflow detection.  For example in the second image below the log was captured at 8K with a 8/8K gyro/pid loop.  On the initial inversion it takes roughly 1ms to go from +32767 to -32768 - but there about 8 samples in between (this is what's preventing the slew filter from working).  So maybe we say that "if the gyro exceeds a change of X dps in Y milliseconds" then we're in overflow detection.  Chop the PID's.  I think we can adjust the range and time to be short enough to avoid false triggering.

Now... how to know when the quad has slowed down enough to exit from the overflow and re-enable PID's?  This is tough because after the initial overflow messiness the gyro stabilizes at -32768 and just stays there.  When the spinning slows down there's again a noisy transition period where the value jumps around and finally recovers.  See third screenshot.  So I'm not really sure what to do in this case.  Maybe we just consider an overflow event "unrecoverable" and keep the PID's disabled until disarm.

[image]

[image]

[image]

ctzsnooze, 2017-12-14:
I'm hoping the slew filter will somehow manage both the initial overflow transition and the recovery.  Ideally it should hold, in the 32k example above, a positive yaw value until the data comes back into a positive range and has at least a couple of acceptable steps in that range.  I don't think that can be achieved at all in 8k mode.

Your finding that the inversion happens perhaps quite a lot above 2000 fits with my bathroom floor testing when I was spinning the quad.  I don't think I was getting more than 2500; I could easily get to 1500 and 1950 with some practive, but probably not 3000.  So perhaps that's why I could not trigger a reversal with that approach - just didn't get it to spin fast enough.

ctzsnooze, 2017-12-14:
Actually killing all PIDs while spinning fast is perfectly good, and you can still recover in flight.  See

https://www.youtube.com/watch?v=p6ylT_zK7g0

These deliberate spins were on an MPU6000 with all PIDs killed above 1500 deg/sec.  The impact caused a self-limiting spin that caused some climb because I was in FFF at the time, yaw rate slowed down quickly in the absence of any motor activity speeding it up, and I could recover in flight.

Also the logs I posted here showed several impacts that triggered zero PIDs but not flyaways.

What seems to happen with zero PIDs is just that the motors don't exaggerate the problem.  Very quickly the pitch/roll precession stops, and the spin itself slows down fairly quickly without the motors actually doing anything about it, at which point you get control back in 0.3s or so.

Pop a bit of code like what I suggested earlier, works very well, as a pilot the spins are simple less aggressive, shorter in overall duration, and easier to fly out of.  On all gyros.

etracer65, 2017-12-14:
An alternative (possibly better) approach might be to use a system that looks at error (measured - commanded) rather than just the absolute rate coming from the gyro.  A generalized system that examines the **error** over time and if it's increasing then we're in an uncontrolled spin and we should chop the PID's.  Likewise recovery would be when the error has been reducing consistently over a time period.  At that point we could re-enable the PID's.  The detection logic should be applied separately to all axis but killing the PID's should apply to all axis once the event is detected.  The timing of the enable/disable might be a little tricky to get right as I'd think you could be more aggressive on the start event but probably have to have a longer period of "improvement" to trigger the disable (to avoid another spin up).

I think this would handle the gyro overflow as well as likely improve recovery from de-syncs or other uncommanded roll events.

Additionally by looking at error rather than absolute gyro rate we avoid the possibility of false triggering for people who enjoy really high rates.

ledvinap, 2017-12-14:
@etracer65 : Can you try your rig with Fc mounted at small angle? This should couple rotation onto roll&pitch axes and actual speed will be clearly visible (angle rate on R&P will stay in limits and sampling is good for much faster spins)

@ctzsnooze : Disabling (or strongly attenuating) roll & pitch on fast YAW reasonable(gyroscopic forces will keep spin stable). With known yaw spin direction, it may help to use reasonable (and constant) yaw command in opposite direction.
Spin may be probably stabilized when roll and pitch commands are applied with phase advance (gyroscope). Not sure about required angle, but for fast spins it may be 90deg (the math should not be hard)

@etracer65: What you describe would need mathematical model of quad, comparing modelled measurements with measured ones(Kalman filter can realize this). Simpler implementation may be hard to tune - look at PRs related to crash detection ...

etracer65, 2017-12-14:
@ledvinap The logs posted above are with me holding the drill in my hand during the run.  So I'm absolutely sure there's some small angle already there on the roll/pitch.

Also, I don't understand the need for mathematical models, Kalman filters, etc.  Put very simply, if the error on an axis is increasing (or not improving) over time despite the flight PID trying to correct it, then that's an indication of a problem.  No need to predict what's happening.  In the case of the gyro overflow the gyro output has erroneously gone to -2000dps and is holding there.  The PID loop is pushing the wrong direction (positive) but the physical model is not slowing down.  The PID loop is not "improving" error over time.  This would be a clear detection of the overflow event.  The goal with this is to reliably detect when "something really bad" has happened.  At this point zeroing the PIDs might be the response or maybe something else, but we need to detect the event first without false positives.

ledvinap, 2017-12-14:
@etracer65 : The idea is to mount FC so that drill rotation axis does not match yaw axis exactly. It should cause R and P gyros to 'see' the rotation. Your setup is too accurate, mismatch is hidden in other noise.

etracer65, 2017-12-14:
@ledvinap I'll try to figure out a way to mount at an angle without too much work.  I don't want to significantly take apart or change the rig as I feel like we'll need to do a lot more testing and it's a good control that can consistently replicate the problem.

How about another option... What if I enabled the accelerometer and tested with the Z axis at 45 degrees perpendicular to the ground by simply tilting the drill.  Or is the accelerometer data too slow or filtered for the purpose?

etracer65, 2017-12-14:
@ledvinap I did a test with the accelerometer on and the rig tilted about 45 degrees from vertical.  The accel X axis showed a clear signal (IMU is mounted significantly off center in this board).  My earlier guess was remarkably accurate in that the overflow started at a measured 3970dps.  Max spin during the run was held at about 5967dps.  I repeated the test and reliably calculated around 4000dps for the onset.

This run was BF 3.2.2, accelerometer enabled, 32KHz mode active, 8/8K gyro/pid, GYRO_RAW debug_mode.  Since this was in 32K mode the slew filter was working so the yaw gyro output doesn't show the overflow.  Look at the debug[2] for the raw yaw gyro.

[icm20608g-bf322-32k-8k-raw-accel.BBL.zip](https://github.com/betaflight/betaflight/files/1559922/icm20608g-bf322-32k-8k-raw-accel.BBL.zip)

[image]

[image]

martinbudden, 2017-12-14:
Interesting. We could estimate the absolute gyro rate from the frequency of the acc oscillations and we could derive the sign of the gyro rate by the phase difference between acc oscillations on the X and Y axis. What do you think @ledvinap ?

etracer65, 2017-12-14:
@martinbudden If the chip is mounted in the center of the board (or close to it) the acc signal would be very weak and probably difficult to distinguish from noise.  It just happens that on this board the chip is mounted quite close to the edge so the signal looks good.

etracer65, 2017-12-14:
So I started diving into the code and datasheet for the ICM-20608-G chip to better understand the config and see what options are there for the hardware lowpass filter.  The goal was to try running with 32KHz **disabled** and remove the lpf that is preventing the slew filter from working (due to the smoothing).  I found you can't actually completely disable the lpf, just select different bandwidth cutoffs.  First in accgyro_mpu6500.c (icm20608g is detected as a MPU6500), the line:

`const uint8_t raGyroConfigData = gyro->gyroRateKHz > GYRO_RATE_8_kHz ? (INV_FSR_2000DPS << 3 | FCB_3600_32) : (INV_FSR_2000DPS << 3 | FCB_DISABLED);`

is where the gyro scale and lpf is set.  2000dps is always the scale, but the lpf settings vary for 32K vs. 8K mode.  The selection of the lpf type is controlled by `FCHOICE_B` and `DLPF_CFG`.  In 32K mode FCHOICE_B = FCB_3200_32 which is 2.

```
enum fchoice_b {
    FCB_DISABLED = 0,
    FCB_8800_32,
    FCB_3600_32
};
```

So this selects the hardware digital lpf this way:

[image]

With 32K mode disabled, FCHOICE_B = FCB_DISABLED or 0.  This means that the lpf filter selection is then determined by `DLPF_CFG` which correlates to `gyro_lpf` in the CLI.  By default `gryo_lpf` is set to "OFF" which is 0.  So that gives us the yellow highlighted setting.  Note the difference from the 32K version.

[image]

However, by changing the `DLPF_CFG` (which is `gyro_lpf`) we can select a different lpf setting.  By setting it to 7 we match the 32K settings.  In the CLI, the value for 7 is `gyro_lpf = EXPERIMENTAL`

```
static const char * const lookupTableGyroLpf[] = {
    "OFF",
    "188HZ",
    "98HZ",
    "42HZ",
    "20HZ",
    "10HZ",
    "5HZ",
    "EXPERIMENTAL"
};
```

So I tested with **32K mode off**, 8/8K gyro/pid and with gyro_lpf = EXPERIMENTAL.  This eliminated the extra smoothing that was thwarting the slew filter and it once again began working.  Here's what it looks like (raw gyro on top, filtered gyro on the bottom).  No gyro reversal occurred.

[image]

So is this any different than running in 32K mode from a noise perspective?  Not sure.  There will definitely be less filtering/more noise than before.  So there are now two ways to get the slew filter to work:

Use 32K mode or `set gyro_lpf = EXPERIMENTAL`.

It's interesting to note that for 32K mode there is one additional lpf setting that has even more bandwidth (and lots more noise I'm sure).  You'd have to mod the code though to set FCHOICE_B = FCB_8800_32.

martinbudden, 2017-12-14:
I've created a PR that checks for gyro overflow and zeros the PIDs when gyro overflow is detected, see PR https://github.com/betaflight/betaflight/pull/4754

Gyro overflow is when gyro exceeds a threshold, set `ysttm_threshold` in the CLI. I suggest a value of 1900 while testing, but that can probably be increased to 1950 or even higher.

ctzsnooze, 2017-12-15:
These are really good developments.  Thanks so much!

I recall in blackbox that 8khz mode, the gyro trace did look appear very smoothed out. Now we know it the hardware filter is at 250kHz, that's an interesting finding.   I'll try 'experimental' also on the weekend.  First I'll check if 32k stops permanent yaw spins, and if the noise is manageable.  If 32k is good I'll see if 8k experimental does the same.  Losing a 250Hz low pass filter from the gyro data may or may not be a problem, don't know, might need to biquad the gyro instead of PT1.

Martin thanks for ysttm_threshold.  Maybe consider a different name?  Cutting the PIDs above a given gyro rate won't actually stop YSTTM events, which are of infinite duration and are due to gyro reversal.  It will, however, improve behaviour in yaw spins that either (a) are at rates below the reversal threshold or (b) happen on gyros that don't reverse.  Maybe as an alternative name, pid_cut_threshold?

ctzsnooze, 2017-12-15:
One small query... in 32k mode, should the gyro filtering be done first, given the extreme noisiness of the signal?  Theoretically we had some discussion about this.  I now wonder if  gyro LPF should be appled first, before any PID calculations, at least in 32k mode, given how noisy it gets?

martinbudden, 2017-12-15:
I think the name should have `ysttm` in it, since the reason for setting the threshold is to reduce the frequency and severity of YSTTM events. There's no implication that it cures them entirely.

As for gyro filtering - all gyro filtering is done before the PID calculations.

And @ctzsnooze , a big thanks for all your testing, especially since it puts you at risk of incurring flyaways and losing your quad. Can you do some more testing with the threshold set at 1950 and, if that doesn't improve things, try lowering it to 1900.

ledvinap, 2017-12-15:
Acc measurements: centripetal force vector will not move with respect to gyro chip, only magnitude is yaw speed dependent (it will be pointing to center of rotation - fixed point on board). 
Only force of gravity will cause periodic signal and only if yaw axis is not pointing up (aligned with gravity vector). It may be usable, but not trivial to evaluate and there will be situations with no usable signal.

`gyro_lpf = EXPERIMENTAL` : It may be reasonable compromise. One problem with this setting is that signal bandwidth 3.2k is very close to nyquist frequency 4k. This will cause some aliasing. But from noise bandwidth figure, is seems that high order filter is used at this stage.

It may be interesting to try FCB_8800_32 - there is chance that the bug is later in the filter chain..

etracer65, 2017-12-15:
@ctzsnooze When you test PR4754 and the ysttm_threshold, you need to make sure the slew filter doesn't intervene.  So test with 32K mode disabled and gyro_lpf = OFF.  Otherwise the slew filter might catch and prevent the inversion.

Also, the changes in PR4754 will stop YSTTM events.  If won't actually prevent the gyro inversion, but it will detct the high gyro rate and disable the PID loop until it reaches a reasonable value (ysttm_reset) for a substantial period of time (25ms).  This will prevent the PID loop from reinforcing the spin and causing a flyaway.  Take a look at the comments https://github.com/betaflight/betaflight/pull/4754 - I ran a number of tests with overflows to see how the PIDs would react.  So the flight testing part is to see how long it takes the aerodynamic forces to slow the spin after the initial impact back down to the reset value and whether this provides controllable recovery.  It may be possible to even increase the ysttm_reset some to allow the recovery to quick in quicker, but this may require increasing the grace period to prevent false triggering on the gyro overflow noise.

If you can capture any blackbox logs it would be extremely helpful.  If so, use `set debug_mode = GYRO_RAW` and try to capture at 2KHz or better.

Note that the ysttm_threshold logic will reduce the flight controller's response to a crash and increase how long it takes to recover (at least for crashes causing high yaw rates that trigger the code).  So in general we're trading reduced performance to work around the gyro overflow bug by letting the spin initially "slow down on its own" before reacting.  For non-affected gyros like the MPU6000 this is not desirable so we would run with `ysttm_threshold = 0` to disable it.  How much difference it really makes is debatable though and probably needs testing with and without the ysttm_threshold enabled for non-affected gyros.  I'm mainly thinking about my own racing experiences where clipping a gate gives the quick couple of spins with the motors screaming to recover - then getting back under control and continuing.  I've personally never had a YSTTM event.  With the ysttm_threshold active and the gyro rate exceeding it the quad will basically be ballistic until aero forces slow it down so recovery will take much longer.  But it seems to be the price we have to pay for at least the ICM206xx gyros.

etracer65, 2017-12-15:
@ledvinap  Testing with FCB_8800_32 as the lpf config didn't appear to have any noticeable effect on the overflow behavior.  The overflow still kicks in right at 4000dps and the trace looks similar.  I'm sure there's more noise in the signal overall but I'm not really in a position to test that.  The "idle" trace before the spinning starts looks noisier as you'd expect.

etracer65, 2017-12-15:
Ever since I started using the accel to figure out the actual dps at which the overflow occurs, it's bothered me that it turned out to be right at 4000dps.  Consistently, every time.  That's too specific a number.  It's also 2x the full-scale 2000dps configured for the ADC.

Before doing some more testing today I had to charge the battery for my drill.  With a fresh charge I was able to get back to the faster spin rates like before and now top out at a measured 8700dps (calculated from the accel trace).  With the faster spins I started to notice an additional "glitch" in the raw gyro trace.  Calculating the rate at the onset and exit of the glitch it appears that it occurs once the rotation exceeds 8000dps.  Data is a little shakey because my drill steps from about 7000 to 8700dps in the final step so the transition is not smooth.  But I can replicate the extra glitch.

So 2000dps scale, 4000dps overflow, 8000dps glitch...

Starts to look like overflow in the 16bit ADC in the chip.  Each "glitch" point is one bit shift left.  Not much data to go on but interesting nonetheless.

So I did another test where I set the full scale to 1000dps instead of 2000dps.  I was hoping to see the glitches at 2000dps, 4000dps and 8000dps.  Unfortunately the first overflow still occurred at 4000dps so I'm not sure what to make of that.  Possibly the scaling is applied after the ADC - not sure.

[image]

martinbudden, 2017-12-15:
@etracer65 , I'm pretty sure the overflow is in the ADC. For 16-bit ADC, max unscaled value is 32767. Gyro output is scaled by 16.4 and 32767/16.4=1998. For a gyro rated at +/-2000dps you need one extra bit to ensure you don't get internal overflow. So actual overflow point is 1998*2 =3996 dps. So you get output something like this:

Actual spin rate | Reported spin rate
--- | ----
0 | 0
1000 | 1000
1500 | 1500
1998 | 1998
2000 | 1998
3000 | 1998
3996 | 1998
3997 | -1998 (overflow has occurred)
4000 | -1998
7991 | -1998
7992 | 1998 (sign bit has flipped again)

etracer65, 2017-12-15:
@martinbudden Yes, this mostly matches what I'm seeing experimentally.  However the second overflow at around 8000dps doesn't seem to result in a sign reversal.  Instead the output shifts from -1998 to -1874dps and holds for the duration exceeding 8000dps.  Then on slowing below that threshold it again returns to -1998.  Clearly linked to the bit overflows in the 16bit twos-complement ADC, just strange behavior.

ledvinap, 2017-12-15:
@martinbudden : There is also some (small, relatively low frequency) signal mixed into overflow, is looks like this signal is corrupted by 2^14 of 2*2^14 jump ..
https://user-images.githubusercontent.com/17088539/34009077-8299849c-e0d5-11e7-9e8f-758bee599fae.png.

martinbudden, 2017-12-15:
`1874*16.5 = 30733.6 = 0x780D`, so there is something going on at `0x7800`

martinbudden, 2017-12-15:
@ledvinap , I don't understand your comment...

ledvinap, 2017-12-15:
@martinbudden: If you look at overflow traces, there is relatively small and continuous signal over which 16k jumps are superimposed.

martinbudden, 2017-12-15:
I see what you mean. This certainly has implications for the cutoff value - it can't be too close to 1998.

etracer65, 2017-12-15:
I don't think it would matter as the overflow detection would trigger on the first full scale value and then the 25ms countdown would keep getting reset.  I'll try some testing later this evening with 1995 as the cutoff.

etracer65, 2017-12-15:
@martinbudden I tested with ysttm_threshold = 1995 and everything worked correctly.  I also tested with ysttm_reset all the way up to 1850 and that was good as well.  The PIDs were kept zeroed during the gyro inversion and recovered after the event.  The upper limit for ysttm_reset should probably be 1850 as the 8000dps overflow seems to stick the gyro output at 1874dps (see the bump in the middle of the spin).

The noise observed in the raw gyro doesn't really matter in this case because the overflow detection latches on the first full scale value and the the delay and the hysteresis of the lower ysttm_reset makes the noise get ignored.  Also the code is looking at the filtered gyro so that noise isn't even seen.

Lastly, it looks like there should be validation to ensure that ysttm_threshold is higher than ysttm_reset.  Having them reversed would cause the code to not work.

[image]

ctzsnooze, 2017-12-15:
So much happening!

1950 and 1500 should be good threshold values.

I don't know how I can build a binary of PR #4754?

etracer65, 2017-12-16:
@ctsnooze What target do you want?

ctzsnooze, 2017-12-16:
Sorry mate I just figured it out, got the PR into github desktop and can build it.

etracer65, 2017-12-16:
No problem.  We don't want the slew filter to be active so be sure to test with 32K disabled and with gyro_lpf = OFF (the default).  If this code works out I think the slew filter will be removed.

denovich, 2017-12-16:
The proposed solution looks like a good work around for < 32khz. 

But for those of us flying at 32khz, isn’t the slew filter superior? Why remove it? 

(Or did I misunderstand?)

For those of us running at 32khz, I would think we still want the slew filter.  The proposed looks more like a work-around for those who run at a lower gyro rate.

denovich, 2017-12-16:
Also, if you want to see all the possible gyro modes, Kalyn had it figured out. Not much in the way of comments, but he has all the valid permutations and combinations defined.  And hey, it’s gpl’d: https://github.com/OrneryD/RaceFlight-One/blob/current/src/flight_controller/src/drivers/invensense_mpu_icm_device.c

ctzsnooze, 2017-12-16:
Looks like Raceflight use an error based 'spinstopper' that not only sets all PIDs to zero but also throttle to 0.05 (idle, I guess).
Search 'spinstopper' in https://github.com/OrneryD/RaceFlight-One/blob/current/src/flight_controller/src/pid.c
I think the approach we are follwing is better.
Might be interesting to see how/if RF took specific measures to deal with overflow in ICM206xx's in 32k gyro mode.

martinbudden, 2017-12-16:
@denovich , the slew filter is not being removed. This checking can be enabled in addition to the slew filter.

etracer65, 2017-12-16:
@martinbudden The YSTTM gyro threshold and the slew filter won't play nice together. The slew filter (when working) will "hold" the last gyro reading as soon as it detects the next reading to be more than 2<<14 away.  It'll continue to hold this until a reading is found that isn't more than 2<<14 away and then the cycle repeats.  So in the context we're seeing from the gyro in overflow, it'll hold the 32767 raw value as soon as the first inversion occurs.  But the YSTTM gyro threshold will also be triggered because of the high gyro rate and the PID's will be zeroed.  So the "benefit" of having the slew filter kind of "correcting" the gyro keeping the "last good" value and letting the PID controller actively work to correct the spin will be nullified.

Also the slew filter is "fragile" in that it's only looking at values one sample apart.  We know the data coming from the overflow is extremely noisy and chaotic.  The slew filter seems to work most of the time when in 32K mode, but I've seen cases on the test rig where it arbitrarily holds some intermediate value because it was within 2<<14 of the previous and is in the transitional region during the gyro overflow.  Worst case is this happens multiple times during the noise period and ends up locked in the inverted portion of the range.

The YSTTM threshold detection logic will always work and because of the hysterisis in the delay and lower reset range it won't re-trigger falsely.

martinbudden, 2017-12-16:
Agreed. The slew filter and the overflow detection should not be used together.

I'd sort of assumed the slew filter was a config setting, but I've just checked and it is just a build flag. I'll change it to being a config setting, and then the user can use the overflow or the slew filter as they see fit.

etracer65, 2017-12-16:
@martinbudden I'll try to get the revisions to the overflow detection tested.  I have an indoor race to go to tonight so I can't guarantee I'll be able to get to it (have to complete some repairs!).

Overnight I was thinking about yet another solution...  kind of a hybrid approach.

The good things about the YSTTM overflow detection are that it's reliable, shouldn't false detect, and can be useful for all uncontrolled spin events (not just gyro overflows).  The bad thing is zeroing the PIDs leading to uncontrolled ballistic flight until aero forces slow down the spin.  I feel like this will substantially reduce the recovery of gate clips, etc. in general as most occurrences don't result in a YSTTM event.  Experimentally we've seen that the ICM206xx gyros are fine up to 4000dps so the 2000-4000dps range which would be recoverable is now going to be in the zeroed PIDs region.

The slew filter (when it works) is very good at recovery because it makes the affected gyros appear to not have had the overflow to the PID controller and recovery is at the maximum.  The bad thing is the slew filter's detection is unreliable and somewhat fragile.

So I was thinking about a design that takes the YSTTM overflow detection method and combines with the slew filter's concept of the gyro data supplied to the PID controller.

So detect the overflow with the same kind of ysttm_threshold, use the 25ms grace period and the hysteresis of the ysttm_reset - but do this on a per axis basis.  At the time the overflow is detected, save the current gyro reading for the axis (don't really need the actual value, just the direction really).  While overflow is detected, instead of zeroing all the PIDs, use the stored gyro rate for the currently overflowing axis rather than the measured reading.  Once the overflow event stops for that axis then revert back to measured readings.

So the idea is that the PID controller can attempt to aggressively recover because it will have "good" data (at least the correct direction and that the rate is at least 2000dps).  There would probably be some overshoot on the recovery because of the gap between the threshold and the reset as the pid controller will "see" 2000dps all the way until the gyro gets back to 1800 and the detection resets.  This probably won't matter as the pilot had likely already centered the sticks and the PID controller is trying to get back to zero on the axis anyway.  This could be mitigated by reducing the reset gap.

Won't be able to do it today, but maybe I'll try to work up an example pull request.

denovich, 2017-12-16:
The spin stopper when enabled was super effective. I was onnof the first to put it to the test. I had to suppress the reflex to disarm, but I basically flew through a tree. Hitting a branch it would be “stunned” for a moment. And by the time my eyeballs caught up to the action I had contol again, until the next branch.

But the feature was short lived. Some builds had it, some didn’t. Kalyn wasn’t a fan for some reason that he wouldn’t reveal at the time.

martinbudden, 2017-12-16:
@etracer65 , you've probably heard me say it before, but one of my rules of development is "Get it working and then get it working well". As you point out, there are several ways this can be improved. However, rather than continue working on this PR, I'd rather get it to a state where we are confident it works and then merge it in. We can then start on a new PR which incorporates your suggestions  - indeed I've had similar thoughts myself.

One of the problems is - can we reliably know sign of the spin? That is suppose you are flying with no yaw and hit a gate and that causes you to spin at 4000 dps. If the internal gyro readings are something like 0,1000,2000,3000,4000, that is the quad spins up to 4000dps over the course of 4 gyro readings then you ascertain the sign of the spin. But suppose the internal gyro readings are 0, 4000 - ie the quad spins up to 4000dps in just one reading. Then your sign will be incorrect. So the question is: can a quad go from 0 to 4000dps in 125 microseconds? I don't know, I think probably not, but it might be possible.  I need to do a Fermi estimate...

etracer65, 2017-12-16:
@martinbudden That's fine - just throwing out food for thought.

Let me know when the code is ready to test again.  I saw a lot of updates and it currently doesn't compile.

```
./src/main/sensors/gyro.c: In function 'checkForOverflow':
./src/main/sensors/gyro.c:641:17: error: too few arguments to function 'cmpTimeUs'
             if (cmpTimeUs(currentTimeUs - overflowTimeUs) > 25000) {
                 ^~~~~~~~~
In file included from ./src/main/io/beeper.h:20:0,
                 from ./src/main/sensors/gyro.c:59:
./src/main/common/time.h:40:27: note: declared here
 static inline timeDelta_t cmpTimeUs(timeUs_t a, timeUs_t b) { return (timeDelta_t)(a - b); }
```

ledvinap, 2017-12-16:
@martinbudden : Hitting quad arm with small hammer should provide some estimate of angular acceleration involved in crash. My wild guess is that angular acceleration will be order of magnitude lower.
Impact interval on golf ball is about 0.5ms ...

martinbudden, 2017-12-16:
@ledvinap the angular acceleration is not caused by the quad arm hitting the gate, it's caused by a rapidly spinning prop hitting the gate.

For a 2500kV motor at 16V thats 40,000rpm unloaded. Say 30,000rpm loaded. 500 revolutions per second, 180,000dps. Say half of that angular speed is transferred from the prop to the quad in 0.5ms. So we get 90,000dps transferred in 0.5ms, ie 180,000dps in 1ms or 180dps in 1microsecond. So in 125 microseconds we get 22500dps.

So clearly half of angular speed is not transferred from prop to quad. But if a tenth is transferred then we still hit 4500dps,

So can quad accelerate from 0 to 4000dps in 125 microseconds? It certainly seems feasible from above estimates.

martinbudden, 2017-12-16:
@etracer65 OK, it's ready to test again.

etracer65, 2017-12-16:
You'd also need to take into account the conservation of  momentum in comparison of a probably 15g prop/bell vs. 400g+ quad though.

martinbudden, 2017-12-16:
Indeed. But it could be 15g propellor and 250g of quad. And also some of the forward momentum of the quad will be converted into angular momentum.

So it is entirely feasible that after clipping a gate we don't even know the direction of the spin.

But we could use the accelerometer data to at least get the direction of the spin.

brycedjohnson, 2017-12-16:
Was anyone getting in touch with invensense?  I remembered seeing something a long time ago but not sure if anything came out of it.  If no one wants to contact them I could shoot a message to their support...

etracer65, 2017-12-16:
I emailed them again on Monday and posted in their developer forum.  No response to either yet.

ledvinap, 2017-12-17:
@martinbudden : IMO structure stiffness will be main limiting factor when transferring momentum. But my knowledge in this field is very limited .. 

Accelerometer data will disappear when axis of rotation aligns with gravity. And precession may generate unwanted signal - it may be quite hard to get reliable data

[19 more replies]

closed 2018-05-05 by etracer65