betaflight/betaflight #4739: Current sensing/claibration bug

Dimand, 2017-12-11:
So I spent a bit of time last night calibrating my current flight controller as it seemed like the current was reading low and my batteries were suffering. Using a bench power supply and a Keysight calibrated current meter I set about measuring the correct scaling factor and offset for the particular shunt resistor on my board.

Now while the scale seemed to make sense I could not figure out why the values put into the offset made things go crazy. I have calibrated thousands of other instruments over my career and have never had such issues.

So this morning I cloned the repository to have a look at the code and found this in main\sensors\current.c

```
static int32_t currentMeterADCToCentiamps(const uint16_t src)
{

    const currentSensorADCConfig_t *config = currentSensorADCConfig();

    int32_t millivolts = ((uint32_t)src * ADCVREF) / 4096;
    millivolts -= config->offset;

    return (millivolts * 1000) / (int32_t)config->scale; // current in 0.01A steps
}
```

The issue here appears to be that the offset is been subtracted before the scaling factor is applied. 

This should be in the form y=mx+b (as current sensors are generally linear sensors) but is currently in the form of y=mx-mb.

Can I suggest the function be changed to something like this

```
static int32_t currentMeterADCToCentiamps(const uint16_t src)
{

    const currentSensorADCConfig_t *config = currentSensorADCConfig();

    int32_t millivolts = ((uint32_t)src * ADCVREF) / 4096;
    millivolts /= (int32_t)config->scale;
    millivolts += config->offset;  // I guess you could keep this as a minus if you really want.

    return (millivolts * 1000); // current in 0.01A steps
}
```
The scales used here also make things difficult but I guess thats the price we pay for keeping everything as integers. Is the performance hit bad enough that we can't switch to floating point values and keep everything in sensible units?

Hopefully a change like this helps all the people that have had a hard time figuring out why they cant get their offset right and have been living with bad current values at low amps.

mikeller, 2017-12-11:
What you say makes good sense, @Dimand, thanks for looking into this.

Changing to float will not have an unacceptable performance penalty on the currently (as of 3.3) supported MCUs, they all have FPUs.

The only negative impact I can see in this change is that all of the targets that define custom defaults for the calibration of their on-board current meters will have to change the default calibration (or rely on the user to calibrate before use).

Can you come up with a pull request to get this changed?

Dimand, 2017-12-12:
Thanks for the feedback @mikeller. I also understand that changing this will impact factory calibrations and user configs, however an incorrectly calibrated current sensor will have no effect on actual flight and voltage warnings will still tell pilots to land no matter what the current is set to in an attempt to save batteries. As it is betaflight defaults to current values that have a decent margin of error most of the time anyway.

I will have a go at writing up something and making a pull. Be warned that my knowledge of c and git are ameture at best.

mikeller, 2017-12-12:
No problem, we're happy to help new contributors. And thanks for looking into it.

grizewald, 2018-01-03:
Please, don't merge this!

The way it is programmed is entirely correct!

If the current sensor is a shunt resistor, the offset should always be zero.
On the other hand, if the current sensor is a Hall effect sensor, it is likely to have a non-zero starting voltage and this is what the offset is used for.

Merging this will break things.

Dimand, 2018-01-03:
The voltage offset should be zero on a shunt resistor but in reality that is almost never the case, there are many potential causes but often it is the ADC that causes the offset to be required from my experience.
This code still allows for any offset needed on either hall effect or shunt resistor sensors and removes the dependence of the offset on the scaling factor which makes more sense mathematically. Also most pilots are going to calibrate their systems by directly measuring current as you can combine all the linear offsets into the one correction.

jflyper, 2018-02-23:
@grizewald I have a new comment on #4743