betaflight/betaflight #4740: pid_sum_limit not appearing to work

ctzsnooze, 2017-12-11:
I was recording yaw spin events and noticed in the blackbox logs that PID sum was exceeding the limits set in the CLI.  This failure to constrain PID_sum on yaw, in particular, is probably exacerbating the yaw spin problem.  It appears that there is absolutely no constraint on PID_sum, not that appears in the logs, anyway.

If there is limiting, shouldn't it appear in the logs?

Code is nightly build of 3.3.  My CLI settings:

```
set pidsum_limit = 500
set pidsum_limit_yaw = 400
```

Here is pid_sum for roll at 700 or so...

[image]

Here pid_sum for pitch at 1131

[image]

Here pid_sum_yaw during a yaw spin at 1204

[image]

Here's the blackbox log from which the above images were taken.

[YSTTM_event1.BFL.zip](https://github.com/betaflight/betaflight/files/1549675/YSTTM_event1.BFL.zip)

etracer65, 2017-12-11:
@ctzsnooze  Just to be clear since I'm not sure...  Is this with your custom build where you were limiting the PID response above a certain gyro threshold?  If so maybe some unexpected effect of that mod is affecting the limiting?

I looked at my logs from the spin rig and the overflow events with runaway yaw and the yaw PID_sum seems to limit at about 460% (default settings in this test).

In the screenshot you can see the gyro inversion from the overflow (failed slew filter because not running in 32k mode) and then the yaw PID_sum clearly limiting.  This is with the ICM20608G gyro.

[image]

ctzsnooze, 2017-12-12:
Maybe I don't know what the numbers should be.  My PID sum on yaw was 120% but the numerical value was very high.  When the yaw limit is set to 400, what limit should I see in the numerical values of the log?  I was expecting no more than 400.  But I see over 1000 on yaw and over 700 on some rolls.  It isn't my custom code because that only sets the individual elements of the PID system to zero when yaw exceeds 1500 deg/sec.  What I'm getting is a PIDsum that is extremely high when that code is not active.

ctzsnooze, 2017-12-12:
I mean to say that blackbox is showing the highest possible value it can display for PIDsum

etracer65, 2017-12-12:
@ctzsnooze Ok, I did some digging... It looks like the PID_sum limiting is working properly but you can't see that in the blackbox.  The code doing the limiting is in mixer.c:

```    float scaledAxisPidRoll =
        constrainf(axisPID_P[FD_ROLL] + axisPID_I[FD_ROLL] + axisPID_D[FD_ROLL], -currentPidProfile->pidSumLimit, currentPidProfile->pidSumLimit) / PID_MIXER_SCALING;
    float scaledAxisPidPitch =
        constrainf(axisPID_P[FD_PITCH] + axisPID_I[FD_PITCH] + axisPID_D[FD_PITCH], -currentPidProfile->pidSumLimit, currentPidProfile->pidSumLimit) / PID_MIXER_SCALING;
    float scaledAxisPidYaw =
        constrainf(axisPID_P[FD_YAW] + axisPID_I[FD_YAW], -currentPidProfile->pidSumLimitYaw, currentPidProfile->pidSumLimitYaw) / PID_MIXER_SCALING;
```

The PID_sum values for each axis are not fields written to the blackbox log.  Instead they are calculated fields computed by the viewer as a simple sum of P + I + D for the axis.  Since the original unconstrained P, I and D values are written to the log you can't really see the limiting happening in the mixer.

I guess it would be nice to have the actual constrained values used in the mixer written to the log as it would give a truer picture of what's happening.

ctzsnooze, 2017-12-12:
It would be helpful to see that constraining.  Close this and make a new issue?

etracer65, 2017-12-12:
Probably a good idea.  The new issue would be a feature request rather than a bug report.

ctzsnooze, 2017-12-13:
Michael Keller pointed out that this isn't a blackbox viewer issue, but an issue for betaflight to determine what should be logged.  So this 'issue' should remain open.

He agreed that the default data to be sent to blackbox should be the actual limited PIDsum values, with perhaps a debug alternative of the non-limited values.

etracer65, 2017-12-13:
Bug report or feature request - whichever is easier to track is fine with me.

However this is going to be bigger than just sending the constrained values to the log rather than the original PID calculated terms.  This is because the limiting happens on the sum of P + I + D an then is constrained to +- the pidSumLimit.  In the log currently doesn't contain the pidSum, only the individual P, I and D values.  The blackbox viewer recalculates a display value by adding together the discrete P, I and D again.  Since there's no reasonable way to convert the constrained pidSum back into it's P, I and D components you can't write limited values back to the log as individual elements.

So I think the enhancements will need to involve adding an actual pidSum element to the log for each axis that will contain the constrained value used in the mixer and then modifying the blackbox viewer to have it use the actual value rather than calculate it from the P, I and D values.  Not sure any additional debug values are needed since if the "old style" pidSum is wanted in the blackbox viewer then it can simply be the calculated value like it is now (instead of the "new" actual value in the log).

Might get a little tricky in the viewer to check versions of the log and behave in the new mode vs. the current if you are trying to view an old log as it won't have the pidSum values.

mikeller, 2017-12-13:
How about just adding a debug value for the limited PID sum? That would do the trick without any compatibility woes.

ctzsnooze, 2017-12-13:
Yes sorry this is a bit of a can of worms.  Debug alternative sounds simplest.

etracer65, 2017-12-13:
That's fine - if the goal is to simply debug that the pidSum limiting is working (which I'm confident it is).  Sounds more like the goal is that it "more correct" to report the used pidSum values rather than the incorrectly calculated values in the viewer as they are meaningless.  Also you can only have one debug_mode at a time so you couldn't see "real" pidSum and raw gyro data (for example).

I don't think the changes to do it the right way are too tough - I was just noting them.  The blackbox viewer already has some version checking logic as new fields have been added over time that needed to be conditionally handled.

Communication would be important so that everybody understands that the pidSum values have changed to be actual representations.  People often look at the pidSum in the log for an indication of the activity on an axis and might be confused by the change.

Alternately we could leave the existing calculated pidSum alone and simply add a new "constraned pidSum" as a new field - basically the same as adding it as a debug field.