betaflight/betaflight #4742: FPV cam control; send to OSD menu option, disable OSD (for boards with built-in OSD)

csurf, 2017-12-12:
Feature request for FPV camera control for boards with built-in OSD:
- Can we please move the FPV cam control access into the betaflight OSD menu?

I was thinking of something like this:
- when *not* in betaflight OSD menu, disable the stick commands for the FPV camera control 
- in order to enter FPV camera control menu, user must enter OSD menu and select a menu option 
- At this point, completely disable/turn off the betaflight OSD (so that the FPV cam menu is more clearly visible)

- We'd need to detect when the user is finished working within the FPV cam menu:
-- use a stick combo to let FC know that we're done (more complicated and clunky?)
-- use a user-configurable stick inactivity timeout. Once the timeout expires, re-enable betaflight OSD and send user back to previous position within the OSD menu (seems simpler to implement it this way)

is something like this feasible?

DieHertz, 2017-12-12:
How would you achieve that if CMS and cam control use the same sticks for navigation?

csurf, 2017-12-12:
so, are you saying something like an additional, external 'exit button' that the user can somehow navigate to in order to close out of the 'mostly-empty-OSD-menu-screen" that's meant as a background/placeholder for the FPV cam menu? Or, am I getting it wrong?

I'd be cool with however this could be worked out. If there's an easy way to clearly exit out of 'joystick mode' that also signals that "it's time to also exit FPV cam control mode", then great. Maybe by simply lowering the throttle stick? dunno

DieHertz, 2017-12-12:
Most cameras have more than one menu triggered by enter today, this proposal doesn't take them into account at all. I don't feel like camera control should be integrated into bfcms in current state, it's impossible to get it uniform and complete.

csurf, 2017-12-12:
this is why I think timeout would be easiest option. Just let it expire and things go back to normal. it would only have a meaningful effect if you had already decided to 'enter camera settings mode' via the menu option. Other than that, the OSD is completely separate from what happens in 'joystick mode'

DanNixon, 2017-12-12:
@DieHertz Integrating it isn't what was being suggested. The same stick commands would be used either way, this would just add the ability to blank the BF overlay.

csurf, 2017-12-12:
@DanNixon,
yeah, blanking the overlay and keeping from accidentally going into "joystick mode" are the main goals here. blanking is simply to make it easier to work with FPV cam settings OSD. Addning a BFCMS menu option for "cam control" simply helps to keep you out of "joystick mode" until you really want to use it...

DieHertz, 2017-12-12:
I have a change in the works that allows setting camera mode on a switch to trigger it with the old-fashioned stick command, OSD render may be disabled while this mode is active. Don't see a reason to duplicate it as a CMS menu item

csurf, 2017-12-12:
seems like a waste of a switch though and overcomplicates things.
What's the issue with adding it to the menu? seems more straight-forward IMO

DieHertz, 2017-12-12:
Doing it from the OSD doesn't as anything to the equation.
As I already mentioned, there are cameras that have more than one menu triggered by different keys, adding all of them to the CMS is not viable, better leave it to the actual user of each camera.
Disabling the BF OSD display can be achieved by the change I'm working on.

csurf, 2017-12-12:
@DieHertz I believe you might be misunderstanding the concept. 
@DanNixon, you mind chiming in here? Perhaps I'm not explaining it well enough...

At the bare minimum, CMS menu option would only do two things:
- clear the screen of all CMS/OSD overlay graphics
- wait for a stick timeout to determine when to reenable CMS/OSD overlay

It would have absolutely nothing to do with joystick mode or trying to interpret the joystick commands that are being sent. It simply temporarily disables the CMS/OSD, waits for a stick idle timeout to expire, and then re-enables it...

DanNixon, 2017-12-12:
From what I gather @DieHertz 's suggestion is pretty much the same as yours, except activated via a stick command or mode rather than a CMS menu entry. IMO that is a better approach for the sake of being more generic.

@DieHertz 
> adding all of them to the CMS is not viable

That isn't what I was suggesting, the OSD and camera control would remain completely separate apart from camera control stick commands only being allowed in this specific menu.

csurf, 2017-12-12:
well, whatever works I guess, but the feature request was for sending it to the CMS for boards that support it. It avoids having to add another mode to the ever-growing list, and having to sacrifice an AUX switch for such a trivial task. It doesn't seem like that much of a big deal to add an option to temporarily disable the OSD (and enable joystick mode) *within* the CMS itself (keeps things together logically by task).

wind0r, 2017-12-12:
If it a trivial task simply implement it as a proof of concept and create a pull request to discuss further

Also why you need a CMS Soltuion? 
You cant free up any stick command with the CMS Entry. You still need all currently implemented stick commands for controlling the camaras. 
To disable OSD while in CamControll for sure is good feature, but handling timeouts and monitoring inputs will not be possible in a clean and generic way for all camaras.

So the Disable OSD on a switch is the most sane solution and the best part is not a single line need to change since it is allready implemented ;)

csurf, 2017-12-12:
lol, so much antagonism...
This was a feature request, and not a pull request...

The purpose of a "feature request" is to seek the expertise of those who are already familiar with a specific aspect of the code in order to propose a new feature. If I was already up to speed on the CMS code, I wouldn't have wasted time posting the request. I would have simply hacked together the "proof of concept" that you speak of (which is something I might now have to consider doing since there doesn't appear to be much support around here).

BTW, I don't see you you mean by "a clean and generic way for all cameras". This is completely independent of the 'joystick mode' functionality and has nothing at all that is "cam-specific". It doesn't try to interpret the joystick commands at all. It simply disables the OSD and ignores the stick inputs while the user adjusts the cam OSD. Once the user stops moving the sticks, the 'blank CMS menu screen' will time out and everything will return to normal.

having to waste an AUX switch and add yet *another* freakin mode that needs to be enabled & disabled on a per-need basis is a crap solution that isn't future-proof, unless the plan is to regress back to the oldskool methodology of the multiwii days. If there's *now* a built-in OSD (CMS) for FC configuration, then why not use it? makes no sense to keep stacking more switches, especially for this type of adjustment...

DieHertz, 2017-12-12:
Seems like we have another trollcop here, guys

wind0r, 2017-12-12:
This is nothing against your idea or person. Developers just write harsh sometimes if people do get the further problems and think everything is easy ;)

With openTCO and/or CRSF more cam controll features will be possible, but currently we just are emulationg joysticks which has the limits described above and below.

"The purpose of a "feature request" is to seek the expertise of those who are already familiar with a specific aspect of the code in order to propose a new feature"

And execlty those people tell you that your request need another thought. Writing multply fast message doesnt help there. Please read all messages again and try to understand the problem better. @DieHertz  even told you so befor. 

This has nothing to do with antagonsim. Sometimes people simply get annoy telling the same stuff multply time also writing that this Request is easy and ever other solution is overcomplicates doesnt help either. Also sentence like "another freaking mode" and  "crap", "waste" arent word you want to read when you deal with such stuff.

Please just keep it classy.

To keep it "short"
"We'd need to detect when the user is finished working within the FPV cam menu"
This isnt possible so all other requests can not be implemented

or a little longer:
"'blank CMS menu screen' will time out" as other and I allready said this isnt possible since every cam is different (a clean and generic way for all cameras). For sure someone simply can add a timer which counts down and enable the osd again but this will result in plenty of problems. (User may still has cam osd open, cam osd may allready closed and user is confused why osd isnt back and different other cases)
Having a sane and repeateable solution is to prefer.

So it just some back to a CMS/OSD entry to enable cam Controll, that may could be nice but i wonder if it is worth. Just entering the Stickcommand ist faster also having the cms entry will not remove the stick command. You still need to remember that to navigate to the cam osd having a bfosd entry will not replace/remove any stick command.

"and send 'ENTER' joystick command to FPV camera in order to enable camera settings menu"
as @DieHertz told above. just sending enter isnt possible. some cams need a long press some need a short and some another pattern.

"What's the issue with adding it to the menu?"
Having a osd entry which just sends enter will create plenty of issues we do not want to adress. We dont have that nice easy world here. Thats what we try to say in multply posts.

"having to waste an AUX switch and add yet another freakin mode that needs to be enabled & disabled on a per-need basis is a crap solution that isn't future-proof"
beside those nice words even with a cms solution something like switch need to be implemented for the users who dont have a osd and also you simply do not need to use that.

I personaly also do not see any use of a new mode or switch. As soon as you armed stick commands are disabled and my osd doesnt discurpe the cam osd and if it would i simply would use the feature to disable the osd.

Just a personal opinion and further context
"seems like a waste of a switch though and overcomplicates things.
What's the issue with adding it to the menu? seems more straight-forward IMO"
I would hold back which such statements. The solutions talked about are way more straight forward then the advisde one. Also it allready was descriped why the menu cant simply be added and also your implementation wouldnt be that straight forward as you think. You said "keeps things together logically by task" .. this is exaclty what betaflight does. cam controll is one task, radio channels, osd is one task and timing is one task... so the you need to connect all those task just for having a cms entry which disables osd and enables it (unreliably) after some time.

Abancalari, 2017-12-12:
If yaw right  serves as enter, why can't yaw left serve as exit?

csurf, 2017-12-13:
@Abancalari, yes... it seems you "get it" as it's not as complicated as it seems.

@DieHertz, actually trollcop was equally against adding new features and didn't approve of anyone trying to 'think outside the box' and/or doing things differently from what he was used to. 
...Wait, isn't this also your stance on the issue???

@wind0r, again, you've completely misunderstood the concept and have completely overcomplicated the original concept. 
There are many things wrong with your response:
- you continue to insist that this must somehow be tied to camera control. it isn't. I have no idea what you are talking about with avoiding stick commands or trying to combine them. All this is meant to do is to DISABLE joystick mode until you go into the CMS menu and then manually ENABLE it by selecting the appropriate menu option.  At the same time, the screen would be blanked simply so that you can work with the FPV cam OSD. At that point, you can then enter your FPV cam settings without having the betaflight OSD getting in the way 

- The stick timeout (or stick movement detection) was simply an idea of how to signal to the CMS that it's time to re-enable the OSD. In fact, this entire feature could work stand-alone without the user ever having entered the FPV camera menu at all. It's not very complicated... it simply would require a CMS option that disables the OSD and places it in stand-by until signaled somehow (yaw left, timeout, throttle low, or whatever).

- The issue you claim with 'automatically sending the ENTER command isn't compatible with all cameras' wasn't essential to getting this feature to work. It was just a small 'extra' bonus feature that I suggested in order to help speed things up by skipping a step, but it's not necessary or critical to the feature implementation. It could simply be left out, and then leave it to the user to manually enter the FPV cam menu; problem solved.

- moving forward, most boards targeting betaflight will have an on-board OSD (many, if not most, already do). Boards that don't have an OSD can simply fall back to the 'aux mode' switch (which @DieHertz claims to have already developed anyway). This feature request was for BOARDS WITH A BUILT-IN OSD. If the CMS is going to be

- The current implementation of FPV cam control is basically broken. It has joystick mode always enabled (except when armed apparently), so it's possible to enter the FPV cam OSD at ANY TIME, with the betaflight OSD graphics in the background, thus confusing things. Plus, it's very easy to accidentally enter the FPV camera OSD by inadvertently messing with the sticks. 

My idea was to simply DISABLE joystick mode in the same way that it's disabled when the quad is armed. In addition to blanking the OSD.

- my annoyance is with developers that tend to lack the ability to see things through the perspective of others and feel compelled to force their views and opinions upon others about what they consider to be the right solution. A perfect example of this behavior is when someone requests "feature A", and then someone else comes along and says "no, I don't agree with your request, I prefer to take your request and convert it into feature B... trust me, it's *better* than yours, so we must now do it my way..." This is where egos clearly get involved and things become very counter-productive... 

- If you don't agree with a feature request, simply ignore it and move on. The rest of us will either determine if it's feasible and try to implement it, or scrap it. There's no need for people to come and try to hijack it...

For those still interested, If I have time this month, I will try to dig into the CMS/OSD code and see what can be done about mocking up the 'dummy / blank screen' menu option and hacking together a proof of concept. 

All other critics are welcome and encouraged to simply move on...

wind0r, 2017-12-13:
You also may should think outside the box ;)
It currently possible to Enter ANY StickCommand at ANY TIME!
There may should be cms entry and aux modes for alle the different Commands why only for cam controll.

There may just should be a CMS Entry  and AuxSwitch to disable Stick Commands at all.

Remember if you play around with the Sticks while disarmed. You can do more dangerouse stuff then just open Cam Controller. You can recalibrate Gyro and Accel which totaly isnt good when Quad isnt leveld or in your hand. You can Switch Profiles which also isnt that cool if you dont do that on purpose and even worse you can Arm your Quad with Stick Commands.

We need to find a better way else the next Person will come around and say i accidentally changed pid profiles there should be a cms entry for that and so on.

For the time befor that you simply can setup a close range for the min_command max_command so this only happens if you sticks are far far outside and waste a switch for disabling osd

Woody9999, 2017-12-15:
An additional consideration is that not everyone has OSD on their quad. I've successfully setup the cam control on 3 quads and it's working well with the stick commands. 
A switch to enable the function would be fine too, but there are probably more important things to work on.

KnzHz, 2018-02-23:
In fear of stepping into a hornets nest, as a user and Github lurker, let me add my two cents, trying very carefully not to step on anmybodys toes here:

I have been thinking the exact same, why not add camera control "entry" to OSD, it is way more user friendly. I agree that the biggest drawback is "how to get out of it". In the name of full disclosure I would prefer to get rid of ALL stick commands that are not run through OSD or activated that way, it seems to me that stick commands for calibrating etc. is very old school and "in the know" instead of clean and user friendly from a "new user" point of view. Don't get me wrong, I don't propose killing them off, but a setting to disable all other build in stick commands other than OSD and the features implemented there would be preferable in my book, I never use them other than when forced to because there's no other way. Also, my memory sucks so I have to google the stick commands each time :P

I think this argument is almost the same as "why add GUI option when there's already a command line option, just use that". Yeah, well, it's not that user friendly or clean. 

I suggest maybe not removing OSD completely but removing all elements and then on the bottom of the screen write comment on how to get out of camera menu like "Re-enable OSD by full pitch, full throttle" or whatever command you know won't happen in any menu navigation. Yes, users can do that without exiting camera control but they can do that with a switch as well, yes? And the point wind0r made is important, if you enter camera control(which is outside of beat flights control other than forwarding commands) all other stick commands should be disabled.

I also like the idea of having it on a switch, I just don't see why we can't have both, the OSD menu now is so freaking cool, I use Deviation so not having LUA scripts on OpenTX, OSD is my best friend :)

regnets, 2018-05-09:
I am absolutely agree with @KnzHz , currently it is awkward to enter the camera menu with the stick command and feels out of place. Sometimes i (as a noob) get both menu overlapping each other and i am working as an IT-Consultant. I don't want to know how confusing this sometimes has to be for someone who isn't that techie.

tavdog, 2022-05-23:
Time to close?

haslinghuis, 2022-05-23:
Fixed in #6425

closed 2022-05-23 by haslinghuis