betaflight/betaflight #4748: BF3.2.3 - SPRF4E target - ESC telemetry not working

dk7xe, 2017-12-14:
FW - BF3.2.3
FC - SPRACINGF4EVO
Configuration - Octo X flat
ESC - KISS32A FW118e 

After upgrade from BF 3.2.0 to BF3.2.3 the ESC telemetry values for combined voltage and current are showing 0V / 0A while the individual ESC values are correct. with BF3.2.0 everything has worked correctly.
It seems something is broken in the code to sum up the A values and calculate average of the V values.

[image]

cli dump
`dump cli

# version
# Betaflight / SPRACINGF4EVO (SP4E) 3.2.3 Dec 11 2017 / 08:04:33 (cb962eda1) MSP API: 1.36

# name
name RQR-Monster

# resources
resource BEEPER 1 C15
resource MOTOR 1 C06
resource MOTOR 2 C07
resource MOTOR 3 C09
resource MOTOR 4 C08
resource MOTOR 5 B06
resource MOTOR 6 B07
resource MOTOR 7 B01
resource MOTOR 8 B00
resource SERVO 1 A09
resource SERVO 2 A10
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A03
resource PWM 1 A03
resource PWM 2 A02
resource PWM 3 A09
resource PWM 4 A10
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 A01
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 C10
resource SERIAL_TX 5 C12
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 C11
resource SERIAL_RX 5 D02
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource INVERTER 1 NONE
resource INVERTER 2 B02
resource INVERTER 3 NONE
[... 574 lines cut ...]
set gps_pos_i = 0
set gps_pos_d = 0
set gps_posr_p = 34
set gps_posr_i = 14
set gps_posr_d = 53
set gps_nav_p = 25
set gps_nav_i = 33
set gps_nav_d = 83

# rateprofile
rateprofile 0

set rc_rate = 100
set rc_rate_yaw = 100
set rc_expo = 0
set rc_expo_yaw = 0
set thr_mid = 50
set thr_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 10
set tpa_breakpoint = 1650

# `

mikeller, 2017-12-14:
Are **all** 8 individual ESC values showing? The sum / average will be set to 0 if data from one or more ESCs are missing (to indicate to the pilot that the values can not be trusted if one ESC telemetry wire has come off.

dk7xe, 2017-12-14:
Hi Mike, that behaviour i know. With BF3.2.0, just some minutes before changing to 3.2.3, everything worked ok. But i will check tomorrow again including reverting back to 3.2.0 to prove that it is 3.2.3 related.

dk7xe, 2017-12-15:
stupid me. issue was caused by low voltage battery

closed 2017-12-15 by dk7xe

mikeller, 2017-12-15:
Too low to boot up all ESCs?

dk7xe, 2017-12-18:
indeed.. i should have seen it on the screenprint ;(