betaflight/betaflight #4750: Onboarding Documentation: Developer/Hacker Resources

KazW, 2017-12-14:
Let me start this by saying that this is an amazing piece of software! As a web developer that uses many open source technologies professionally, using open source in my hobbies as well is just awesome-sauce.

I'd like to work on connecting a Raspberry Pi to a Betaflight FC and have telemetry analyzed in real-time using TensorFlow to predict/recommend tuning steps or repairs that should be done to the copter. Possibly, even display warnings to the pilot on the OSD, such as `Replace Prop #2`. Maybe even have AI tune the copter in-flight.

However, getting to know Betaflight as a developer has a few roadblocks. The lines in the `README.md` and `CONTRIBUTING.md` about developer docs mention a docs/development folder that no longer exists. Also, `build_docs.sh` appears to reference markdown files that no longer exist. After reading the wiki, I eventually found old copies of these files here: https://github.com/martinbudden/betaflight/tree/master/docs/development
The problem is, those lines that I mentioned and those files I tracked down are several years old. It's unclear how much still applies.

While the directory structure and file names are fairly well named and self-explanatory, for me, trying to find something specific involved a game of following the calls once I tracked down the file.

## TLDR; Proposal and Offer

I'd like to get to know the Betaflight codebase better and contribute at the same time. I want to start by writing documentation for the telemetry protocols so that hardware hacking is easier.

Before I write any markdown files and open a PR, I'd like to know if this would be a welcome PR and if there are any special considerations I should know about.

If this would be welcome, being pointed to any existing resources on telemetry protocols would be super helpful. Suggestions about which telemetry protocol is best or simplest to document first would also be appreciated. I'd like to document a telemetry protocol that outputs the same collection of data found in the black box files, at a high time-resolution.

Thanks!

DanNixon, 2017-12-14:
The `docs` folder was relevant in Cleanflight, in Betaflight most of the documentation resides in the Wiki (there was the suggestion to move back to the `docs` folder (#3914), but that has not happened yet).

Updating documentation is always welcome. Currently this could probably just go into the WIki (this is still just a Git repo with markdown documents so is easy enough to move back to the main repo if that ever happens).

> I'd like to document a telemetry protocol that outputs the same collection of data found in the black box files, at a high time-resolution.

It sounds like what you want is to implement a serial blackbox device (see `case BLACKBOX_DEVICE_SERIAL:`'s in [`blackbox.io.c`](https://github.com/betaflight/betaflight/blob/master/src/main/blackbox/blackbox_io.c)), this will not get the time resolution of flash or SD card however.

I'm not aware of any telemetry protocols that transmit *all* the data that goes into blackbox.

mikeller, 2017-12-14:
I agree with @DanNixon that serial blackbox output is probably the way to go for this.

The `docs` folder was removed from Betaflight exactly for the reason that it was never updated. I am trying to bring it back (#3914), in the hope of motivating people to update it. At the same time, I see value in using the wiki for a lot of the user facing documentation, because it sets a low bar for edits, and as such allows users that are not developers and / or don't want to burden themselves with learning git to update the documentation.

conkerkh, 2017-12-27:
I would suggest to look at MSP which I believe is still used on BF. It’s essentially bidirectional and you can do a lot of cool stuff like controll motors etc. Lastly I think it might be hard just to judge it on so little variables what’s going on with copter and suggested repairs although I must admit it’s very interesting😄