betaflight/betaflight #4751: define new hardware target

nyway, 2017-12-14:
Hi
According our demand we designed a type of FC hardware PCB , the interfaces definition are different with the current type of hardware , so we hope to customize a type of FC . Customizing a name for FC and then people could find and upgrade this FC in the betaflight firmware flasher list, For example we want to make a type of FC hardware which will be named with XXXX405.
My question is how could we acchieve that and what procedure does it take ?

richard-scott, 2017-12-14:
A Pull Request would need to be submitted but as for the changes you would require that would depend on the board you are making. 

This seems to be an example of a PR required https://github.com/betaflight/betaflight/pull/1783.

Maybe this process should be documented on the Wiki?

Have a look at this [search ](https://github.com/betaflight/betaflight/pulls?utf8=%E2%9C%93&q=is%3Apr+label%3A%22New+Target%22) to see previously added targets to get an idea of what information you may need to provide.

DanNixon, 2017-12-14:
Also if your new FC is based on an existing target you need only add is as a variant of the existing target (e.g. if only pin mapping is different).

See #2070 for an example of this.

mikeller, 2017-12-14:
Have a look at the existing target definitions in https://github.com/betaflight/betaflight/tree/master/src/main/target . For the board designer, these definitions should be pretty obvious. You will have to submit a pull request that contains a target definition for your target, and it will be reviewed and then accepted if it passes the review.

Just for you information: Target definitions that are merely copies of the design of an existing target with a new name are **not** acceptable and will be rejected.