betaflight/betaflight #4756: PREARM in the BFC Modes tab doesn't change colour when activated.

richard-scott, 2017-12-14:
I created an issue [ticket](https://github.com/betaflight/betaflight-configurator/issues/790) with the Configurator repo and was told to re-open one here.

When testing Modes the PREARM Mode doesn't change colour when you enable the switch on your TX. This is different to all the others which change to orange once the switch is activated.

I have been told that this is due to the Firmware not reporting the mode status back to the Configurator correctly.

dthubereng, 2017-12-15:
Should PREARM be prevented for any reason. I understand that ARM has number of prerequisites. I think I see where the PREARM is not getting set in msp_box.c, but I'm not sure of it's overall operation. I can take a shot at fixing this.

richard-scott, 2017-12-15:
Could this fix also go into the 3.2.x branch?

dthubereng, 2017-12-15:
I think it should, but I’ll wait for approval to do so.

mikeller, 2017-12-15:
@richard-scott: Is there any particular use case that you can think of that this fixes that justifies the risk of breaking things in 3.2 by cherry-picking this?

richard-scott, 2017-12-15:
It will fix the original problem... that is with the 3.2 branch.

mikeller, 2017-12-15:
@richard-scott: Can you describe the use case (i.e. 'problem') that you have with 3.2, and what exactly does not work for you in 3.2?

richard-scott, 2017-12-16:
@mikeller - Go into BFC, move to the Modes Tab and enable your "PREARM" switch that you have configured. The line doesn't get highlighted as the ARM, or HORIZON (I could go on) do so you have no idea that PREARM has been activated other than the RX signal marker is moved into the range you have selected for this.

This is with Failsafe selected, but not PREARM:
[image]

and this is with Failsafe not selected but with PREARM selected:
[image]

As you can see, the dark grey box on the left does not change colour.

richard-scott, 2017-12-16:
I've just tried build [#427](https://ci.betaflight.tech/job/Betaflight/427/artifact/obj/betaflight_3.3.0_ANYFCF7-427-59a6055cd.hex) of the 3.3.x branch and its still broken. Has this been merged yet?

dthubereng, 2017-12-16:
I think what Mike is asking is whether or not there is a case that this is causing an issue with the general operation of the FC beyond just not being able to see the PREARM condition highlighted when the mode is active? There is a potential risk of breaking the current 3.2 code and that needs to be weighed against what appears at this time to only be a purely cosmetic bug. Yes, it's a bug, but can it wait until 3.3 is released.

As for your last comment, my fix hasn't been merged into the master branch yet, so you won't see it fixed. You could try checking out my fix branch.

richard-scott, 2017-12-16:
oh, in that case then no, 3.2 is functional with it the way it is.

richard-scott, 2017-12-16:
I've just tried the latest 3.3.x branch and it works as expected.
Thank you very much :-)

closed 2017-12-16 by richard-scott