betaflight/betaflight #4760: Standardize OSD controls for camera vs. Betaflight OSD

Ryanf55, 2017-12-15:
Can the yaw and aileron controls for camera control and Betaflight OSD be standardized?

jflyper, 2017-12-16:
Can you elaborate? Do you want to change how sticks are used during camera control?

Ryanf55, 2018-03-24:
Sorry I did no see the reply. Basically, depending on whether you are in camera control, or in betaflight OSD, yaw and aileron function different, such as exiting the current menu, or changing the value. They are different. Is there a way to standardize it, or does that not work because the cameras all implement OSD contol differently?

mikeller, 2018-03-24:
@Ryanf55: Yes, that is the case here. Betaflight does just forward the stick movements to the camera as 'left button push', 'right button push'. How the camera menu then interprets the button pushes is up to the individual cameras.

Ryanf55, 2018-03-25:
Ok. I'll close the issue.

closed 2018-03-25 by Ryanf55