betaflight/betaflight #4762: 3.2.3 Betaflight F4 FC Turtle Mode

BoosiveFPV, 2017-12-15:
I have recently set up my board on the latest release. Everything is working correctly apart from Turtle Mode. When I activate TM my motors start spinning. Is this supposed to happen now? They never used to spin apart from when TM was first released. Am I missing something or has it gone back to its original functionality?

Please advise,
Boosive

brycedjohnson, 2017-12-16:
They should only be spinning when on the pitch and roll axis, nothing has changed on that since the early 3.2RCs.  You have to disarm -> go into turtle mode (crashflip) -> rearm.

BoosiveFPV, 2017-12-16:
That is exactly what I am doing but my motors spin as soon as I arm in Turtle Mode which theu shouldnt. Tried 3.2.2 and still the same as well.

brycedjohnson, 2017-12-16:
Want to grab a picture of your modes and a diff all in the CLI?  It is likely you have something configured wrong because it is working for everyone else.

BoosiveFPV, 2017-12-16:
Ive set up "flip on crash" to a switch. Flick switch to position, arm and then motors should only spin when I roll or pitch. It was working fine on my old board. Just since I changed to a new board so its super wierd that its gone back to the original TM. Its literally just one switch that you need activate for it to work.

phillipsj70, 2017-12-16:
Turtle mode doesn't work for me. I just gave up trying. Props don't spin correct direction to flip.

brycedjohnson, 2017-12-16:
@phillipsj70 Separate issue, You likely don't have your ESCs updated to a version that supports it.

@BoosiveFPV Follow the steps below.  You did 1 and 6 but none of the others.

## If you believe there is an issue with the firmware itself please follow these steps:
1.) Describe your problem;
2.) Include ways to reproduce the problem;
3.) Provide information about your flightcontroller and other components including how they are connected/wired;
4.) Add the used configuration and firmware version;
5.) Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/); and
6.) Remove this Text :).

BoosiveFPV, 2017-12-18:
Only the rear two motors spin when I arm in turtle mode. The front two react as normal. Only when I move the pitch/roll stick.

brycedjohnson, 2017-12-18:
Sounds like you don't have the your sticks calibrated or centered.  If you want help please provide the info above.  Looks like something isn't configured right, I don't think this is a firmware issue.

Or look here for more help:
## Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack: https://slack.betaflight.tech or other user support forums & groups (e.g. facebook).

BoosiveFPV, 2017-12-18:
That was it! Pitch was off so trimmed it out. Its always the simple things haha. Thank you very much for your help. Just tuning tomorrow and Im set for my event this weekend. Thanks again xx