betaflight/betaflight #4763: Esc 3 issue

alexsvq, 2017-12-16:
Hi everyone. I built a quad three weeks ago, i used first an old naze32 rev6 with Some dys30a esc ( the blue ones), with Bheli 16.6. When I was going to fly the aircraft, I saw that the esc 3 was forward respect the other three ones(1059 ESCs 1,2,4 and 1100 that 3 esc). I claim banggood for that and they sent me another esc. Magic happened when I checked that the esc3 was also with same problem. I have changed the FC( with a Dys F4 pro same issue).
I went desperate when I leave the quad to a Hobby mate for check the problem and when he mounted an F3 with simple pdb the esc worked fine. 
I have the same config that he used, and the worst stuff that can occur: One DIFFERENT quad ( a 120 ) that worked fine suddenly suffers the same s@h7'it.
What's the core of this ? I don't know what to do. The problem is real because when I go top throttle i have gigant vibration. The issue is always the same: The esc 3 is around 1100 when the others are on 1059 constant.
Firmwares used : Betaflight 1.7,1.2,1.21; Bheli 15.5,16,16.6,16.7
Motors have the orientation( CCW and start on Back right motor) : 1,3,4,2.
FCs used: Naze32 rev6, sp Racing F3, Dys F4 pro.
I'm pretty sure this is a problem of Betaflight because on kiss doesn't happens.

jflyper, 2017-12-16:
@alexsvq 
> I'm pretty sure this is a problem of Betaflight because on kiss doesn't happens.

I think you have some problem with basic configuration and/or hardware, as there are numerous happy users with SP Racing F3 and DYS F4 Pro.

Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack: https://slack.betaflight.tech or other user support forums & groups (e.g. facebook).

Btw,

> Firmwares used : Betaflight 1.7,1.2,1.21; Bheli 15.5,16,16.6,16.7

If 1.7 1.2, 1.21 are actually all Betaflight versions, they are extremely outdated.

closed 2017-12-16 by jflyper

alexsvq, 2017-12-16:
I have Submit my problem on RC groups and no one answer me. THE PROBLEM IS
NOT TRIVIAL, a lot of hobby mates have prooved my configuration and they
worked so, you are understimating my issue. I have seen Many problems in
github issue's forum that are of the nature of mine and the weren't closed.
I have used all the new betaflight firmwares since 2.1.7; and same with
Bheli.
I have changed all the hardware configuration and no one worked.
I use right the Betaflight and Bheli firmwares and the config because my
Group of hobby mates uses betaflight.
It still being a betaflight issue because on kiss It hasn't happens.

jflyper, 2017-12-16:
@alexsvq Okay, please paste output of CLI command "diff", either SP Racing F3 or DYS F4 Pro.
Please use the latest stable firmware.
I will personally load my test boards with your diff and see if something happens.

alexsvq, 2017-12-16:
Diff of sp racing f3, the lastest stable firm:

# diff
# Betaflight / SPRACINGF3 (SRF3) 3.2.2 Nov 11 2017 / 00:44:04 (77782e602)
MSP API: 1.36

feature -RX_PPM
feature -TRANSPONDER
feature RX_SERIAL
feature ANTI_GRAVITY
feature DYNAMIC_FILTER
serial 2 64 115200 57600 0 115200
aux 0 0 1 1700 2100
aux 1 13 0 1700 2100
aux 2 28 1 1700 2100
set baro_hardware = NONE
set serialrx_provider = SBUS
set min_throttle = 1050
set motor_pwm_protocol = DSHOT600
set align_board_yaw = -45
set small_angle = 180
set pid_process_denom = 1
profile 0

set i_pitch = 25
set d_pitch = 50
set p_roll = 62
set i_roll = 25
set d_roll = 50
set p_yaw = 60
set i_yaw = 25
rateprofile 0

set rc_rate = 170
set rc_rate_yaw = 160
set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 80

#

jflyper, 2017-12-16:
I'm a way from my bench now, and it will take a day or two to get back.

jflyper, 2017-12-17:
@alexsvq You said you have the same problem with DYS F4 Pro with exact same settings? My SPRF3 test quad has a gyro problem. I'll test with DYS F4 Pro and I need a diff from it if it's different.

alexsvq, 2017-12-17:
It's the same config: sbus, 2/8 GHz, dshot on 4,5... Including stock PIDs
and aux channels uses.

jflyper, 2017-12-18:
@alexsvq 
Okay, the local test quad was loaded with the above config and tested.

- I had to serial RX connection to RX1, in order to accommodate unmodded SBUS receiver.
- Upon arming and throttle at idle, all motors spin at 1045~1049. I consider this okay as the digital idle is 4.5%.
- The quad hovered just fine in angle mode.