betaflight/betaflight #4764: Quadcopter Flips Out

NinjaSauce1, 2017-12-16:
I have been having problems with a few of my quadcopters where they keep flipping out. After hundreds of hours of troubleshooting, I have scrapped 2 of those projects. My latest build came out pretty nice and its doing the same exact thing, so I have been determined to fix this problem. I have narrowed down the problem to very specific parameters.

Description of issue:
My quad flips out on the Pitch axis only on 4S batteries. Once it goes past a certain angle, it either gets "stuck" or it flips wildly and cannot recover.

I have tried everything that I can think of, so this is my last resort. My latest attempt included flashing the firmware to 3 different versions. None of them were any different. I have also compared the CLI dump & diff to some of my working boards. Nothing really stands out. 
My next attempt is going to delete Betaflight completely and re-install. I usually have a bunch of jibberish at the top of my CLI and I have always ignored it.
If anybody has any ideas, please let me know.
I should warn you that I have probably tried everything obvious, and it is far too much to list.

My board is the TattooF4S. 
Target: LUXF4OSD

I have been documenting the problem on youtube, so there might be enough info for someone to solve this problem, but I am completely stumped. 

Quadcopter Flips Out (&Blackbox): https://www.youtube.com/playlist?list=PLwip8UXl_WqjOHM6VekLUc4p2QQ21w3-T

If anybody has ANY suggestions, please let me know. 
Thank you

etracer65, 2017-12-16:
This probably a more appropriate question for the rcgroups forum (https://www.rcgroups.com/forums/showthread.php?2464844-Betaflight-Flight-Controller-Firmware-Discussion-Thread/), but posting more imformation about what the actual problem is would help.  Just saying it's "flipping out" isn't very helpful.  Flipping out when?  What are you doing when it happens.  What mode?  Angle/horizon/acro?

Also, if you want to link to a video illustrating the problem, link to it directly from here and tell us exactly where to look and what we're looking for.  Trying to watch a bunch of multi-minute long youtube videos to try and figure out what you mean is difficult.

Next if you have a blackbox log, zip it up and attach here.  Again describe what the problem is and where it appears in the log (time wise).  Otherwise again we're looking through multiple logs not know what we're looking for.

Lastly, post a "diff" output from the CLI so we can see your settings.

NinjaSauce1, 2017-12-17:
Ok. I apologize for posting this, but as I said, its a last resort. I have tried numerous forums and that is why I ended up with so many videos. Each time, someone would ask to see something different.

Im sorry for not describing the problem more completely. I have posted on so many forums, that I may have left out some info.

The exact problem is that my quad flips wildly on the pitch axis once it goes past a certain angle. This only happens on 4S. It feels like the gyros reset every microsecond once I go past about 80 degrees and as soon as I stop all stick input, thats the point where the gyro locks back in. Usually pitching at a random angle. If it happens to "lock in" at a reasonable angle, I can land, but it almost always ends up at an extreme angle, often inverted. As far as I can tell, the Pitch D term seems to be doing something that I cannot explain. It looks like it is doing its own thing and has extreme spikes regardless of the other values.  Its very difficult to explain, but I will link the best video. That being said, it flies perfectly on 3S. 
As far as the black box logs, if you look at any of them and select one flown on 4S, you will see the problem. The latest one has the debug notch on. With the Pitch PIDs & gyro graph, just look at any point where you see a large throttle jump and immediately after. I will try to point out a specific log as soon as I have time. I will also post the diff file.
The main reason why I am posting here is because this is the third quadcopter that has done the exact same thing. I was wondering if there might be a glitch in some firmware causing these types of problems. Honestly, Im out of options.

NinjaSauce1, 2017-12-17:
This happens in all flight modes.
https://youtu.be/7jvGdtnTdIY

Blackbox files:
https://www.dropbox.com/sh/0rev0x6my2el2xz/AADzOOzWG60woztlo_HLHVuea?dl=0

The first file on the top of the list has an example of a crash. Log #11. The crash is at the very end at 1:06.

The bottom file on the list has the notch debug turned on, so the gyro pre notch charts can be viewed. Log #5 shows a lot of throttle punches and pitch input. This is with the D term notch filter tuned. Shows a lot of spikes that may relate to the problem. 

I will get the diff as soon as I can.
Thank you!

NinjaSauce1, 2017-12-17:
Basically, I wanted to see if this is a known issue with Betaflight.
Or, am I just getting EXTREMELY unlucky?
Is it possible to have an instance of Betaflight that somehow adds a glitch into firmware?
Here is the diff:

# diff

# version
# Betaflight / LUXF4OSD (LUX4) 3.2.1 Oct 15 2017 / 19:38:39 (434db9a36) MSP API: 1.36

# name
name 225 Grams!

# resources

# mixer

# servo

# servo mix

# feature

# beeper
beeper -RX_LOST

# map

# serial
serial 0 64 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 900 1275
aux 1 1 2 1725 2100
aux 2 2 2 1300 1700
aux 3 13 1 900 1225
aux 4 28 1 1300 1700

# adjrange

# rxrange
rxrange 0 1100 1900
rxrange 1 1100 1900
rxrange 2 1095 1895
rxrange 3 1100 1900

# vtx

# rxfail

# master
set gyro_notch2_cutoff = 90
set baro_hardware = NONE
set min_check = 1010
set max_check = 2000
set serialrx_provider = SBUS
set dshot_idle_value = 250
set motor_pwm_protocol = DSHOT300
set pid_process_denom = 1
set osd_rssi_pos = 40
set osd_tim_1_pos = 54
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 377
set osd_crosshairs = 200
set osd_ah_sbar = 200
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_disarmed_pos = 138
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_min_rssi = OFF
set osd_stat_endbatt = ON
set debug_mode = NOTCH

# profile
profile 2

set dterm_lowpass_type = PT1
set dterm_lowpass = 90
set dterm_notch_hz = 156
set dterm_notch_cutoff = 90
set p_pitch = 35
set i_pitch = 35
set d_pitch = 30
set p_roll = 30
set i_roll = 35
set d_roll = 20
set level_limit = 72

# rateprofile
rateprofile 0

#

etracer65, 2017-12-17:
My first observation is that you seem to have some odd choices in your config.  Initially your digital idle is likely quite a bit too low.  The default is 4.5 and you're running 2.5.  This could lead to the motors stuttering or de-syncing - particularly on flips/rolls or transitioning from minimum throttle.  Even the default is often too low and needs to be raised as it depends on the motors and esc's.  You need to find the minimum speed that the motors will run reliably at under load.

Next roll/pitch P are very low.  While this won't likely lead to it "flipping" out, it could lead to instability when recovering from rolls/flips that could cause you to lose control. Also from the blackbox log pitch D looks to be too high.

Strange choices on the filtering.  The dterm notch range with the cutoff all the way down to 90hz is unusual and may be interacting with the dterm lowpass.

All these things are departures from the defaults for unknown reasons.  I would start by simply going back to all default settings and reconfigure the minimum needed to fly.  Leave the digital idle at the default or maybe even increase it to 5.0 or even 5.5 for testing.  Use default PID and filter settings.

Once it begins flying normally then you can begin tuning things like the digital idle, PIDs and filters.

quadnoob, 2017-12-17:
I have had the same problem. This is how i fixed it. Reduce start up power on the blhelli config. Then find a higher motor idle on your config.

NinjaSauce1, 2017-12-17:
Again, these are just the current settings from the latest test. I have tried 3 different firmwares with everything that you describe. If you are looking at the D term notch, you will notice that my peak motor noise runs from about 101hz to 156hz. This is well below the stock setting where the D term notch cutoff is 160hz. That isnt even in the range where I would need it. So I tuned it to encapsule my motor noise frequency. I know its not perfect, but it is certainly a better range for me. The only reason that I tuned it is because of the unexplainable spikes in Pitch D term. It did not help. In the Blackbox logs, the Pitch D goes crazy even at the absolute lowest number. The lower it goes, the less noise there is, but it becomes unflyable.
I have tried PIDs of every imaginable combination. The stock PIDs make the motors twitch, shutter, and get extremely hot. I had to lower the P terms in order to get it to fly at all.

Im using BLHeli32. I have tried reducing the startup power, the ramp up power, the motor timing, and demag compensation. I have tried throttle idle percentage from 2.0 to 7.0. My min throttle has been 1070. Are you saying to increase min throttle or idle percentage? Because its definately not the idle percentage. That was literally the first thing that I tried because it flies perfect on 3S, but goes crazy on 4S. I thought for sure it was because of the difference in idle speed. If anything, the min throttle seems too high already, and everybody has been telling me to increase it.

NinjaSauce1, 2017-12-17:
So I just tried 8 percent throttle idle. Changed the Pitch PIDs and brought the D term way down. It did not work. Still flips on the pitch axis. Its almost like the gyros reset at the exact second where the motors are switching from front to back during a flip on the pitch axis. I can tell you that this idle is way too high. Besides, if it flies on 3S, wouldnt I want to lower the idle?

etracer65, 2017-12-17:
The gyros are not "resetting" and everything looks fine with them in your logs other than lots of noise on the pitch axis (too much D seems to the main issue).  Still think you should try resetting to the defaults and reconfiguring to rule out any configuration issues.  You've said that you changed so much back and forth it's easy to forget what you changed and then end up with conflicts.  Also never use a dump from an older version to restore your settings as this can cause settings corruption.  Go to CLI and type "defaults".  For initial tests lower the roll/pitch D to about 15.  You're running BF 3.2.1 and 3.2.3 is the current - consider updating.  If this doesn't fix it I doubt it's a Betaflight issue.  Nobody else is reporting these kinds of problems and you've said it's persisted across different versions.  If the problem persists than capture another blackbox showing the problem.  Be sure to `set debug_mode = notch` and capture with at least 2KHz logging rate.

Overall it sounds more like an ESC issue.  Are you running the latest BLHeli_32?  32.3 is the latest.  Sounds like you've messed with the BLHeli settings as well.  Unless you have some extremely high-KV motors I would just go back to the BLHeli_32 defaults.  Next to suspect would be an ESC failing or a motor failing.  Both of which would be more stressed by 4S.  Double check that your motor screws are not touching the windings.  If you can't visually confirm, back the screws off a little to test.  A weak short would be worse on 4S.

NinjaSauce1, 2017-12-17:
I have reset to defaults each time I flashed the firmware. I started with version 3.1.7. Then I went to 3.2.2, then to 3.2.1. Each time, I started with the default settings and never used any old dumps. I will start over with 3.2.3. That is the only one that I have not tried because it must be only a few days old.
I am using the latest version of BLHeli 32. Each time I flash the firmware, I return all the ESC settings to default. 
I suspected a bad ESC or motor, but since this has happened on 3 different flight controllers with 4 sets of motors, I am finding that extremely hard to believe.

quadnoob, 2017-12-18:
lower start up power on the esc ?

NinjaSauce1, 2017-12-18:
I have tried 50 percent through 10 percent on BLHeli 32. I guess they call it ramp up power now.

NinjaSauce1, 2017-12-19:
Just received an email from the manufacturer. Finally. They told me to turn off the temperature protection on the ESCs and see what happens. Does this make any sense? 
So I was swapping motors to determine if there is a proplem with an individual ESC and I got a tiny bit of solder on a mosfet. Im pretty sure that its not a problem, but I am not going to plug in a battery until I am able to remove the solder. These mosfets are touching the motor pads, so its either a very poor design, or its completely fine.
And Im ready to give up on this project. It appears that there is no solution to this problem. Looks like Im gonna have a pile of banged up spare parts. This project was a disaster.

bigtuni, 2018-01-06:
any success? Found this post as my quad also goes crazy when accelerating.

NinjaSauce1, 2018-01-07:
Nope. I have a few ideas tho. After a lot of research, I think that the most likely problem is that the voltage regulator for my gyro is bad.

richard-scott, 2018-01-07:
Do you have extra capacitors on your ESC's? Most other boards that have issues like this get fixed with low esr ones... Have you tried them?

NinjaSauce1, 2018-01-07:
Yup. First thing I tried. Added a 1000uf low esr cap.

richard-scott, 2018-01-08:
@NinjaSauce1 just the one? Most people add one per ESC? and maybe also on for the VBAT.

NinjaSauce1, 2018-01-08:
Its an AIO.

richard-scott, 2018-01-08:
Matek per chance?

Even an AIO has  positive and negative for an ESC.

NinjaSauce1, 2018-01-09:
No. Its a "true" AIO. Has fc, 32 bit escs, osd, and pdb. Its the TattooF4S. Its identical to the Sunrise Model and the Lumenier Alpha. Similar to the Asgard. I didnt even know that people added individual ESC capacitors to 4 in 1 ESCs.

richard-scott, 2018-01-09:
Oh, an FC+ESC AIO, I assumed you mean just the FC... People don't generally add capacitors to a 4 in 1 ESC other than maybe across the battery.

I would try unsetting the following (back to defaults), as they are listed in a "diff" this means you've changed them:

> set dterm_lowpass = 90
> set dterm_notch_hz = 156
> set dterm_notch_cutoff = 90

Also, try increasing your ESC Protocol from DShot300, I'm sure your board would support DShot1200 being BLHeli32:

> set motor_pwm_protocol = DSHOT1200

Also your Idle Throttle value is **WAY** too low. The default i believe 4.5 so try this:

> set dshot_idle_value = 500

You need this idle value set high enough so it doesn't cause issues at zero throttle when flying.

Most likely thing is the last one (dshot_idle_value), that controls the minimum signal sent to an ESC for  low throttle movement. A roll asks one side of the quad to put the ESC's into low throttle, if this is too low a signal for the ESC, then the ESC may reset/stop working on those motors hence the unstoppable rolling.

NinjaSauce1, 2018-01-09:
That D term notch filter was tuned with blackbox. 
Obviously, I tried the defaults. Motors got really hot.
This board cannot run DShot 1200. I have tried each DShot protocol with 8k8k, 8k4k, 8k2k, 4k4k, and 4k2k. Does not fix the problem.
I have tried the throttle idle percentage from 2.0 to 8.0. It seemed like the most logical thing to look at, but it does not change anything.

Trust me, I have tried EVERYTHING.
There is no obvious solution.

NinjaSauce1, 2018-05-05:
After some experience, I believe that this problem could possible be linked between the motor timing on the ESCs and the accelerometer's maximum speed of which it can read date. Generally, the MPU6000 is limited to read a maximum of 2000 degrees per second. I believe that if your motor timing is too high, it could cause the accelerometer to be unable to read the data fast enough. Especially if your accelerometer is damaged or not great quality. This is just something that I have noticed. I hope this helps.

Morbidfpv, 2018-05-07:
I have this problem with lumenier alpha flight controller. No matter what I do nothing helps. It will be ok for a bit but then the problem will happen on its own again or when I make any pid changes etc. it seems to freak it out more. I have this aio on another quad and it never had this problem. Did you ever figure out what the prob is? I’m pretty much done and assume the flight controller is bad

closed 2018-06-13 by stale[bot]