betaflight/betaflight #4772: omnibus F7 V2 board and it takes around a minute to arm (Sensor Calib)

jamborloi, 2017-12-17:
I got a omnibus F7 V2 board and it takes around a minute to arm. In the OSD it is indicated CALIB or sensor calibration is on going. I am running BF3.3, with ICM20608 gyro running at 16k/8k and dshot1200. I tried 8K/8K still sensor calib takes very long time which means arming also takes a very long time. I got other omnibus F7 not the V2 same configuration as my V2 board but this one runs on BF3.2.1. Is it the firmware problem?

etracer65, 2017-12-17:
When running with 32K mode enabled the gyro will have a much higher bandwidth hardware lowpass filter enabled that will let through much more noise.  This will affect gyro calibration.  You may have to increase the `moron_threshold` setting to allow calibration in this configuration.  Also try disabling 32K mode and running 8/8 for the gyro/pid loops to see if calibration works normally.  You also just might have a "noisy" gyro (ICM-206xx series are more noisy than older MPU6xxx version) and the reduced hardware lpf is just letting too much through.

jamborloi, 2017-12-19:
Hi I increase the moron_threshold to 60 from 48 and it initializes quickly.
It might be the vibration it detects when I plug in the batt and put it on the ground.
Thanks

denovich, 2017-12-19:
No. etracer’s reply was as perfect of an answer as anyone can expect. He even resisted making a joke about you needing a higher moron threshold. He’s a better man than I.

thiagohmoreira, 2018-01-22:
I can say I'm having the same issue with not being able to arm the F7 v2 with 32k/8k, even setting the moron_threshold to 200 (upper limit). No luck with changing to 8k/8k also. 

Watching the Sensors tab on configurator I noticed that once I plug in the battery the noise starts. Connected via USB, gyro readings are waaay less noisy. 

So I'm assuming I need to better isolate the FC in my system to get rid of this dirty power. Right now the FC, VTx (unify pro hv) and camera (eagle 2 pro) are being powered from vbat, and I don't have any capacitors or anything. I'll try to put some caps around and see how/if it improves.

Using the MPU-6000 I can see similar behavior, but in a smaller degree. The quad is able to arm and fly (8k/8k).

Destacatyl, 2018-02-20:
Something is wrong with the DC regulator in your board. When you connect a 2S battery your board will probably calibrate without problems. I had the same problem so the seller sent me a new board. The new board calibrates and arms without problems.

I also tried to find a place to cut the VBAT to seperate the power for the FC and the measurement for OSD but I was not successful so far.

thiagohmoreira, 2018-04-08:
So, just as a quick update, I've changed the setup to be feed the board from the ESCs 5V rail. It works great as long as I don't connect VBAT to it. In the moment there is anything connected to it (even using a clean power supply), the ICM gyro readings get crazy, with the MPU it stays pretty reasonable (and flies well).

I've added capacitors in many places, also did what @Destacatyl mentioned, with capacitors on the rails, and got same results.

It might be that my board has some issue, but at this point is too late to return it anyway ;/ Too sad...

closed 2018-05-15 by stale[bot]