betaflight/betaflight #4779: OMNIBUSF7V2 won't boot in certain configurations [bisected]

andrejpodzimek, 2017-12-19:
`git bisect` says that de5b0d53ffff22037a1d96409aa88ffcf51423ce is the culprit. The board always boots just fine in default configuration. Once the following settings are applied, it reboots normally before the culprit commit, but crashes after that commit:

```
name Max

resource SERIAL_TX 3 NONE
resource SERIAL_TX 11 E06
resource SERIAL_RX 3 NONE
resource I2C_SCL 2 B10
resource I2C_SDA 2 B11
resource SPI_SCK 4 NONE
resource SPI_MISO 4 NONE
resource SPI_MOSI 4 NONE

feature SOFTSERIAL
feature GPS
feature TELEMETRY
feature RSSI_ADC
feature LED_STRIP
feature ESC_SENSOR
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

serial 0 16384 115200 57600 0 115200
serial 5 2 115200 57600 0 115200
serial 30 32 115200 57600 0 115200

set gyro_sync_denom = 3
set gyro_use_32khz = ON
set mag_hardware = HMC5883
set max_check = 1950
set blackbox_device = NONE
set min_throttle = 1030
set dshot_idle_value = 200
set motor_pwm_protocol = DSHOT1200
set align_board_roll = 180
set align_board_yaw = 270
set current_meter = ESC
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set gps_auto_baud = ON
set pid_process_denom = 1
set osd_stat_max_dist = ON
set osd_stat_max_alt = ON
set vcd_video_system = 1

profile 0
rateprofile 0
```

Crash symptoms: The device si no longer recoginzed on USB; I don't get the `/dev/ttyACM0` device and `dmesg` says this:

```
[170858.294456] usb 3-1.4: new full-speed USB device number 120 using xhci_hcd
[170863.347778] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[170868.681151] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[170868.887720] usb 3-1.4: device not accepting address 120, error -62
[170868.887774] usb 3-1-port4: unable to enumerate USB device
```

The above configuration works and boots perfectly fine before de5b0d53ffff22037a1d96409aa88ffcf51423ce.

martinbudden, 2017-12-19:
@andrejpodzimek , thanks for testing. I think PR https://github.com/betaflight/betaflight/pull/4780 should fix this.

andrejpodzimek, 2017-12-20:
Thanks, that works. I've just booted from today's master without issues.

closed 2017-12-20 by andrejpodzimek