betaflight/betaflight #4783: Strange stage2 failsafe behavior with Betaflight 3.2.3

jstremmler, 2017-12-19:
I own two copters using the EACHINE modified Omnibus F4 Pro FC. Both of them show a strange failsafe behavior in failsafe stage2.  

Stage1 works perfectly and stage 2 as well, when I specify "drop "in stage 2. 

But when I specify in stage2 "landing" then only motors 1, 3 and 4 go to the specified rpms and then stop after the specified time.

But motor2 stops immediately when the copter switches to failsafe stage2. I observe the same behavior at both copters.
( I know that we shouldn't use landing in failsafe stage2 and dropping the copter immediately after a permanent loss of signal is more safe, but I just wanted to test "landing" in failsafe stage 2 on the bench when switching my TX off.)

mikeller, 2017-12-19:
@jstremmler: GitHub is for questions and discussion around the **development** of the firmware and tools. For questions around the configuration and use of Betaflight, please visit [RCGroups](https://www.rcgroups.com/forums/search.php?query=betaflight%20cleanflight%20eachine%20manufacturer%20support).

closed 2017-12-19 by mikeller

jstremmler, 2017-12-20:
I was informed in RCGROUPS that this is not a problem of my FC's and my RX's. Neither is it a configuration problem of Betaflight! So, if motors 1,3, and 4 on my quads do their job and motor 2 does stopp immediately when the FC is going to stage 2 of failsafe in Betaflight then this is a Betaflight bug. And it is a serious one, because this behavior in case of a failsafe can make the situation more unsafe than immediately dropping ones copter.

DanNixon, 2017-12-20:
Post the followibg information and I'll take a look:
- Firmware Git revision
- `diff`
- ESC model
- ESC firmware

jstremmler, 2017-12-20:
Hi Dan,

1. Firmware Version: Betaflight / OMNIBUSF4SD (OBSD) 3.2.3 Dec 11 2017 / 07:56:44 (cb962eda1) MSP API: 1.36

2. Diff:
# name
name Real1S

# resources

# mixer

# servo

# servo mix

# feature
feature LED_STRIP
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper
beeper -ON_USB

# map

# serial
serial 0 1 115200 57600 0 115200
serial 5 64 115200 57600 0 115200

# led
led 0 0,15:W:LI:0
led 1 1,15:N:AI:0
led 2 2,15:S:AI:0
led 3 3,15:E:LI:0

# color

# mode_color
mode_color 6 0 2
mode_color 6 1 6

# aux
aux 0 0 0 1800 2050
aux 1 1 1 975 1200
aux 2 2 1 1425 1550
aux 3 3 1 975 2050
aux 4 13 3 1800 2050
aux 5 19 4 1800 2050
aux 6 26 4 950 1200
aux 7 27 3 950 1200
aux 8 28 1 1625 2050

# adjrange

# rxrange

# vtx

# rxfail
rxfail 3 h

# master
set gyro_notch1_hz = 0
set gyro_notch1_cutoff = 0
set gyro_notch2_hz = 0
set gyro_notch2_cutoff = 0
set serialrx_provider = SBUS
set dshot_idle_value = 500
set motor_pwm_protocol = DSHOT1200
set failsafe_delay = 10
set failsafe_off_delay = 50
set failsafe_throttle = 1320
set failsafe_kill_switch = ON
set bat_capacity = 1400
set vbat_min_cell_voltage = 32
set vbat_warning_cell_voltage = 34
set vbat_scale = 112
set ibata_scale = 317
set beeper_dshot_beacon_tone = 1
set deadband = 2
set yaw_deadband = 2
set pid_process_denom = 1
set osd_vbat_pos = 2072
set osd_rssi_pos = 2049
set osd_tim_1_pos = 54
set osd_tim_2_pos = 2390
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 377
set osd_crosshairs = 200
set osd_current_pos = 2369
set osd_mah_drawn_pos = 2338
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_avg_cell_voltage_pos = 76
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 258
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_min_rssi = OFF
set vcd_video_system = 1

# profile
profile 0

set dterm_notch_hz = 180
set dterm_notch_cutoff = 110
set anti_gravity_gain = 3500
set setpoint_relax_ratio = 30
set dterm_setpoint_weight = 200
set p_pitch = 72
set d_pitch = 30
set p_roll = 64
set i_roll = 48
set p_level = 20
set i_level = 20
set d_level = 100

# rateprofile
rateprofile 0

set rc_rate = 75
set rc_rate_yaw = 75
set rc_expo = 27
set rc_expo_yaw = 27
set thr_mid = 38
set thr_expo = 30
set roll_srate = 86
set pitch_srate = 86
set yaw_srate = 82

3. ESC Model:

Special Edition High Quality Racerstar Tattoo 30A BLheli_32 ST 2-4S RaceSpec ESC DShot1200
https://www.aliexpress.com/item/Special-Edition-High-Quality-Racerstar-Tattoo-30A-BLheli-32-ST-2-4S-RaceSpec-ESC-DShot1200-Mluti/32824709259.html

4. ESC-Firmware:

BLHeli-32 Version 32.3

Thank you very much for your assistance! Pls note in the diff that I catually hvae specified "drop" for failsafe and the failure only occurs with failsafe stage 2 set to "land". And please note that I set the beeper_dshot_beacon_tone to 1 and use my ESCs and motors as well as a beeper (orsk only with version 32.3 of BLHeli)

Best wishes

Jürgen Stremmler
jstremmler@aol.com

Verschickt: Mi, 20. Dez 2017 11:01