betaflight/betaflight #4784: Wrong pin configuration on UART3 for OMNIBUSF7

andrejpodzimek, 2017-12-19:
[Betaflight](https://github.com/betaflight/betaflight/blob/b924ac0c9ceb772d4f544834a8700df126506f5c/src/main/target/OMNIBUSF7/target.h#L88-89) and [iNav](https://github.com/iNavFlight/inav/blob/912d131556302cf6ea67d59308e43dfe65ca8026/src/main/target/OMNIBUSF7/target.h#L91) disagree on the ordering of the two pins. Only one of them can be correct. Based on [this observation](https://github.com/betaflight/betaflight/issues/4531#issuecomment-352754944), I believe that iNav may be actually right. I2C devices on `UART3` seem to work on iNav, but not on Betaflight.

jflyper, 2017-12-19:
Good finding.

However, it's the other way around. BF is correct and iNav is incorrect (unless reverse option is configured in the driver 😉 ).

Datasheet:
[image]

BF: https://github.com/betaflight/betaflight/blob/master/src/main/target/OMNIBUSF7/target.h#L88-L89
```
 88   #define UART3_RX_PIN            PB11
 89   #define UART3_TX_PIN            PB10
```

iNav: https://github.com/iNavFlight/inav/blob/master/src/main/target/OMNIBUSF7/target.h#L91-L92
```
 91   #define UART3_RX_PIN PB10
 92   #define UART3_TX_PIN PB11
```

I will submit a PR to iNav for a fix.

closed 2017-12-19 by jflyper

jflyper, 2017-12-19:
Btw, on Omnibus F7, I2C2 on PB10(TX3/SCL) and PB11(RX3/SDA) is the proven configuration.

[image]

```
# resource list
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: BARO_CS
...
B10: I2C_SCL 2
B11: I2C_SDA 2
...
```

andrejpodzimek, 2017-12-20:
Many thanks for looking into this, @jflyper! Indeed, iNav has exactly the same I2C configuration on those pins, only the `UART3` configuration differed.

In the mean time, [**I got comass data on iNav**](https://github.com/betaflight/betaflight/issues/4531#issuecomment-352956851). But I still have no clue why it doesn't work for me in Betaflight. It would be awesome if someone more familiar with the hardware could take a glance at the two drivers side-by-side and try to spot anything worth testing in Betaflight. With @McGiverGim's hack, the Radiolink SE100 compass works out-of-the-box in iNav.

(I'm not sure how precise it is and what the absence of the gain data means, but at least some data is there to start with.)