betaflight/betaflight #4790: Receiver does not leave the binding mode

Simon74SH, 2017-12-20:
Hello everybody
I have problems with my DSMX receiver I want to bind my transmitter to the receiver (Spektrum DX9) in the console I enter the following
set spectrum_sat_bind = 9
set spectrum_sat_bind_autoreset = 0
to save
I'm in bind mode, so far so good.
Transmitter successfully connected LED lamp lights up permanently.
Now I give the command
set spectrum_sat_bind_autorst = 1
Save
to exit bind mode press Enter reboot the system LED is off.
Under "dump" is also binding mode "1"
Now I pull the USB cable off and put a lipo on it or again the USB cable, I'm still in bind mode.
I just can not turn off the bind mode.
Please help me

LG Simon

Simon74SH, 2017-12-20:
I went to this manual

https://www.propwashed.com/spektrum-receivers-with-betaflight/

richard-scott, 2017-12-20:
Don't use this part of your setup:

```
set spectrum_sat_bind_autoreset = 0
```

It will work fine without that, and will autoreset out of bind mode back to normal mode.
The Propwashed instructions are not great as they force the RX into bind mode all the time, so you need to rebind after the first power cycle which isn't what you want.

Simon74SH, 2017-12-20:
Hello now with the following command the binding mode can leave
"set spectrum_sat_bind = 0"
All data in betaflight are displayed correctly.
Now I connect a lipo but the engines do not start even though "Throttle" is displayed in betaflight and works.
Am an absolute novice what this system is concerned therefore I apologize for stupid question that may be clear to others

AndersHoglund, 2017-12-20:
Why following (potentially incorrect/incomplete) instructions on some foreign web-sites, when there are instructions here?: https://github.com/betaflight/betaflight/wiki/Spektrum-Rx-setup

Why would it be strange and an issue that you are stuck in bind mode when you actually turn OFF automatic reset from bind mode? Then you will have to do that manually your self. Or keep auto reset ON.

As for arming, that is also covered in the WiKi: https://github.com/betaflight/betaflight/wiki/Arming-Sequence-&-Safety
As advised there, use the CLI `status` command to see what is preventing arming.

closed 2017-12-20 by AndersHoglund