betaflight/betaflight #4796: 3D on a switch failsafe bug 3.3

thongraegu, 2017-12-20:
Build 430, failsafe stage 1 puts throttle to "low" position and this means motors start spinning full throttle to the opposite direction in a 3D setup.

brycedjohnson, 2017-12-20:
Think you could toss a diff all from the cli as well?  Wondering if it is dshot or multishot and just checking how you have it set up.

thongraegu, 2017-12-20:
Its dshot600, motors set bidirectional, 3D enabled in betaflight, 3D on a switch enabled.

# version
# Betaflight / BETAFLIGHTF4 (BFF4) 3.3.0 Dec 18 2017 / 02:38:29 (5bfd5107a) MSP API: 1.37

# reset configuration to default settings
defaults nosave

# name

# resources

# mixer

# servo

# servo mix

# feature
feature 3D
feature AIRMODE
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper
beeper -ON_USB

# map
map TAER1234

# serial
serial 1 64 115200 57600 0 115200
serial 5 0 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1700 2100 0
aux 1 13 1 1300 1700 0
aux 2 35 1 1700 2100 0
aux 3 38 2 1300 2100 0

# adjrange

# rxrange

# vtx

# rxfail
rxfail 3 h

# master
set gyro_notch1_hz = 0
set gyro_notch2_hz = 0
set acc_hardware = NONE
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1020
set max_check = 1950
set rssi_channel = 8
set serialrx_provider = CRSF
set dshot_idle_value = 400
set motor_pwm_protocol = DSHOT600
set failsafe_delay = 6
set vbat_min_cell_voltage = 31
set vbat_warning_cell_voltage = 31
set current_meter = ADC
set battery_meter = ADC
set ibata_scale = 236
set beeper_dshot_beacon_tone = 4
set yaw_motors_reversed = ON
set 3d_deadband_low = 1450
set 3d_deadband_high = 1550
set 3d_neutral = 1500
set pid_process_denom = 1
set osd_warnings = 63
set osd_vbat_pos = 2402
set osd_rssi_pos = 2093
set osd_tim_2_pos = 2421
set osd_mah_drawn_pos = 2411
set osd_warnings_pos = 2312
set osd_stat_max_spd = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set vtx_band = 0
set vtx_channel = 2
set vtx_power = 5
set vtx_freq = -17
set vcd_video_system = 2

# profile
profile 0

set dterm_lowpass_type = PT1
set anti_gravity_gain = 6000
set dterm_setpoint_weight = 150
set p_yaw = 90

# profile
profile 1

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

set rc_expo = 10
set rc_expo_yaw = 10
set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 80

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# restore original rateprofile selection
rateprofile 0

# save configuration
save
#

thongraegu, 2017-12-20:
^^ On that one I have now setup throttle failsafe as "hold" instead of default "low"

Teracis, 2018-02-13:
FYI for failsafe in 3D mode, I've always set the arm channel to hold at the disarmed state, thereby disarming the quad immediately and preventing any possible 3D throttle issues. Just a thought/workaround whilst this is being worked on. Maybe that could be incorporated as a default instead of throttle going low? Arm goes low?

mikeller, 2018-02-13:
I think the problem is that when 3D comes into play, there are multiple variables to control: Arm switch, 3D switch, throttle (itself dependent on 3D switch). This will make it hard to establish meaningful defaults that work for all configurations. Disarming is one potential option, but it might not be desirable in all situations, since it can mean complete loss of control even for short transient loss of signal situations.

Teracis, 2018-02-13:
Fair call Mike, I wasn't paying attention to the fact that people use these rigs for things other than LOS acro and FPV where a disarm state due to lack of signal is desirable. Maybe as part of the default for 3D on a switch the 3D part could be disabled during failsafe? That or notes are added that put the onus back on the user to set up and test their failsafe correctly before flying, there's too many different modes and uses to cater for them all in the defaults as you say.

DanNixon, 2018-02-13:
> I've always set the arm channel to hold at the disarmed state, thereby disarming the quad immediately

That does not actually disarm the quad. Assuming your stage 2 procedure is DROP you are probably just seeing the transition from stage 1 to 2.

To verify this try setting the maximum stage 1 timeout possible and see if you get the same effect.

Teracis, 2018-02-13:
Really?
I'll have to give that a test, I never used to even bother with it and always set stage 1 and 2 to be immediate however started putting the stage 1 to disarm as part of that. I won't do it tonight but I'll get that tested and report back. If setting the channel for arm to outside of the arm range won't disarm the quad then what does it do?

DanNixon, 2018-02-13:
I'm sure I remember that being the case when looking at #3108. Although, that may not be the case anymore. If you do end up testing it, it would be good to hear back.

mikeller, 2018-02-13:
@Teracis:

> I wasn't paying attention to the fact that people use these rigs for things other than LOS acro and FPV where a disarm state due to lack of signal is desirable.

I don't think we are talking about the same here. You are postulating to configure the flight controller to disarm every time that stage 1 is entered due to signal loss, no matter how short the signal loss is. This may be workable for RX systems that have some built in delay before signalling connection loss to the flight controller, but for others that report every single RF frame loss, it will result in mid-flight disarms just from things like somebody powering up a TX in the vicinity.

Teracis, 2018-02-13:
@mikeller Indeed I hadn't realised that connections aren't as robust as the gear I'm using. I've never had a loss of signal (that I've known about) so it hasn't been an issue.
I didn't realise at all that would be the result of single frame drops e.t.c. so obviously disarm won't work. I'll have to test my rig with another tx powering up in the vicinity to see if it causes any issues.
In that case then maybe for stuff like 3D mode on a switch there's a way to disable that mode (or all modes for that matter) whilst doing things like stage 1 failsafe, and setting throttle to the default state e.t.c. Other than that I'm out of ideas unless it's just mentioned that users have to configure this to their safest mode or risk issues.

mikeller, 2018-02-13:
@Teracis:

> Indeed I hadn't realised that connections aren't as robust as the gear I'm using. I've never had a loss of signal (that I've known about) so it hasn't been an issue.

Yeah, I think you make a bad test case for this then - the stage 1 / stage 2 failsafe system was designed exactly for this case - to enable the craft to 'push through' a link interruption of at maximum stage 1 delay length, before considering the link permanently lost and reacting accordingly. If you never get a transient link loss then stage 1 is indeed not needed for you.

> In that case then maybe for stuff like 3D mode on a switch there's a way to disable that mode (or all modes for that matter) whilst doing things like stage 1 failsafe, and setting throttle to the [...]

- doing mode changes on failsafe when in 3D is potentially problematic, since we can't control what state the craft is in when the failsafe happens. What if 3D is disabled by a falisafe event while the pilot is flying inverted?
- the pilot can always configure failsafe to do any mode switches they want, as long as the mode in question is on a switch, by setting the failsafe value for the channel in question. So I don't think any mode changes should be done by default.

Teracis, 2018-02-13:
Yep entirely makes sense, I was just offering my solution to the problem, entirely forgetting that it's unreasonable to accept a disarm for a signal loss. The same thing goes for 3D disabled e.t.c. I'm not flying with it on a switch so the thought never crossed my mind that it could become disabled at the wrong time.

So does this line of thought mean that the issue here is fixed because it was setup incorrectly by the original pilot? Or have I entirely missed the actual issue here? Maybe a confirmation is required on the failsafe tab before the board will arm for the first time or something? Though that would likely cause more issues than it would solve!

mikeller, 2018-02-13:
@Teracis: There is potentially room for improvement over what is currently done by making the throttle value default to something that is idle (but this might make the setup overly complicated, and not be worth it for this reason). Apart from that I am not sure how much the current situation can be improved.

haslinghuis, 2022-05-08:
Closing as inactive

closed 2022-05-08 by haslinghuis