betaflight/betaflight #4797: Softserial ports do not show with HGLRC-XJB-F428  and OMNIBUSF4 firmware 3.2.0 and above

jrwrodgers, 2017-12-20:
Board: HGLRC-XJB-F428
Firmware : OMNIBUSF4

When enabling the softserial feature on versions 3.2.0 and above the softserial ports neither appear in the ports tab nor in the "serial" at the CLI . The ports 30 and 31 are not showing even though softserial is enabled.

Version 3.2.0 and 3.2.3
# feature SOFTSERIAL
Enabled SOFTSERIAL
save

# serial
serial 20 1 115200 57600 0 115200
serial 0 0 115200 57600 0 115200
serial 2 0 115200 57600 0 115200
serial 5 0 115200 57600 0 115200

Version 3.1.7 
# feature SOFTSERIAL
Enabled SOFTSERIAL
save

# serial
serial 20 1 115200 57600 0 115200
serial 0 0 115200 57600 0 115200
serial 2 0 115200 57600 0 115200
serial 5 0 115200 57600 0 115200
serial 30 0 115200 57600 0 115200
serial 31 0 115200 57600 0 115200

mikeller, 2017-12-21:
Have you assigned resources for the softserial ports? What does your `resource list` show?

jrwrodgers, 2017-12-21:
Allocated serial tx 11 to B14 , the original PPM pad

'># resource
resource BEEPER 1 B04
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 A02
resource MOTOR 5 A01
resource MOTOR 6 A08
resource PWM 2 B15
resource PWM 3 C06
resource PWM 4 C07
resource PWM 5 C08
resource PWM 6 C09
resource LED_STRIP 1 A01
resource SERIAL_TX 1 A09
resource SERIAL_TX 3 B10
resource SERIAL_TX 6 C06
resource SERIAL_TX 11 B14
resource SERIAL_RX 1 A10
resource SERIAL_RX 3 B11
resource SERIAL_RX 6 C07
resource INVERTER 1 C00
resource LED 1 B05
resource SPI_SCK 1 A05
resource SPI_SCK 3 C10
resource SPI_MISO 1 A06
resource SPI_MISO 3 C11
resource SPI_MOSI 1 A07
resource SPI_MOSI 3 C12
resource ADC_BATT 1 C02
resource ADC_RSSI 1 A00
resource ADC_CURR 1 C01'

'># serial
serial 20 1 115200 57600 0 115200
serial 0 0 115200 57600 0 115200
serial 2 0 115200 57600 0 115200
serial 5 0 115200 57600 0 115200'

jflyper, 2017-12-21:
@jrwrodgers 

Use "`resource list`" to see the actual mapping.

I tested the configuration on the original Omnibus F4 AIO, freshly flashed with 3.2.0, gave `resource` and `feature` commands, and the serial 30 persists.

```
# diff

# version
# Betaflight / OMNIBUSF4 (OBF4) 3.2.0 Sep 28 2017 / 16:38:25 (ceb6f70) MSP API: 1.36

# resources
resource SERIAL_TX 11 B14

# feature
feature SOFTSERIAL
```
Then
```
# serial
serial 20 1 115200 57600 0 115200
serial 0 0 115200 57600 0 115200
serial 2 0 115200 57600 0 115200
serial 5 0 115200 57600 0 115200
serial 30 0 115200 57600 0 115200
```

Please start with fresh flash, then immediately enter `resource` and `feature`.
If it does not work, please consult the board manufacturer or some other places on the net.

closed 2017-12-21 by jflyper