betaflight/betaflight #4798: I am using the furious fpv radiance but motor 1 wont spin up and my 3 other motors work can anyone help me?

DarkmatterFPV, 2017-12-21:
# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack: https://slack.betaflight.tech or other user support forums & groups (e.g. facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1.) Describe your problem;
2.) Include ways to reproduce the problem;
3.) Provide information about your flightcontroller and other components including how they are connected/wired;
4.) Add the used configuration and firmware version;
5.) Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/); and
6.) Remove this Text :).

jflyper, 2017-12-21:
# Your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

# Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack: https://slack.betaflight.tech or other user support forums & groups (e.g. facebook).

closed 2017-12-21 by jflyper

richard-scott, 2017-12-21:
The original board doesn't look to DShot from what I can see. Try a different ESC protocol if you are trying to use DShot, maybe Multishot.