betaflight/betaflight #4799: [Feature request] disable osd entry stick command

wx4cb, 2017-12-21:
Using a betaflight f4 board,  latest version, x8r on sbus in a fixed wing 

Im badically just using the bff4 board for the osd and gps right now, but i have had an issue on a couple of occasipns where if the stick positions are just right it will enter the osd mwnu mid flight.

Im assuming this is because im technically not arming the board for flight.

It qould be nice to be able to disable the throttle mid yaw left pitch up command to enter the osd menu.

Any thoughts on this or a way to get it to igbore them?

jflyper, 2017-12-21:
@wx4cb 
If arming doesn't harm your application, then arming would be your "OSD disable switch".

wx4cb, 2017-12-21:
No arming doesnt hurt as i have a throttle cut switch, i guess i can just put that on an extra channel output.

wx4cb, 2017-12-21:
i'm also assuming that because it's not being armed, the home position (which the direction/distance osd item relies on) isn't being recorded

jflyper, 2017-12-21:
I think so, too. You are not getting flight statistics neither (if you are interested).

closed 2017-12-21 by wx4cb