betaflight/betaflight #4804: [FEATURE REQUEST] Less decimal places on osd for GPS

wx4cb, 2017-12-21:
Currently we are using 7 decimal places for the LAT/LONG on the OSD from the GPS.

I propose reducing this to (at most) 5 decimal places which leaves extra room on the OSD especially if you are using the larger fonts.

Maybe this could be configurable via a CLI variable?

DanNixon, 2017-12-21:
I'll take a look once #3417 is done, this adds a nice way to have the precision configurable.

wx4cb, 2017-12-21:
that's not a prob. I tried it myself but it appears that the tfp_format and tfpsprintf functions in comomn/printf.c that it uses to merge them all into buff isn't length modifier compatible. IE: you can't do %05d for example. if that could be modified to accept those, that would enable all sorts of goodies :D - take a look also at the way INAV does it as theirs uses %07d to have 7 decimal places (working on that one to 5 too :D

McGiverGim, 2017-12-21:
The number of decimals that the OSD shows is not a fixed length. It depends of the position gps you are.
Is not the same the "4.xxxxx..." or the "120.xxxx...". The number of decimals shown in each case is different.
Having a fixed number of decimals can occupy a different size in your OSD depending on your position. So I think it's better to limit the length total and not the decimals total. And you can have negative numbers too so "-120.xxxx..." will occupy more.
An easy solution is to put the elements into the right, letting part outside the screen 😋

wx4cb, 2017-12-21:
i realise that looking more into the code. but my point was there's not really much point in having more than 4 numbers (5 at the most) after the decimal point. 

Having them just "drop off" the side of the screen doesn't work if they're on the left side of the OSD like mine. But that being said, you can't place them on the right half outside of the screen, the configurator won't let you.

[image]

DanNixon, 2017-12-21:
> The number of decimals that the OSD shows is not a fixed length

I find that a very odd decision in the first place.

I assume this was done to ensure the element always occupies the same amount of screen space, f so the correct way would be to space pad the integer part.

As it is now the precision of your GPS readout is dependent on where in the world you are flying.

McGiverGim, 2017-12-21:
Four decimal places is too low precision. You can have 11m or precision only, that can be a world depending on the place. Five decimals is 1.1m, that is more or less convenient. Six decimals is 0.11m, a good value.
The configurator let me some time ago put the elements outside the screen, but I've not tried it in the latest version.

McGiverGim, 2017-12-21:
>>>  I assume this was done to ensure the element always occupies the same amount of screen space, f so the correct way would be to space pad the integer part.

I think that this was the idea, but is not true. When the number is negative, it occupies a position more than when it isn't. For me, padding the integer and limiting the decimals to 5-6 is right, but will be:
symbol + sign + integer (3 pos) + decimal (5-6 pos) = 10-11 length

DanNixon, 2017-12-21:
Thinking about it, maybe it should be the integer part that is optionally removed. After all the part of the coordinates you will probably care most about are the last few decimal places.

wx4cb, 2017-12-21:
@DanNixon i personally like having at least a semi-complete lat/long int he feed that way if im flying long range or outside of my "home state" so to speak, when i look at dvr footage i know exactly where it is. and as regards the decimal precision, 5 decimals would be more than accurate for finding a lost model (3 feet). but even at 11 meters that's still only a 30 foot area you're looking at.