betaflight/betaflight #4810: Beeper on startup even when disabled

stefanheule, 2017-12-21:
# Problem
When I plug in the battery, the external beeper goes off, even though I turned it off in betaflight.

# Reproduce
Connect external beeper, run `beeper -all; save` in CLI.  Then plug in a battery, and you'll hear the beeper go off.  I expect it to not, since I turned it off.

# Setup
Matek F405-OSD flight controller, with a Matek external beeper.

# Additional comments
My external beeper is pretty loud (which I want, because I want to use it when I need to find my quad), so I really don't want it to go off ever unless I flip a switch on my radio (or maybe if there's no signal), and certainly not when I just plug in a battery.  I thought `SYSTEM_INIT` would control that initial beep, but even with it disabled (and everything else too), I still get a beep.

# Diff
Probably not relevant, but here we go:

```
# version
# Betaflight / MATEKF405 (MKF4) 3.2.3 Dec 11 2017 / 07:51:34 (cb962eda1) MSP API: 1.36

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -MULTI_BEEPS
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE

# map
map TAER1234

# aux
aux 0 0 2 1700 2100
aux 1 1 0 1700 2100
aux 2 2 0 1300 1700
aux 3 13 3 1300 2100
aux 4 28 0 900 1300
aux 5 35 4 1700 2100

# master
set motor_pwm_protocol = DSHOT300
set beeper_dshot_beacon_tone = 5
set pid_process_denom = 1
```

etracer65, 2017-12-21:
What kind of beep?  A single short beep or some pattern?

A single short beep may be unavoidable depending on whether the board designer used and active high or active low signal for the beeper driving transistor.  When the board is powered before the program starts running the pins are set to a default that may trigger the beeper.  Once the firmware starts running it resets the pins to the proper state and the beeper stops.

I looked at the target for the Matek F405 and it is using an inverted beeper so this could be what's going on.

stefanheule, 2017-12-21:
Two short beeps.  I see, thanks for looking into this and explaining, sounds like this is not betaflights problem.  Anything I can do to fix this, or at least work around the problem?  Any hints would be appreciated (though I'll close this since it sounds like it's not a betaflight problem).

closed 2017-12-21 by stefanheule

milos-silni, 2018-05-20:
I have the same issue.. found any solution yet?

stefanheule, 2018-05-20:
I have not, at least with the Matek beeper I don't think there is an easy fix.  I just use a different beeper now.