betaflight/betaflight #4818: [FEATURE REQUEST] Optionally download SD card logs over MSP

etracer65, 2017-12-22:
Just a long shot feature request.  Sometimes the orientation of flight controllers with on-board SD cards make it extremely difficult to access the card - particularly on tight builds.  It would be handy to have the ability to download the log through the configurator like with on-board flash storage.  Yes it would be slower, but being able to get a short log off for tuning would be useful without having to struggle with getting the SD card out and back in.  Besides, the blackbox download through MSP is moderately quick now and most logs can be downloaded in less than 60 seconds.

mikeller, 2017-12-22:
There already is an issue for this: #2850

The idea is to add a block device mode to enable mounting of the SD card as a block device - this will take care of the speed issue as well.

Currently, the download speed over MSP puts downloading a full 8MB flash into the 'minutes' range - extrapolating this to a (modest) 8GB SD card puts this at 'days', which does not seem to be realistic or useful.

closed 2017-12-22 by mikeller

ledvinap, 2017-12-23:
@mikeller : you don't usually fill 8G card with logs..

mikeller, 2017-12-23:
@ledvinap: Depends on how you use them. I have cards in my quads with card readers, and have logging always on, just in case there is an event that requires log analysis. So when I pull the card, there often are 100 or so logs, with 20 MB or so per log. Not quite 8 GB, but more than I would want to download at MSP speeds just to get access to the last log that was recorded.

My point wasn't so much about 8 GB or whatever size, but about the fact that SD card storage has become so cheap that it is an attractive option to just provide enough storage to log everything, and only pull the logs when something needs analysis.