betaflight/betaflight #4838: AIRBOTF4SD doesn't detect baro

etracer65, 2017-12-24:
I'm not a big baro user, but I noticed that the on-board baro on a Flip32 F4/AirbotF4 Dual Gyro (target AIRBOTF4SD - derivative of REVO) wasn't being detected.  The baro appears to be a BMP280 based on it's package and numbering.  The definitions appear to be in the target so I'm not sure what's going on.  Verified in the current 3.3 master as well as the 3.2 maintenance branch.  Not sure if the baro is connected via I2C or SPI as it supports both.  I don't see any I2C resources defined.

Product link: http://www.readytoflyquads.com/flip32-f4-dual-gyro-edition-mpu-icm20608mpu6000

```
# resource
resource BEEPER 1 B04
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 A02
resource MOTOR 5 A01
resource MOTOR 6 A08
resource PPM 1 B14
resource PWM 1 B14
resource PWM 2 B15
resource PWM 3 C06
resource PWM 4 C07
resource PWM 5 C08
resource PWM 6 C09
resource LED_STRIP 1 B06
resource SERIAL_TX 1 A09
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 6 C06
resource SERIAL_RX 1 A10
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 6 C07
resource INVERTER 6 D02
resource LED 1 B05
resource SPI_SCK 1 A05
resource SPI_SCK 3 C10
resource SPI_MISO 1 A06
resource SPI_MISO 3 C11
resource SPI_MOSI 1 A07
resource SPI_MOSI 3 C12
resource ESCSERIAL 1 B14
resource ADC_BATT 1 C02
resource ADC_RSSI 1 A00
resource ADC_CURR 1 C01

# resource list
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: MOTOR 4
A03: MOTOR 3
A04: MPU_CS
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: FREE
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: SDCARD_CS
B04: BEEPER
B05: LED 1
B06: FREE
B07: FREE
B08: FREE
B09: FREE
B10: FREE
B11: FREE
B12: FREE
B13: SPI_PREINIT
B14: FREE
B15: FREE
C00: SDCARD_DETECT
C01: FREE
C02: FREE
C03: FREE
C04: MPU_EXTI
C05: FREE
C06: FREE
C07: SERIAL_RX 6
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE
C14: FREE
C15: FREE
D02: INVERTER 6

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

[image]

jflyper, 2017-12-24:
@etracer65 I'll look into it. Will get back to you.

jflyper, 2017-12-24:
@etracer65 BMP280 is on SPI1, and it wasn't configured at all 😩 

Try this
[betaflight_3.3.0_AIRBOTF4SD-with_BMP280-0.hex.zip](https://github.com/betaflight/betaflight/files/1584942/betaflight_3.3.0_AIRBOTF4SD-with_BMP280-0.hex.zip)

etracer65, 2017-12-24:
@jflyper That fixed it!  Baro is detected and shows valid readings in the sensors tab.  Thanks for your help.

jflyper, 2017-12-24:
@etracer65 Thanks for testing! I'll submit PRs for 3.3 and 3.2.x.

PR #4839 (2017-12-24): AIRBOT{F4,F4SD} Add missing BMP280 on SPI support
Add missing BMP280 on SPI support for AIRBOTF4 and AIRBOTF4SD.

Fixes: #4838

Also for 3.2.x maint, may be?
closed 2017-12-24 by mikeller