betaflight/betaflight #4840: RunCam Split no longer works

andrejpodzimek, 2017-12-24:
RunCam Split worked fine on my build on 3.1.7 and in `master` from ~August. It no longer works with the current `master`; the wiring and configuration is identical, but changes in the RunCam power channel no longer switch the RunCam on/off. Starting and stopping the RunCam using its buttons works as usual. Details about the build and configuration:

```
# diff

# version
# Betaflight / ANYFCF7 (ANY7) 3.3.0 Dec 24 2017 / 18:51:35 (818bda4f7) MSP API: 1.37

# name
name Raiju II

# resources
resource BEEPER 1 B09
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource LED_STRIP 1 E06
resource SERIAL_TX 2 NONE
resource SERIAL_TX 3 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_RX 2 NONE
resource SERIAL_RX 3 NONE
resource SERIAL_RX 4 NONE
resource SERIAL_RX 7 NONE

# mixer
mixer HEX6X

# servo

# servo mix

# feature
feature GPS
feature TELEMETRY
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper

# map

# serial
serial 0 1 115200 57600 0 115200
serial 3 32 115200 57600 0 115200
serial 4 64 115200 57600 0 115200
serial 5 2 115200 57600 0 115200
serial 7 16384 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1700 2100 0
aux 1 1 1 900 1625 0
aux 2 2 1 1625 1875 0
aux 3 3 1 900 1375 0
aux 4 11 1 900 1125 0
aux 5 13 3 1200 2100 0
aux 6 28 4 1700 2100 0
aux 7 33 5 1700 2100 0

# adjrange
adjrange 0 0 2 900 2100 12 2

# rxrange

# vtx

# rxfail
rxfail 5 s 1000
rxfail 6 s 1000
rxfail 7 s 2000

# master
set align_mag = CW0FLIP
set mag_declination = 2180
set magzero_x = -634
set magzero_y = 496
set magzero_z = -388
set max_check = 1950
set blackbox_device = NONE
set min_throttle = 1030
set dshot_idle_value = 200
set motor_pwm_protocol = DSHOT600
set current_meter = ADC
set battery_meter = ADC
set ibata_scale = 366
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set gps_auto_baud = ON
set pid_process_denom = 1
set frsky_unit = METRIC

# profile
profile 0

set dterm_lowpass_type = PT1
set vbat_pid_gain = ON
set anti_gravity_gain = 2000
set yaw_lowpass = 90
set d_pitch = 31
set d_roll = 27
set p_alt = 15
set i_alt = 25
set d_level = 80

# rateprofile
rateprofile 0
```

I thought someone might have an idea when this could have got broken and what to revert. If so, I'm happy to test. Otherwise I'll eventually do a `git bisect` to find out. (But it's been quite a lot of commits since ~August.)

mikeller, 2017-12-24:
Thanks for reporting this. Do you know if it's still working with 3.2.3? If so, you can try 3.2.0, and then start your bisect from there (also, since bisect halves the number of commits in question with every step, it scales very well for large number of commits.

McGiverGim, 2017-12-25:
Maybe here? https://github.com/betaflight/betaflight/pull/4251

@azolyoung the new code is compatible with old firmware?

azolyoung, 2017-12-25:
Hi,

Thanks for reminding, and I'm orry about that, that is my fault :(

Recently I was entangled in something else, forgetting to add support for the older version. Some time ago, I thought our new firmware would be released soon, but it's still under test.

I've create a branch to add old split firmware support, coding is done, I'll test it tonight, if everything work fine, I'll submit a PR.

closed 2017-12-26 by mikeller

David-VG, 2018-01-02:
Hi,

I am using Betaflight 3.3 (build 479) and I can only control my runcam split if i go to the betaflight configurator and click at "save and reboot" (without changing anything, just connect and save &  reboot) every time I change the battery.

I can check more builds if you want me to.

andrejpodzimek, 2018-01-03:
Same here; my RunCam just doesn't respond. Not sure if saving / rebooting does anything.

azolyoung, 2018-01-04:
Hi David-VG & andrejpodzimek,

I've pull the newest code from bf master branch, and test it with my split, after port setup and modes setup, either connecting configurator or not, the split seems everything work fine with transimetter :(, May I know how to reproduce the problem? and what the build date of 3.3(build 479)?

azolyoung, 2018-01-04:
Hi David-VG & andrejpodzimek,,

If you are convenience, I can build a fw from the newest code for your FC to test it, Thanks

David-VG, 2018-01-04:
Hi, 

I was testing betaflight_3.3.0_BETAFLIGHTF4-479-5a448ea9b built the day 01-01-2018 at 9:28.

To reproduce the error I just need to turn on the taranis and plug a battery to the FC.

If I use the runcam related switches at the taranis i get no response.

If I then plug the usb cable to the fc and do a click at "save and reboot" at betaflight configurator the transmiter switches do their job until I power off and on again the FC.

I can try your new build if you want me to.

David-VG, 2018-01-04:
Just one more thing: I a using a Runcam Split v2.

andrejpodzimek, 2018-01-04:
I'm using v1. There's no response from the camera whatsoever, no matter if I plug in the battery or the USB into the FC. One FC is `ANYFCF7`, the other one is `OMNIBUSF7`, both with the same type of RunCam Split attached.

David-VG, 2018-01-04:
BetaFlightF4 here.

azolyoung, 2018-01-05:
ah, I think I know the reason, May be there is a bug in the logic of  initialize device, I'll make a branch to fix it, and upload the firmwares to here.

Thanks for your help

azolyoung, 2018-01-06:
https://s3.amazonaws.com/rctmpfiles/betaflight_3.3.0_ANYFCF7.hex
https://s3.amazonaws.com/rctmpfiles/betaflight_3.3.0_BETAFLIGHTF4.hex
https://s3.amazonaws.com/rctmpfiles/betaflight_3.3.0_OMNIBUSF7.hex
https://s3.amazonaws.com/rctmpfiles/betaflight_3.3.0_OMNIBUSF7V2.hex
hi @andrejpodzimek @David-VG, please try the above firmwares, I've test in my split ,it' works when runcam split was boot after flight controller. 

Note: when you downloaded the above firmwares, the file extension might change to "txt", please remove it(keep the file extension as "hex"), and use Betaflight Configurator to flash it to your FC

David-VG, 2018-01-06:
I cant try this build:

I use FPort with this quad and it seems broken. My taranis cant "talk" with the quad so i cant send orders to the runcam (with this build i wouldnt be able to fly also).

David-VG, 2018-01-06:
(The FC is in failsafe now)

David-VG, 2018-01-06:
Ok, I have read the changelogs from my last working setup. I've noticed they changed the default serialrx_inverted so my FPort worked with this new build.

And yeah, the patch fixed the problem with the runcam :)

knockoutperformance, 2018-01-06:
Runcam split 2 with dysf4pro hooked to UART 6 has the same problem as David-VG. If I plug in to dysf4 with USB, runcam will take commands from taranis q7. When unplugged, quits taking commands. Even if unplugging battery and replugging back in, its no go until I plug in USB and do a fc reboot.

andrejpodzimek, 2018-01-07:
I've tested the build on `ANYFCF7`. (My `OMNIBUSF7V2` uses the SD card pins for sofserial, which requires source&hack to make it work.) **It works** on `ANYFCF7`, to some extent.

I was able to start the camera using 3 channel changes (e.g. 0->2000->0->2000) and stop it using 2 changes. This makes the camera controls somewhat asymmetric, so one can't know for sure whether it's started, unless the drone is on the ground and the tiny camera beeper is audible. Previously it used to start and stop after exactly 2 channel changes, very reliably. Also, in roughly 1 case out of 3, only 1 channel change was sufficient to stop the camera, whereas a start (AFAICT) always required 3 changes.

azolyoung, 2018-01-08:
Hi guys, 

Thanks for the testing, I'll submit a PR to fix this bug.

and yes, we are preparing a new split firmware that is support some exactly commands about camera control, like start recording, stop recording, turn on wifi......, and after the new split firmware test done, we'll add these feature to configurator(actually, for now I' have no idea how to add these exactly behaviors control to configurator)

David-VG, 2018-01-09:
@azolyoung I think a command to record just while one channel is ON would be great. (For instance for recording just while armed).

David-VG, 2018-01-09:
@azolyoung also keep in mind that doing some clicks you can completly lose video (also mid flight).

andrejpodzimek, 2018-01-09:
Flipping the channel down->up->down is also OK, IMO. Recording only when the channel is on/high would limit the usability of TXs that have buttons. My Devos (10 and 12E) only have 2&ndash;3-position switches, but I don't mind flipping a switch back and forth, as long as it behaves in a consistent way. Currently it has a random feel to it &mdash; 3 channel flips start the recording and 1 or 2 flips stop it.

I haven't set the mode and wifi switching channels at all, exactly due to the fear of losing the video mid-flight. I don't fly FPV, so is't not a safety-critical issue in my case, but it would still be very disappointing.

David-VG, 2018-01-09:
Every single aux channel in betaflight works the same way the arm switch work: you switch it on or you switch it off, only runcam works as you do a button click.

andrejpodzimek, 2018-01-09:
Channels work all the same, but control elements on the TX are not the same — some TX types have (in addition to switches) also buttons that need to be held in order to keep a channel value up. These buttons are quite often configured (by default or by choice) to control cameras on some of the more “proprietary” drones. That’s why I wouldn’t consider using two channel changes (button press and release) for camera control a bad solution. The only important thing is that it needs to be consistent across multiple record and stop iterations.

estevanb, 2018-03-03:
Hi! I am using Matek F7. I updated the firmware to Betaflight 3.3.0 and my Runcam split V2 stopped working as well..

azolyoung, 2018-03-07:
@estevanb @explorethetruth 

Hi, I'm sorry for replying late, are you using the same aux with the split and arm? 
for now, I know there is a issue of rcsplit caused by the slow initializing, so we add a reinitializing when you try to trigger the modes of rcplist(e.g power button, wifi button), but there is a case, if your quad is arming, we won't reinitialize it, becuase initializing might make the quad hung for some seconds(1~3 seconds).

might be we need the async-initializing for rcsplit

explorethetruth, 2018-03-07:
@azolyoung 
I use classic arming by sticks method (I don't use any AUX for arming). But two buttons of rcsplit are attached to the same AUX.

estevanb, 2018-03-07:
I have 5 separate switches.

1 for pre-arm
1 for arm
3 for split

5 separate channels

Estevan
+1 (469) 407-8483

estevanb, 2018-03-19:
@azolyoung , I have just updated the matek f722-std to the latest Betaflight firmware and Runcam Split V2 is still not working.

When I flip the power switch to start recording I get the motors to give a short beep just like when we are saving a setup in Betaflight.

Appreciate if you can check on this issue.

andrejpodzimek, 2018-03-28:
I've just updated Runcam Split (the first 5V-only version) to firmware 2.x.x from January 2018. FC control stopped working completely. There is no response from the camera whatsoever. It used to work with the old firmware, albeit in a [somewhat quirky way](https://github.com/betaflight/betaflight/issues/4840#issuecomment-355828587). After the camera firmware update it just doesn't work at all. The camera itself appears to work fine. WiFi works, the recording button works etc.

This problem has been around for quite long. Would it be possible to introduce multiple RunCam options into the serial port settings, so that each type just works without this much of a hassle?

andrejpodzimek, 2018-04-05:
The problem still persists with 3.3.1.
Because this issue is closed (and it may be a different problem, as I already have the new RunCam firmware installed), I'll file a new one.

PR #4845 (2017-12-26): add old split firmware support
Fixes #4840
PR #4912 (2018-01-09): Improve initialization timing for RC Split
Fixes #4840

sometime the FC has prepared, but  runcam split might still initializing, so the FC can't init the runcam split.
this PR add a reinitialize logic when user trigger the camera control action