betaflight/betaflight #4847: F4 Dshot does not work

jflyper, 2017-12-26:
With #4843, all F4 targets WILL now use DMA burst feature for Dshot.

This change will
- Drastically reduces  DMA stream usage thus reduces motor output selection constraints, and
- slightly better motor command timing skew (depends on number of timers used).

However, as the change introduces different stream usage compared to the legacy driver, there may be targets which experience new stream collisions.

If you found an F4 board not working after #4843, please report here with the following information.

- Target name
- List of DMA stream assignments, obtainable by CLI command `resource list`, for
1. Pre-#4843 firmware (working state)
2. Post-#4843 firmware (non-working state)

mikeller, 2017-12-26:
@jflyper: I assume adding comments here for boards that are confirmed working would help as well (for other owners, to know it's working for their board)?

jflyper, 2017-12-26:
@mikeller Very true.

mikeller, 2017-12-26:
BEEROTORF4 motor 4 does not work with DMAR - fixed by #4849.

mikeller, 2017-12-28:
Added #4859 to enable the use of `target.h` files to track untested / unsupported / supported status for Dshot DMAR.

How to use:
- if a target is known to work with Dshot DMAR, add `#define USE_DSHOT_DMAR` to its`target.h`;
- if a target is known to **not** work with Dshot DMAR, add `#define USE_DSHOT_DMA` to its `target.h`;
- all targets that have neither of these defines will be built with Dshot DMAR enabled in nightly builds (to maximise test / feedback coverage), but built with it disabled for releases (safe option).

wolverin3, 2017-12-28:
Is this DMA burst already in BF 3.2.3?

mikeller, 2017-12-28:
@wolverin3: All 3.2 releases after 3.2.0 are maintenance releases, and contain only fixes for safety issues, bugfixes and target changes (i.e. no new features like DMA burst).

wolverin3, 2017-12-28:
May I know how to get the firmware with DMA burst so I can test it out on my CL Racing F4 and Hobbywing F4?

mikeller, 2017-12-28:
@wolverin3: https://ci.betaflight.tech/job/Betaflight/lastSuccessfulBuild/artifact/

wolverin3, 2017-12-29:
Tested on CL Racing F4 (1st version) with DShot600 and 8k/8k. Everything seems fine.

Hex: betaflight_3.3.0_CLRACINGF4-469-3c166a3e4.hex

ps: I'll test it out on OMNIBUSF4SD later and update you soon.

mikeller, 2017-12-29:
@wolverin3: Now that #4859 has been merged, would you mind opening a pull request to add `#define USE_DSHOT_DMAR` to `target.h` for this target?

wolverin3, 2017-12-29:
I don't mind at all but to be honest, I dunno how.

TheIsotopes, 2017-12-29:
Benchtest latest BF3.3.0 on MatekF405 and Revolt ... on both all 4 motors spinning ... and also DMA list shows 2 TIMUP instead of 4 Motors 👍

mikeller, 2017-12-29:
@wolverin3: All good then, I can do it.

AirBreak69, 2017-12-29:
Flight test with BF3.3.0 build 1912 on DYSF4PRO (OMNIBUSF4): No issues, 2 TIMUPs.

AndersHoglund, 2017-12-29:
Quick hover test with DYSF4PRO, seems OK.
```
# version
# Betaflight / DYSF4PRO (DYS4) 3.3.0 Dec 29 2017 / 17:56:29 (fe44d26) MSP API: 1.37

```

mikeller, 2017-12-30:
Thanks for your feedback @wolverin3, @AirBreak69, @AndersHoglund! Updated the targets in #4872.

ratmole, 2018-02-05:
SPRACINGF3: only 2 motors spinning...
rebuild with #define USE_DSHOT_DMA -> everything ok...

closed 2018-04-21 by DieHertz