betaflight/betaflight #4851: LEDs work right after boot, but deteriorate into randomness afterwards

andrejpodzimek, 2017-12-26:
I have a build using `OMNIBUSF7V2` with 30 LEDs installed. The first 16 LEDs are on [these devices](http://www.helipal.com/hglrc-motor-led-set-4pcs.html), 4 in a series, followed by 6 LEDs on [this device](http://www.armattanquads.com/pod-led/) and 8 LEDs on [this device](http://www.helipal.com/mateksys-h-shape-ws2812b-led-and-buzzer-dual-modes.html) (which doesn't have a `D-out` pad, so it has to be connected last). All the LEDs are powered from the 5V outlet on the PDB.

(The FC has its own separate 5V regulator. The build is for 8" propellers, so the length of the wiring is quite substantial. I'm mentioning all of this just in case if it could be a grounding / interference issue of some sort.)

Symptoms of the problem: Right after boot-up, the LED colors are exactly as configured. Throttle response, fixed colors, indicators etc. &mdash; everything works perfectly. But within a few seconds the LEDs start to flicker a bit and eventually turn into a pretty neutral (somewhat random) color that (mostly) doesn't have much in common with the initial configuration. When motors are running (DShot 1200), the blinking  / flickering gets quicker.

So the LEDs appear to work fine and initially get their colors right (including the very last ones in the chain), but this gradually turns into chaos. Not sure what could possibly be wrong. I've tried this with the current `master` and with 3.2.3.

These are the LED settings. I also tried to set them all to a constant color (with no throttle response or indicators) to check how that would work. But it's still the same; the colors are accurate for a few seconds and then deteriorate slowly. The crazily blinking randomness appears to be influenced (in terms of frequency and other characteristics) by the throttle.
```
led 0 13,15:ESD:CW:8
led 1 14,14:ESD:CW:8
led 2 15,13:ESD:CW:8
led 3 15,15:ESD:CI:8
led 4 15,2:NED:CW:4
led 5 14,1:NED:CW:4
led 6 13,0:NED:CW:4
led 7 15,0:NED:CI:4
led 8 0,13:SWD:CW:8
led 9 1,14:SWD:CW:8
led 10 2,15:SWD:CW:8
led 11 0,15:SWD:CI:8
led 12 2,0:NWD:CW:4
led 13 1,1:NWD:CW:4
led 14 0,2:NWD:CW:4
led 15 0,0:NWD:CI:4
led 16 9,10:S:ATW:0
led 17 8,10:S:ATW:0
led 18 7,10:S:ATW:0
led 19 6,9:S:ATW:0
led 20 7,9:S:ATW:0
led 21 8,9:S:ATW:0
led 22 9,6:N:COI:4
led 23 9,5:N:COI:4
led 24 9,4:N:COI:4
led 25 9,3:N:COI:4
led 26 6,3:N:COI:4
led 27 6,4:N:COI:4
led 28 6,5:N:COI:4
led 29 6,6:N:COI:4
```

The rest of the diff follows, just for the record. I'm using the MOSI pin of the SD-card slot (`PE6`) for SmartPort over SoftSerial. The LED chaos looks the same when I disable that, so it's probably irrelevant.
```
name Max

resource SERIAL_TX 3 NONE
resource SERIAL_TX 11 E06
resource SERIAL_RX 3 NONE
resource I2C_SCL 2 B10
resource I2C_SDA 2 B11
resource SPI_SCK 4 NONE
resource SPI_MISO 4 NONE
resource SPI_MOSI 4 NONE

feature SOFTSERIAL
feature GPS
feature TELEMETRY
feature RSSI_ADC
feature LED_STRIP
feature ESC_SENSOR
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

serial 0 16384 115200 57600 0 115200
serial 5 2 115200 57600 0 115200
serial 30 32 115200 57600 0 115200

set gyro_sync_denom = 4
set gyro_use_32khz = ON
set magzero_x = 131
set magzero_y = -334
set magzero_z = -44
set max_check = 1950
set blackbox_device = NONE
set min_throttle = 1030
set dshot_idle_value = 200
set motor_pwm_protocol = DSHOT1200
set align_board_roll = 180
set align_board_yaw = 270
set current_meter = ESC
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set gps_auto_baud = ON
set pid_process_denom = 1
set vcd_video_system = 1

profile 0
rateprofile 0
```

jflyper, 2017-12-26:
OMNIBUSF7 and OMNIBUSF7V2 have a problem on LED output circuit. Did you mod your board?

mikeller, 2017-12-26:
In addition to this; I think it's worth checking your voltage levels, and potentially adding one (or more, considering the number of LEDs you are running) diodes: https://github.com/cleanflight/cleanflight/blob/master/docs/LedStrip.md#connections (last paragraph)

andrejpodzimek, 2017-12-26:
OK, now I see that this is probably #4429. So I'll close this, because it's an (almost) obvious duplicate.

@jflyper  I haven't made any substantial modifications. it's in a pristine state, apart from the `PE6` wire (going to the SD-card slot and working as a SoftSerial). I'm connecting it to 6S directly, so I added [this capacitor](https://www.conrad.ch/de/elektrolyt-kondensator-radial-bedrahtet-35-mm-220-f-35-v-20-o-x-l-8-mm-x-15-mm-teapo-sy-220uf35v-8x15mm-1-st-443987.html) to the power pads. (However, I'm not sure how useful the capacitor is, because the total capacity I can measure on the main battery input (i.e., including the capacitors the ESCs have on board) is somewhere around 1.5 mF, which makes the extra 220 µF look quite small.) [This is the PDB.](http://www.armattanquads.com/f1-integrated-power-distribution-board-5v-12v-with-xt-60/)

@mikeller Thanks for the pointer; I'll measure the actual voltage on the 5V output. Should the "diode" be a standard rectifier diode (with a suitable forward voltage drop) or a Zener diode with a more precisely defined "backward" Zener voltage? (There are 1V and even 0.3V Zener diodes available.)

In summary, IIUC, all diode experiments are futile unless I resolder + remap the LEDs from the LED pad elsewhere, right? (Because the LED pad is flawed somehow on this board model.) One option may be the MISO pin on the SD card slot. So in `target.c` I'd remove&hellip;
```
DEF_TIM(TIM4, CH1, PD12, TIM_USE_LED,   0, 0 ), // LED
DEF_TIM(TIM9, CH1, PA2,  TIM_USE_NONE,  0, 0 ), // UART2_TX (unwired)
```
&hellip;and add&hellip;
```
DEF_TIM(TIM9, CH1, PE5,  TIM_USE_LED,  0, 0 ),  // MISO -> LED
DEF_TIM(TIM9, CH2, PE6,  TIM_USE_NONE, 0, 0 ),  // MOSI -> SoftSerial
```

Would that^^^ work? (The SoftSerial works fine for me already, but I'm not sure whether this could cause conflicts on `TIM9`.) The magic numbers are from the MCU manual [as mentioned here](https://www.rcgroups.com/forums/showthread.php?2884546-Omnibus-f7/page37#post38821973).

closed 2017-12-26 by andrejpodzimek

mikeller, 2017-12-26:
@andrejpodzimek: The forward voltage drop of a standard diode should be fine - the fact that your LED strip works under certain conditions shows (and the data sheet suggests) that the voltages are just borderline out of spec.

andrejpodzimek, 2017-12-27:
Here comes a facepalm feeling. I should have checked the 5V supply before anything else. After disabling 3/4 of the LEDs (set to `black`), everything works. Colors remain accurate etc. LEDs that don't shine exhibit random dim blue flickering, but that's hard to even spot against the operational LEDs.

The 5V supply has 4.9V when 8 LEDs are on, but drops to 2.6V when all 30 LEDs are on. That's it.

Then I probably don't need to switch to another LED pad. (I don't mind the blue flickering.) A better 5V power supply is what I need. The drone has 3 separate 5V supplies &mdash; for RunCam Split, for the FC+RX+GPS and for LEDs. That's why the problem wasn't too obvious. Had I had a single 5V power supply, the FC would have kept crashing.

ledvinap, 2017-12-28:
@mikeller: The diode in power line is probably only masking other problem. In most cases using series resistor on data line to suppress reflections should work better. 50-300 ohm should do ..

mikeller, 2017-12-28:
@ledvinap: No, in this case the diode is not masking another problem, but addressing the problem (as explained in https://github.com/cleanflight/cleanflight/blob/master/docs/LedStrip.md#connections that  ) that the 3.3V high of the GPIO is less than the specified 0.7 * Vin for the LED to recognise a logical high when driven at 5V.

ledvinap, 2018-01-02:
@mikeller : Hysteresis voltage is 0.35V typ, so actual decision level much closer to half input voltage (and probably centered a bit lower that half of VCC). Input specification in datasheet is copied from CMOS specification.

andrejpodzimek, 2018-01-07:
This story has (hopefully) come to a happy-ending. First, I bought [this BEC](http://www.helipal.com/3a-switch-mode-ubec.html). (It may seem bulky, but it's an 8" build, so what.)

With just the BEC, all LEDs were 100% white. Great. At least I knew there was enough power. The voltage in the all-white state was >4.8V and I left the drone sit for a quarter an hour. The fancy new BEC didn't get even warm!

Then I started to tackle the signal issue, based on the diode advice above. I tried what was usable [from this package](https://www.banggood.com/100pcs-8-Values-Diode-Bag-Assortment-Kit-1N4148-FR107-1N5408-For-Arduino-p-1031130.html?rmmds=myorder&cur_warehouse=CN). Only two of the diodes were rated for 3A, so I stuck with those. (If the LEDs take 60mA each at most, then I should have 1.8A with 30 LEDs. So a 3A rating for both the diode and the BEC sounds safe.)

First I tried a single [1N5408](https://www.diodes.com/assets/Datasheets/ds28007.pdf). Some LEDs worked, but some remained white. (Depending on the LED PCB manufacturer, presumably.) Some progress, but far from ideal.

My multimeter (which has a diode testing mode) estimated the drop to be only around 0.38V for the 1N5408, so I thought the drop had been insufficient and tried two 1N5408s in a series. This yielded a flickering randomness, the same state I had seen initially with the weak BEC. It turned out that the drop was in fact 0.75V per 1N5408 under load, so even a single 1N5408 was too much of a drop.

Eventually one single [1N5822](https://www.diodes.com/assets/Datasheets/ds23003.pdf) saved the day. It's a Schottky diode with a very low voltage drop. Colors look OK on all LEDs right now.
[image]

ledvinap, 2018-01-07:
@andrejpodzimek : Did you try the series resistance?
BTW: According to 'theory' in https://github.com/cleanflight/cleanflight/blob/master/docs/LedStrip.md#connections, only voltage on first LED need to be shifted, output is referenced to LED power supply, so it will be well within limit for rest of chain. So dropping voltage for first led/module only should save some power ...
Also placing more tolerant module first may fix this problem ...

andrejpodzimek, 2018-01-08:
Sadly, 4 of the least tolerant modules are connected before a tolerant one and a somewhat intolerant one ends the chain. So I’m better off putting all the LEDs behind a diode. Putting the tolerant module first would be too much rewiring, now that I have the build finally in one piece and airworthy.

I didn’t check the resistance, because I naïvely believed that what my multimeter showed me as a diode voltage drop (without any load, simply in the diode testing mode) would be also the drop value under load, constant up to (say) the rated current. Well, nope. Now I know that things are (as usual) more complicated than my childish assumptions about electronics.

mikeller, 2018-01-08:
Looking flash, @andrejpodzimek. :sunglasses:

andrejpodzimek, 2018-12-28:
> BTW: According to 'theory' in https://github.com/cleanflight/cleanflight/blob/master/docs/LedStrip.md#connections, only voltage on first LED need to be shifted, output is referenced to LED power [...]

Almost a year later, I can confirm that the theory works. :+1: This time I'm building a drone with 32 LEDs (8 blocks of 4). (It's an octocopter, hence the large number of LEDs.) It's (indeed) OK to only have a diode on the first block of LEDs. Then all the LEDs down the chain just work. Without the diode, the first LED on the first block flickers and the rest stays off.

Also the signal level is an interesting thing. On `ANYFCF7`, repurposing `PWM6` for LEDs makes it work even without the voltage-reducing diode. It's slightly flickery, but all the 32 LEDs work and show reasonable color. The problem with octocopter + DSHOT1200 is that I need ~all available timers and DMA channels and one cannot share a timer between motors and LEDs. So I had to move LEDs to UART1 (using `TIM1`, which is unused in the default build). That's when I had to add the diode. So it seems that the LED signal level produced by repurposed `PWM6` differs from what repurposed UART1 RX or TX generate (tried both).