betaflight/betaflight #4860: Unable to calibrate Blheli32 Current sensor input in 3.2.3

NitroNAF, 2017-12-28:
I have started building my drones relying on the Telemetry from my BlHeli32 ESC rather than a dedicated ADC input. I noticed the ability to Scale voltage and Current disappears when ESC is selected as the source for Vbat and Current. 

Manually changing ibata in CLI does not affect the sensor readings.

mikeller, 2017-12-28:
Yes, this is not implemented.

etracer65, 2017-12-28:
The BLHeli_32 developer said this was something they would look into adding.

https://www.rcgroups.com/forums/showpost.php?p=38743623&postcount=2069

mikeller, 2017-12-28:
Nice! Having this added to the ESC firmware sounds like a much better place for it to go.

etracer65, 2018-05-05:
The ability to calibrate the ESC current sensor is available in the latest BLHeli_32 firmware.

closed 2018-05-05 by etracer65