betaflight/betaflight #4862: Blackbox log viewer hangs on current 3.3 master logs

etracer65, 2017-12-28:
Trying to do some dev testing and I'm finding that logs generated by builds from the current 3.3 master cause the current release blackbox viewer to hang (100% CPU).  Tried downloading the current blackbox viewer source but it looks like work is in progress and Chrome won't load it as an extension - complains about default localization and the version numbering being incorrect.  Tried changing both and got it to register in chrome but it won't launch.

So not sure if it's a problem with 3.3 or changes that haven't yet been made to the viewer to support those changes.

Frustrated because I have a log captured of a takeoff spaz/intentional crash event I'd like to analyze but can't view it.

Here's the log in case it helps:

[baseline - gyro alignment incorrect.BBL.zip](https://github.com/betaflight/betaflight/files/1591619/baseline.-.gyro.alignment.incorrect.BBL.zip)

etracer65, 2017-12-28:
Trying again at gyro/pid at 4/2K and 2K logging (blackbox_p_ratio = 64) works.  So not sure if it's a view issue or some problem with the way the logs are written when using 2.6K/2.6K.

mikeller, 2017-12-28:
Yes, there seems to be a bug in the log viewer causing it to hang when trying to open logs with 'unusual' logging rates: betaflight/blackbox-log-viewer#77

etracer65, 2017-12-28:
I wasn't sure if it was a viewer problem or maybe something broke with recent commits in 3.3.  Looks like it's a view issue - I'll close this.

closed 2017-12-28 by etracer65