betaflight/betaflight #4867: RXLOSS CLI

Vosem-ADMIN, 2017-12-29:
I have connected my SBUS Turnigy IA6C like is shown on the MatekSys website: http://www.mateksys.com/?portfolio=f405-osd
but when I connect I cannot arm or do anything involving the transmitter. When I type status in CLI it shows up as RX LOSS. What can I do?

McGiverGim, 2017-12-29:
For user support, please use rcgroups or slack. This is for bugs or problems with the software itself.

But to give you some ideas, the Turnigy IA6C is a FlySky receiver, with iBus protocol (not sBus). If I'm not wrong the sBus is inverted and the iBus isn't. So check the diagrams on matek site and use the iBus one, not the sBus one.

Vosem-ADMIN, 2017-12-29:
Thanks so much

McGiverGim, 2017-12-29:
Go to the slack chat https://betaflightgroup.slack.com/ you will find the #support channel inside it. I'm pretty sure you will find the help you need.

Vosem-ADMIN, 2017-12-29:
Thanks but now with some fiddeling got it working 👯

mikeller, 2017-12-29:
@Vosem-ADMIN: Can you close this issue then please?

closed 2018-01-03 by martinbudden