betaflight/betaflight #4868: anti_desync_power_setup missing parameter in 3.2.0 ?

MattMgn, 2017-12-29:
Hello,

I can't set this param. I am running BTFL 3.2.0 on Matek F405 :
```
# set anti_desync_power_step = 10000
Invalid name
```
After reading all params with `dump ` cmd in CLI, it's actually missing

However, this param has been add in 2.8
https://github.com/betaflight/betaflight/wiki/Betaflight-specific-CLI-commands

Does anybody know this param still exist ?

Regards

Matt

etracer65, 2017-12-29:
The parameter no longer exists.  In 3.0 the equivalent function was "max_esc_throttle_jump".  This was completely removed in 3.1.  Sometimes features turn out be less useful then expected or not used and are subsequently removed.

MattMgn, 2018-01-03:
Ok, thanks

closed 2018-01-03 by MattMgn