betaflight/betaflight #4875: FPORT wiki F3 correction

elepete, 2017-12-30:
Just tried the FPORT on a SP-F3EVO with an X4RSB, no modifications just a wire from TX2 to SPORT: Works great!
One little issue in the Wiki documentation for an F3 (specifically the SP-F3EVO) you need:
serialrx_inverted = OFF
serialrx_halfduplex = ON

@FrSky - would be great to add the as in the standard config while jumpering Ch2 and 3 - the TX channels 9 and 10 to outputs 2 and 3

mikeller, 2017-12-30:
Thanks for your feedback, @elepete.

What strikes me as a bit weird is this:

> serialrx_inverted = OFF

In the default hardware config, the X4R-sb SmartPort / FPort pin is inverted, and thus the flight controller needs to be inverted too. Did you modify your X4R-sb to use uninverted SmartPort?

elepete, 2017-12-30:
Fully agree that this is strange and should by theory not be working...
My X4R-SB is not modified (as out of the box).
I now changed to an unmodified out of the box XSR and the setting with just a wire between TX3 and the XSR is per CLI, the same as with X4R-SB meaning inverted=off:
--- start c&p ---
# set serialrx
serialrx_provider = FPORT
Allowed values: SPEK1024, SPEK2048, SBUS, SUMD, SUMH, XB-B, XB-B-RJ01, IBUS, JETIEXBUS, CRSF, SRXL, CUSTOM, FPORT

serialrx_inverted = OFF
Allowed values: OFF, ON

serialrx_halfduplex = ON
Allowed values: OFF, ON
--- end c&p ---

mikeller, 2017-12-30:
Ok, I had a look in the code. The default for inversion that was used was how the existing RX models (with built in hardware inversion) work, not how the FPort specification (https://github.com/betaflight/betaflight/files/1491056/F.Port.protocol.betaFlight.V2.1.2017.11.21.pdf) defines it (no inversion).

#4878 fixes this, now the firmware is aligned with the FPort specification (and `serialrx_inverted = on` is required for legacy RX without the uninversion hack).

elepete, 2018-01-01:
Happy New Year and thank you - now with build #479 all in line with the Wiki and theory :-)

closed 2018-01-01 by elepete