betaflight/betaflight #4876: PIKO BLX EVO has compatibility problem

kaji331, 2017-12-30:
I use piko blx evo FC. When I use 3.1.x，I can not use dshot. Only motor 2 won't work, even the esc supports dshot. When I use 3.2.x (no PIKO BLX, only FF-PIKO-BLX)，same problem of dshot and I can't use sbus(uart3)，only ppm can work.

# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack: https://slack.betaflight.tech or other user support forums & groups (e.g. facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1.) Describe your problem;
2.) Include ways to reproduce the problem;
3.) Provide information about your flightcontroller and other components including how they are connected/wired;
4.) Add the used configuration and firmware version;
5.) Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/); and
6.) Remove this Text :).

jflyper, 2017-12-30:
What is the nature of the PIKO BLX EVO?

kaji331, 2017-12-31:
Sorry to the "instruction text", I was going to sleep when I wrote the issue.

There are two welding spots for UART3 on my PIKO FC. One is marked RC3, which I can connect sbus signal to UART3 in configurator for bf 3.1.x. Other is marked PPM, which I can connect ppm signal to UART3 in configurator for bf 3.1.x. I need dynamic filtering. When I used bf 3.2.x, I can still use PPM welding spot, but RC3 spot didn't work for sbus signal or ppm signal.

Not only bf 3.1.x, but also 3.2.x can't use Dshot to my PIKO FC. I saw somebody connected ppm spot to motor signal  spot and did some resource operation in bf configurator, after that PIKO FC can work with dshot. However, his modified PIKO didn't work dshot to motor 4, my PIKO didn't work dshot to motor 2. Neither bf 3.1.x nor 3.2.x.

jflyper, 2017-12-31:
@kaji331 So, what is the nature of the "PIKO BLX EVO"?

kaji331, 2017-12-31:
I do not understand the meaning of "nature"?

kaji331, 2017-12-31:
https://m.gearbest.com/flight-controller/pp_775587.html

jflyper, 2017-12-31:
@kaji331 We are sorry, the flight controller you are referring to is not supported by Betaflight. You have to consult your dealer for support.

closed 2017-12-31 by jflyper

Auxler, 2018-01-12:
You have to set via CLI „set sbus_inversion=ON“ and your FRSKY will work