betaflight/betaflight #4877: Connecting with betaflight configurator - missing COM port

nmw01223, 2017-12-30:
Not sure if this is the right place to put this or not.

Was using the Chrome app configurator, just started the Windows EXE as Chrome apps are about to die.

Problem is, the USB serial port is COM5 on my (W10) system as COM1-4 are otherwise used. Whereas the Chrome app appeared to enumerate the ports and add COM5 to the list, thus allowing connection, the EXE just appears to give you COM1-4 ... .

Could not find a way of giving it a manual connection string that worked.

McGiverGim, 2017-12-31:
I had a similar problem and I solved it with the ImpulseRC driver fixer https://impulserc.com/pages/downloads

Did you tried it?

AndersHoglund, 2017-12-31:
I have no such problem with the latest 10.0.1 version ( 2c84d92c ) on Windows, installed by the new installer. I see COM13 automatically when connecting a BETAFLIGHTF3 FC. This is a FC using STM32 VCP interface and driver. What type of FC are you having problem with?

Also, this is most likely not a firmware issue. It should perhaps be moved here: https://github.com/betaflight/betaflight-configurator/issues

nmw01223, 2017-12-31:
Ah, I think mine is 10.0.0. I'll try to find that version, mine didn't have an installer. I did get it to work with the manual string COM5 in the end. Wouldn't initially - must have been finger trouble. The FC is an Omnibus V4.

I note your point on the other issue list - I thought this one looked wrong, thanks.

nmw01223, 2017-12-31:
Not yet but I may, thanks. It is possible I've got an old version.

AndersHoglund, 2017-12-31:
Also. ImpulseRC Driver Fixer ( and Zadig) only installs drivers for BFC flasher connection to the BOOTLOADER. This is NOT the issue at hand here. 
The issue is BFC connection to BF firmware, and then you need the STM32 USB VCP driver installed. This does not seem to be any issue as it connected fine as a Chrome app. 
Please do not mix up those two different cases.

mikeller, 2017-12-31:
@McGiverGim: This has given me an idea: Now that we have the installer for Windows, we could look into adding a page to the installer that offers the user to download / install the drivers / driver fixer for them.

AndersHoglund, 2017-12-31:
@mikeller brilliant idea. 
The users need all the help they can get with driver installations. I am also getting fed up with all support issues and having to reply with copy-pasting the link to the WiKi install page. 
What if the BFC installer could even do a full job of installing all three drivers, Zadig and ImRCBF are only doing one third of the job. I.e. The BFC installer could install all of: WinUSB, STM32 USB VCP and Silabs CP210x. That would be a blessing, removing a lot of support issues.

mikeller, 2017-12-31:
Not sure if a totally automated full install is feasible for all of these - most of them seem to come with their own installer that needs some interaction. But it could be done by adding a page that told the user that they need these drivers, and had a number of 'Click here to download / install' buttons that downloaded the 3rd party installer and started it for them (Wireshark for Windows does (or at least used to do) this with the PCAP drivers for Windows).
It would be some work to set this up and maintain it, but as you say, @AndersHoglund, it will be a big improvement for what is one area where a lot of users are struggling.

McGiverGim, 2018-01-01:
@mikeller I'm not too sure about that. It is a good idea, but I don't know if it will be useful for the user. The vast majority of them will not know what to install when installing the configurator. They will press next, next, next...
The user detects the problem when executing the configurator the first time and it does not work. And the links to the drivers are in the first page of the configurator.

mikeller, 2018-01-01:
@McGiverGim: I can see your point. if it's just a page that allows the user to either push buttons to install drivers, or click next, then a lot of new users will just click next, and then struggle because they don't have the drivers installed. But if it's a page that has an explanation about the drivers and whyt they are needed, and requires the user to either install the drivers, or check a 'I know what I am doing, and I don't want to install the drivers' checkbox, then I think we should catch these cases.

This will obviously come at a cost of being an inconvenience. But this can be minimised if clicking next is only blocked if this is the first installation of the configurator.

AndersHoglund, 2018-01-01:
@mikeller That sounds as a very reasonable solution, having options for the user to bypass the driver installations. 
Please also remember the two different procedures needed. For the USB VCP and CP210x drivers, you only need to download and make the files available in the right places, Windows PnP will then do the final installation when the device is connected. 
The BOOTLOADER USB device interface is slightly different matter. Downloading and running Zadig or ImpulseRC DF is not a solution. Both requires the bootloader device being connected. And Zadig is also very manual. Instead the BFC Installer would need to do the inf-file modifications and possibly also registry modifications itself, replacing the Win/STM default driver with the "WinUSB" driver.  Without any device plugged in.

The BFC Installer could/should first of all check if all those driver issues has already been done. In that case, the bypass checkbox could be pre-set, or driver install part completely hidden. However, I think it would be good if there was still an option for the user to re-do or repair the driver installations.

mikeller, 2018-01-01:
@AndersHoglund: You are right, USB VCP and CP210x come as zip files. I thought they were installers. As you say, downloading and unpacking them could easily be done - I think it probably should still give the user a choice to opt out of doing so.

Re the dfu driver, the situation is not so easy there. I think at least downloading and putting the executables in a place where they can be found as part of the installer will make supporting these users a bit easier. If either Zadig or the IRC driver fixer can be run in a command line version, then we could add them to a wrapper script that runs them every time before it starts the configurator. Not sure how hard it will be to write a script on our own that does this driver change.

AndersHoglund, 2018-01-03:
Zadig/libwdi sources here: https://github.com/pbatard/libwdi
Might be useful if we want to implement something similar in the installer. 

Also, we should perhaps move this issue and discussion to the BFC github repo.

trickdriller, 2018-02-08:
I am having the same issue. I traced it to not being able to get my Omnibus to show up as a "port" instead of a USB device under Device Manager. I've tried everything the Internet can think of - no solution. Can anyone on here help?

AndersHoglund, 2018-02-08:
Wow, "..everything the Internet can think of..", except maybe reading the WiKi:  https://github.com/betaflight/betaflight/wiki/Installing-Betaflight

We must do something about this now. This is the fourth time only tonight I see this issue. Three users on FB also lost in the Windows driver maze.

trickdriller, 2018-02-08:
Did all that. It doesn't address be issue of the device not being recignized as a "port" instead of a USB device.

AndersHoglund, 2018-02-08:
Then you have not installed STM32 USB-VCP driver correctly and completely. Or your FC is stuck in bootloader mode for some reason.

trickdriller, 2018-02-08:
Negative. USB-VCP drivers are up to date. I got them directly from the updated Betaflight software. And my FC is not stuck in boot mode. I can put it in boot mode and load firmware without any issues. Betaflight even removes it from boot mode for me. I can also unplug the USB and reattach it to get out of boot mode.

AndersHoglund, 2018-02-08:
What? You do not get the STM32 USB-VCP driver from any "updated Betaflight software". You get it from ST Microsystems, as described and linked from the WiKi, Read one more time. Or go here:
http://www.st.com/en/development-tools/stsw-stm32102.html

trickdriller, 2018-02-08:
That link is on the main page of Betaflight, as well as in your link. I have signed up with my email, downloaded the software, installed it using 64 bit, as required - doesn't work.

AndersHoglund, 2018-02-08:
It works. You might need to restart Windows.

trickdriller, 2018-02-08:
Did that, twice. Still not working. Here, I'll attach a screenshot of the issue.
[image]

AndersHoglund, 2018-02-08:
Oh , yes. That was my first reaction, STM driver not completely installed. The ST readme file says:

```
***************
       1- Uninstall previous versions (Start-> Settings-> Control Panel-> Add or remove programs)
       2- Run your "VCP_V1.4.0_Setup.exe" 	   
	   3- Go to Your installation directory - Example, C:\Program Files (x86)\STMicroelectronics\Software\Virtual comport driver	   
	   4- Go to Your OS version directory ([Win7] or [Win8])
			+ Then :
				- Double click on dpinst_x86.exe if you are running a 32-bits OS version
				- Double click on dpinst_amd64.exe if you are running a 64-bits OS version

```
Might need to added to the WiKi, will check.

AndersHoglund, 2018-02-08:
It is already covered in the WiKi. No need to update.
But I can admit the WiKi page is rather poorly written and poorly structured, it is a mess of patch work corrections and additions. Not so strange a user gets lost.

AndersHoglund, 2018-02-09:
@fujin , the IMRC Driver Fixer is NOT applicable in cases like this. IMRC DF only fixes the driver for bootloader connection and flashing (WinUSB driver). It does NOT fix anything for the STM32 USB-.VCP device. Same goes for Zadig. So please only advice using any of those when it is a case of flashing, not in cases like this. That will only causes more confusion.

trickdriller, 2018-02-09:
Fujin: I did try that link as well. I'm attaching the error that popped up when I tried to execute that install from an alternate website. I had the same error pop up when I clicked your link. Any ideas on how to get around it? I tried the "search the Internet" button, then re-opened the link. Same issue.

And on the subject of being nice: I am the most giving person in all of my hobbies - always have been. I am not condescending when I help someone, ever. When someone sounds like they are at wits end with an issue, I do not say things like "Wow, "..everything the Internet can think of..", except maybe reading the WiKi." I have neither the desire, nor inclination to be nice to a person like that.

And yes, Anders, I'm talking about you.
[image]

AndersHoglund, 2018-02-09:
Well,  you did not read the WiKi completely, nor the STM readme file.  The installtion was not complete. Sorry if I did hurt you in any way. Getting a bit too tired for doing polite support.
And as said (to fujin above), running IMRC DF is not applicable in this case. STM32 USB.VCP driver installation has to be done manually and it is a multi stage rocket.

fujin, 2018-02-09:
Ah sorry, my mistake. I also thought it repaired the USB-VCP.

I suggest following my original instructions apart from that, then.

trickdriller, 2018-02-09:
I did read the website, and followed all the steps. The .exe file was extracted and installation was run. When that didn't work, I uninstalled it and did it all over again, as per the instructions you posted right here in this thread. It still did not work. 

I can see, when I try to manually add the drivers, that I have Omnibus F4 available, but I get that error when I try to add the driver. Would you like a screenshot of that error?

fujin, 2018-02-09:
After you extract the installer from STM and run the enclosed exe, there is _yet another_ exe that you need to run which actually installs the drivers. 

I gave the exact path to where it is on your file-system: `C:\Program Files (x86)\STMicroelectronics\Software\Virtual comport driver\Win8\dpinst_amd64.exe`. It is also covered in the README and wiki.

Please re-read my instructions and give it another shot. Also yes, post a screenshot of any error - We aren't mind-readers.

Do you see STMElectronics Virtual COM Port driver in the list of available drivers when you attempt to update the Omnibus F4 device per my screenshot?

trickdriller, 2018-02-09:
I do not see that on the list of available drivers. Not yet, anyway. 

I am making dinner, I'll give this all a try again when the children are settled. Thanks for sticking with me.

fujin, 2018-02-09:
If you do not see that in the list of available drivers during Update Driver then you have not installed the _actual_ driver from the path (in my last comment).

nmw01223, 2018-02-09:
Since I started this thread in December, I did get it sorted out in the end with the help of various youtube videos on sorting out the drivers. The most useful of these was [https://www.youtube.com/watch?v=Ur_mU7Eu69E](url) by Blue Falcon.

It's a minefield, though.

woutput, 2018-02-09:
Hi all :)

>  @AndersHoglund commented on Jan 1 
@mikeller That sounds as a very reasonable solution, having options for the user to bypass the driver installations.
Please also remember the two different procedures needed. For the USB  VCP and CP210x drivers, you only need to download and make the files  available in the right places, Windows PnP will then do the final  installation when the device is connected. The BOOTLOADER USB device interface is slightly different matter.  Downloading and running Zadig or ImpulseRC DF is not a solution. Both  requires the bootloader device being connected. And Zadig is also very  manual. Instead the BFC Installer would need to do the inf-file  modifications and possibly also registry modifications itself, replacing  the Win/STM default driver with the "WinUSB" driver.  Without any  device plugged in.
The BFC Installer could/should first of all check if all those driver  issues has already been done. In that case, the bypass checkbox could  be pre-set, or driver install part completely hidden. However, I think  it would be good if there was still an option for the user to re-do or  repair the driver installations.

I think that the fully automated method of driver installation is the way to go, because that would be most user friendly. I also don't think it is something useful to learn to all BetaFlight users.
I don't consider myself a beginner on Windows/drivers/programming, but to me it was unclear, when I started using a quadcopter & BetaFlight, that the FC can be in two modes. Both of these modes require a different driver and both of these drivers can only be installed/configured when both of these modes have been "shown" to Windows. This took me at least 30 minutes given the many websites and YouTube tips around this that are confusing and/or wrong  (example: "Q: How to program BetaFlight? A:Press boot button" is incorrect in many cases)
Solving this issue completely by automating all of this would help a lot I think.

>  @AndersHoglund commented on Jan 3
Also, we should perhaps move this issue and discussion to the BFC github repo.

Agree :)

trickdriller, 2018-02-09:
Ok, I'm going back at it today. 

nmw01223: the link is broken. What's the name of the video and I'll search for it on YouTube?

AndersHoglund, 2018-02-09:
One more example from FB how misconceptions, myths and disinformation is spreading like wildfire on the driver issue. An advice from one guy to another with problems connecting:

> Connect your quad . Open betaflight but dont connect. Minimize betaflight. Open chrome and type - ipmulse rc driver fixer . Run the fixer till done. Then connect on betaflight . The drone will then connect.

I did not even bother asking where and how that information was initially found. Total nonsense.  We really need to take control on this. If I only knew how to do that, written instructions are not enough. Strange how facts, knowledge, science and truth have a hard time against all the noise.  Sad.

trickdriller, 2018-02-09:
Well, finally, it worked. Fujin and Anders: I'm big enough to admit when I am wrong - and I was wrong. I thought I ran through the entire STM file and understood it. I did not. The link to the Wiki page revealed the step I was missing: navigating through the program files. Once I installed the driver, it was a matter of seconds to get everything tied up. Thank you both for your help.

nmw01223, 2018-02-10:
> nmw01223: the link is broken. What's the name of the video and I'll search for it on YouTube?

It's definitely the right link, just tried it. It opens if you copy/paste it (https://www.youtube.com/watch?v=Ur_mU7Eu69E), but not apparently directly from the embedded link. However, the name is from Project Blue Falcon - 'Making Flight Controller Connect Through USB'.

AndersHoglund, 2018-02-10:
Wow, that Blue Falcon video was one more example of a confused guy doing a lot of unnecessary complicated steps and missing a few important steps. First of all he is calling installing a driver "flashing the driver". Then I get a suspicious if he actually knows what he is talking about. He is totally missing the steps to download and install both Silabs and STM drivers. He assumes some of those are already installed, and uses Windows Serial drivers in some case. He is using Zadig to destroy an installation ("flashing") and then repair it with device manager, when all you need to do is to run the Silabs or STM installers. He is claiming F3 and F4 USB-VCP are different, they are not. They use the same STM USB-VCP driver. What a collection of misconceptions, broadcasted and applauded.

nmw01223, 2018-02-10:
Maybe, but none of that alters the fact that by following his instructions, I got it all sorted out.

I think there's a difference in the way software people (who understand internal structure) and non-software people (who don't) view the world.

AndersHoglund, 2018-02-10:
Yeah sure, everything is easier when you know how things work. You can do the simple steps, without any need to know anything on internals:
1. Download driver.
2. Install driver. (Might be a few steps)
3. Done.

Instead he is messing with Zadig and Device Manager etc etc, that really requires internal knowledge.
Zadig is a sharp tool, you can cut your self and do a lot of damage if you do not know what you are doing. Zadig is only to be used when replacing the driver for STM32 BOOTLOADER. But ImpluseRC Driver Fixer is doing the same job, and much safer as you can not do much damage with it. 
Joshua B has made a video on the simple straight fwd way, linked in the WiKi.

haslinghuis, 2022-05-08:
Closing as inactive

closed 2022-05-08 by haslinghuis