betaflight/betaflight #4882: different Artificial Horizon

BlackPixxel, 2017-12-31:
Hi!

I don't like how the current artificial horizon is fixed on the x-axis and can only travel up and down. When the aircraft rolls, the horizon becomes larger in its overall length, and it only supports rolls to a certain angle. If you go above that angle, it get's stuck at its max angle and you don't get any usefull roll information from it.

In real fighter jets and helicopters the artificial horizon is often perpendicular to the shortest line between aircraft direction and horizon.
I tried to implement that and here is how it looks with the Betaflight OSD

Notice how you can roll at any angle without the horizon getting stuck.
I added an error pointing at the shortest way to the horizon if the horizon gets far away from the screen center or even out of the screen.

The downside is that resolution on the x-axis is limited to the width of a char. That means that if you fly knife edge, the pitch movement would not have such a fine resolution on the horizon as if you fly level or inverted.

I like this approach alot more, because this is how I know it from DCS or Arma.

Some other aspects which I would like to see in general are the following two:

Let the user adjust at least the Y-position of the crosshair/level horizon if they fly with camera tilt.

Let the user adjust the "sensitivity" of the horizon pitch movement. This way the user could set it so that the artificial horizon is always where the real horizon is in the camera image (as it would be in a real fighter aircraft).

DanNixon, 2018-01-01:
> *Let the user adjust at least the Y-position of the crosshair/level horizon if they fly with camera tilt.

This is part on an existing PR.

> Let the user adjust the "sensitivity" of the horizon pitch movement.

This is already done (see `osd_ah_max_pit` and `osd_ah_max_rol` CLI variables).

bkeinert, 2018-01-01:
That looks like a nice acro capable AHI from here - any footage of a real flight?

BlackPixxel, 2018-01-01:
I hope to get some flight footage tomorrow

BlackPixxel, 2018-01-02:
Here is a little flight recording with the modified artificial horizon:
https://www.youtube.com/watch?v=yr96a10hkXw

My plane is not really made for acrobatic flight, but you can still see how the horizon works.
I found the little arrow to be very helpfull during loops. It shows when the loop does not go straight up but sideways instead.

richard-scott, 2018-01-03:
Is there any way to reconfigure the update speed in the CLI? It seems quite slow changing direction when you invert your plane and would be nice to be able to tweak that refresh/update speed.

BlackPixxel, 2018-01-03:
I already increased the TASK_PERIOD_HZ of the OSD in the fc_task.c and I think it is really fast. The default value of 60 causes a very slow updating.

richard-scott, 2018-01-03:
Its just that during one of the barrel rolls it took about 2 seconds for the arrow to change direction once you were inverted.

BlackPixxel, 2018-01-03:
At which minute? To me the arrow looks like it does what it should

Aileron8, 2018-01-04:
I really like it and hope it gets implemented. I fly FPV acro, but could really see the advantage for planes and fixed wing, beginner quad pilots and LOS flyers just starting. I don't see any delay, the arrow disappears, because the crosshair is out of view.

bkeinert, 2018-01-04:
Looks really nice! How much code is this?
Maybe we could have a setting for this, I guess some will probably still prefer the old style.

Another thing, while we are at it: Is this properly centered? Because the default BF crosshair is not centered properly. It's made of 3 chars (-+-), you can not center that on a 30 chars wide display.
I changed the fonts such that the crosshair is in the center of the camera image (one char blank, crosshair made of two chars). To be able to center things on the screen you have to work with an even number of chars.

ItsMeAubey, 2020-09-11:
This is awesome. Bump?

closed 2026-09-26 by blckmn