betaflight/betaflight #4884: Ov showing with dys f4

nkkr77, 2018-01-01:
Iam using dys f4 with vbat monitor enabled also i couldnt see my battery voltage status.
I dont know how to fix this.
i also use a pdb.
is using pdb causing this error ??
thanks

wolverin3, 2018-01-01:
You should post this in DYS F4 forum in RCGroups.

jflyper, 2018-01-01:
@nkkr77 
Your issue looks like a hardware fault or a configuration problem, and unfortunately, this is not a place to solve end user's individual problems.

Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack: https://slack.betaflight.tech or other user support forums & groups (e.g. facebook).

closed 2018-01-01 by jflyper