betaflight/betaflight #4888: Kiss V1 on 3.2.2  not recognizing CRSF

mountaingypsee, 2018-01-03:
Kiss V1 FC with 3.2.2 and crossfire micro will not get a signal through any of the 3 given UARTS.

jflyper, 2018-01-03:
@mountaingypsee Do you have an experience with the particular Crossfire micro working with other FCs?

mountaingypsee, 2018-01-03:
Yes. In fact all my quads have crossfire micro  and was just working on
this quad with Kiss firmware but enjoy betaflight more these days.

mountaingypsee, 2018-01-03:
If this is a me problem, just let me know and I'll go about it but I've done a fair bit of research on how to set it up and unless i need to go into CLI or jump some pads on the FC...... Maybe it is an Issue?

jflyper, 2018-01-03:
I suspect configuration problem(s). That's why I asked "the particular micro" you are using has been successfully used in other crafts.

mountaingypsee, 2018-01-03:
sorry to bother you with this then. Any links on proper config with the Kiss V1 fc CRSF?

mountaingypsee, 2018-01-03:
I checked. I have it wired the same way i did when the board had Kiss firmware on it. I'm not running telemetry and everything is up to date.

jflyper, 2018-01-03:
Can I see CLI "diff" output?

mountaingypsee, 2018-01-03:
I had to leave..... what should it be, I'll check it tomorrow? 
Thanks for your help by the way. Much appreciated.

richard-scott, 2018-01-03:
Have you tried your problem Crossfire RX on a working quad? and vise versa? Maybe the unit is faulty?

mountaingypsee, 2018-01-03:
yes actually i flashed the FC back to Kiss and it works perfectly fine.

mountaingypsee, 2018-01-03:
but id still rather be using betaflight than Kiss..... you can actually tune out oscillations.....

mountaingypsee, 2018-01-03:
[image]

jflyper, 2018-01-03:
Hmm... Can you do `resource list` and see serial_rx/tx 1 are there? (I would prefer copy and paste rather than pic...)

mountaingypsee, 2018-01-03:
gentlemen, the Tx from the the receiver needs to be soldered to the RX tab on the flight controller, not the SBUS tab. Have a great day and thanks for the help

mountaingypsee, 2018-01-03:
I sorted it out. The receivers rx wire needs to go to the Rx pad on the flight controller instead of the set sbus pad.....

jflyper, 2018-01-03:
Do we have pin/pad map that identifies RXn/TXn and SBUS?

mountaingypsee, 2018-01-03:
I moved the signal wire from the SBUS tab to the RX pad and it was good.
With Telemetry on kiss you run the the wire to the Aux1 tab.... not sure if
it works on BF as i dont run telemetry. See the attached photo.

mountaingypsee, 2018-01-03:
im sure it could easily be changed in CLI instead of soldering....

jflyper, 2018-01-04:
I'm a bit uncomfortable with how and why the issue was resolved. Just a personal impression, no offense.

lichtl, 2018-01-04:
I did some testing about half year ago on 3.2.0 (need to double check if this was with frsky sbus or crossfire) Can have a look later to check because I in the believe that it was all working without resoldering

mountaingypsee, 2018-01-04:
be what you want, my quad flies.

mikeller, 2018-01-10:
@jflyper: This is a bit strange indeed. Does the KISS V1 have an inverter on the SBUS pad?

Drewlrox, 2018-01-12:
Kiss v1 Is an f3 so I dont think they would use a hardware inverter on the sbus pad. I do know that they are using softserial for the telemetry part of crsf unlike in a standard betaflight setup where you use both tx and rx of a matching uart with crossfire.

jflyper, 2018-01-14:
@Drewlrox That sounds weird... Any theory behind the use of different serial ports for rx and tx?

Drewlrox, 2018-01-14:
@jflyper I'm not too sure, I was told it's ran at a higher refresh rate but I don't know if that's for the telemetry link or rx link. I assume it's just because that's how it's done for frsky on kiss and they wanted to keep the same 2 pads for crossfire as to not cause confusion. Kiss v1 also doesn't have too many free uarts to just wire up too. Everything already kind of has its place, like osd goes to 1 place and a 2nd option for is is the same uart in one of the side ports for the steele pdb, smart audio and esc telemetry have their own pads,etc. I'm sure they just wanted to use the existing rx pads they use for other recievers.

lichtl, 2018-01-14:
just tested. crossfire indeed do not work on the original pads on any UART while once change crossfire rx to SBUS (inverted) it straight work on UART2 (default)

`# resource
resource BEEPER 1 B13
resource MOTOR 1 B14
resource MOTOR 2 B00
resource MOTOR 3 B15
resource MOTOR 4 A08
resource MOTOR 5 A06
resource MOTOR 6 A07
resource PPM 1 B03
resource PWM 1 A13
resource PWM 2 B03
resource PWM 3 A15
resource PWM 4 A02
resource PWM 5 B11
resource LED_STRIP 1 A06
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 B03
resource SERIAL_TX 3 B10
resource SERIAL_TX 11 A13
resource SERIAL_TX 12 A15
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 B04
resource SERIAL_RX 3 B11
resource I2C_SCL 1 B06
resource I2C_SDA 1 B07
resource LED 1 B01
resource ESCSERIAL 1 A13
resource ADC_BATT 1 A00
resource ADC_CURR 1 A02

# diff
# Betaflight / KISSFC (KISS) 3.2.4 Jan 10 2018 / 22:53:29 (90d32fd56) MSP API: 1.36

set serialrx_provider = CRSF
profile 0

rateprofile 0

# `

mountaingypsee, 2018-01-15:
> Betaflight / KISSFC (KISS) 3.2.4 Jan 10 2018 / 22:53:29 (90d32fd) MSP API: 1.36

There we go! Nice one.

lichtl, 2018-01-15:
I also did a quick test using non inverted SBUS from the micro rx. Didn't worked either (I also tried using the **sbus_inversion = OFF**)

mountaingypsee, 2018-01-15:
And mine is working on the micro Rx without any changes to the board or in CLI.

zukasmichael, 2018-03-25:
does anyone has a final solution for CRSF kiss V1 running betaflight ? SBUS works fine, but CRSF not. I don't want to resolder anything wiring in the picture https://static.rcgroups.net/forums/attachments/7/2/0/8/5/4/a10304473-132-kisscrsf.jpg

Drewlrox, 2018-03-25:
@zukasmichael aux1 pad on kiss is a softserial port. Betaflight needs tx and rx of crsf on the same uart. I don't think there is a way around this.

zukasmichael, 2018-03-25:
@Drewlrox thanks for the answer. Do you know by chance how to get telemetry ( voltage and amperage ) from kiss 24A esc's on kiss fc, they are all connected to TRX pads as kiss requires now. like in the picture https://static.rcgroups.net/forums/attachments/7/2/0/8/5/4/a10304473-132-kisscrsf.jpg

Drewlrox, 2018-03-25:
@zukasmichael a long time ago I actually did use esc telemetry on existing pads but somebody had to compile a firmware for me that was expecting it on tx instead of the typical rx as kiss does it backwards from betaflight. I don't remember exactly what they did but I kind of rememebr somebody commenting saying it can be done without compiling a custom firmware. In a little bit when I'm home I'll try to find the conversation. Maybe somebody more familiar with betaflight here can comment on what the change may be. And I should have a version of 3.2 that is flipped somewhere I could send you if you want.

Drewlrox, 2018-03-25:
@zukasmichael found it, paste this in cli and it should work!!

feature SOFTSERIAL
feature TELEMETRY
feature ESC_SENSOR
serial 2 1024 115200 57600 0 115200
serial 30 32 115200 57600 0 115200
set motor_pwm_protocol = DSHOT600
set current_meter = ESC
set battery_meter = ESC
set esc_sensor_halfduplex = ON

zukasmichael, 2018-03-25:
@Drewlrox thank you a lot, I'm trying to migrate from kiss firmware to betaflight. You saved me a day. Thank You a LOT !!!

zukasmichael, 2018-03-25:
it still does not work. I use BTFL 3.3.0 (Target: KISS)
I'm using RX and TX for crosfire. Not sure how should I go to get telemetry from KISS 24A esc's :[

mikeller, 2018-03-25:
@zukasmichael: Instead of trying to diagnose it end-to-end, first try to get voltage / current to work when looking at the 'Battery' tab in configurator. This will make fault isolation much easier.

Drewlrox, 2018-03-25:
I would double check that all the commands are still current by verifying changes in gui. Beyond that I'm not sure. Hopefully somebody else here that knows more can help.

zukasmichael, 2018-03-25:
@mikeller I did a Sunday with no luck. There is literally 0 info about how to setup betaflight/crossfire/telemetry with kiss fc v1 on internernet :[

Drewlrox, 2018-03-25:
@zukasmichael here is my copy of 3.1.7 that works, maybe somebody can look at it and see what was changed https://drive.google.com/file/d/0B0BzJ7kqbDpPdk5iZERQN1hOcWc/view?usp=drivesdk

danielschafer, 2019-04-07:
Unfortunately this still doesn't work. I tried the hole weekend to get this to work.

It works fine with SBUS, but not with CRSF protocol.

Tried a KISS v1 and KISS v2 F7 board. Both don't work.
I also tried BF 4.0.
Also tried to make a unified target for KISSV2F7 board, but failed since the MSP (USB) is connected to UART4.

mikeller, 2019-04-08:
Not sure why CRSF won't work with KISS hardware. The only thing I could imagine is that the boards might have some on-board drivers / filtering for the UARTs that restrict the maximum speed. Maybe @ronlix can clarify?

Re the unified target for the KISSFCV2F7, I don't think this will ever work, as this board requires a custom boot loader, and therefore the firmware needs to be built with a different memory allocation.

danielschafer, 2019-04-08:
Thanks for your reply.
Actually it's working now on UART1.
UART2 (default for SBUS and CRSF on kiss) and UART3 aren't working!

danielschafer, 2019-04-15:
@mikeller I'm still trying to figure that out.
Why is CRSF not working on UART2 on KISSFCV2F7? Somehow Felix figured it out on his KISS FW. What needs to be changed to have CRSF protocol on Betaflight?

mikeller, 2019-04-16:
@danielschafer:

> Somehow Felix figured it out on his KISS FW.

I think 'figured it out' is not really appropriate here. He (@ronlix) is the designer of this hardware and has access to all schematics - so yes, hopefully he knows how it was meant to be used. We can only guess at this, and we've asked him to provide info (https://github.com/betaflight/betaflight/issues/4888#issuecomment-480709125), but if he does not want to share, and our guesswork is not successful then there is little we can do.

> What needs to be changed to have CRSF protocol on Betaflight?

Your guess is as good as anybody else's.