betaflight/betaflight #4891: More telemetry values for Jeti EX-BUS

RoadyFPV, 2018-01-03:
Hi,

is it possible to implement more sensor values to the Jeti EX-BUS protocol?
like the roll + pitch angle and so on?
I think it must be changd in the config?

mikeller, 2018-01-09:
It will probably have to be added in the telemetry code. I don't know if any of the Betaflight developers out there have Jeti EX-BUS gear available to implement / test this, but any contribution of more values for this would be appreciated.

RoadyFPV, 2018-01-10:
Hi
I cant coding :(  But i think testing is no problem.

mikeller, 2018-01-10:
@RoadyFPV: Fair enough. All help is welcome!

marv-t, 2018-01-15:
What values ​​would be needed?

RoadyFPV, 2018-01-15:
@marv-t:
roll and pitch angle ; rel. Altitude (from start point); Power (Watt);

marv-t, 2018-01-25:
#5030 
Here is an update for the jeti telemetry.
Altitude was already implemented.
New values: Power, Roll angle, Pitch angle and Heading

RoadyFPV, 2020-11-16:
@marv-t is it possible to get a new Telemetry value (total distance traveled)
I dont know exactly, betaflight outputs this value?

RobinatiNav, 2023-04-08:
> Hi,
> 
> is it possible to implement more sensor values to the Jeti EX-BUS protocol? like the roll + pitch angle and so on? I think it must be changd in the config?

Christoph forget about 50 or 100Hz. I tested it with my Jeti setup and telemetry was freezing using both. I know how annoying it is when you cannot help yourself. Sorry for writing you here but as you know I was blocked in iNav. I am going to delete this post after you read it. Here is my email robert.bulko at gmail dot com if you dont mind send me yours please I would like to ask you for the 6.1 hex file for MATEKH743 compiled in GCC 9.3.1 when it will be released. Thank you.