betaflight/betaflight #4895: Cell count detection fails if charged to vbat_max_cell_voltage

jflyper, 2018-01-03:
There has been several reports of cell count being detected incorrectly. I suspect some of these cases are caused by inadequate filtering of the incoming voltage measurements.

Here's how the filter and cell detection responded to 3S LiHV charged to 12.9V. `vbat*` parameters were defaults (`vbat_max_cell_voltage = 43`).

[image]

Things to note are:

A. Filtered voltage overshoots, as much as full 1[V] higher than actual voltage (unfiltered),
B. as soon as filtered voltage begin to settle,
C. isVoltageStable kicks in when filtered voltage is much higher, probably around +0.6[V],
D. and believes it has a 4 cell.

It's also interesting that
E. `isVoltageStable` is fluctuating a lot.

We have been telling people to stay within the 4.2V per cell limit, but the way `filtered` overshoots, lipos charged little over 4.1V per cell may be detected as 4 cell: 4.1 * 3 + 0.6 = 4.3 * 3.

Thoughts?

martinbudden, 2018-01-03:
Based on a min cell voltage of 3.3V and a max cell voltage of 4.3V we have:

cells | minV | maxV
--- | --- | ---
1 | 3.3 | 4.3
2 | 6.6 | 8.6
3 | 9.9 | 12.9
4 | 13.2 | 17.2 
5 | 16.5 | 21.5
6 | 19.8 | 25.8
7 | 23.1 | 30.1
8 | 26.4 | 34.4

so I suggest something like:
```
if (voltageMeter.filtered < 66) {
    batteryCellCount = 1;
} else if (voltageMeter.filtered < 99) {
    batteryCellCount = 2;
} else if (voltageMeter.filtered < 132) {
    batteryCellCount = 3;
} else if (voltageMeter.filtered < 175) {
    batteryCellCount = 4;
} else if (voltageMeter.filtered < 220) {
    batteryCellCount = 5;
} else if (voltageMeter.filtered < 260) {
    batteryCellCount = 6;
} else if (voltageMeter.filtered < 310) {
    batteryCellCount = 7;
} else {
    batteryCellCount = 8;
}
```
on the grounds that when people start their quad, they are likely to have a fully charged or nearly fully charged battery. On the higher cell counts there is considerable overlap, so this assumption is necessary.

ledvinap, 2018-01-03:
As for overshoot - second-order and higher Butterworth does overshoot.
Critically damped filter may work ...

jflyper, 2018-01-03:
@martinbudden That doesn't work. Fully (or over) charged 3S with 4.3V cell voltage (12.9V pack voltage) will produce as high as 13.8~14.0V after filtering, and probably around 13.5~13.6V when `isVoltageStable` kicks for the first time after connection.

@ledvinap It's a biquad, so ... Butterworth, in some sense.

What if we make `isVoltageStable` two stage?

Let `isVoltageStabliziing` be the current `isVoltageStable`, and if `isVoltageStabilizing` persist for settling time, then turn on `isVoltageStable` that can be used to detect cell count.

jflyper, 2018-01-03:
Stash it. It takes forever (about 2~3 secs) to stabilize. Moreover, I don't like watching battery voltage spike and settle...

martinbudden, 2018-01-03:
@jflyper , just adjust values upward accordingly, ie:

```
} else if (voltageMeter.filtered < 139) {
    batteryCellCount = 3;
}
```
The point being we should be checking against min voltages, not max voltages, since battery is going to be fully charged or near so, rather than empty.

ledvinap, 2018-01-03:
@martinbudden: Cell calculation is using `vbat_max_cell_voltage` for this, how would if cascade help?

Filter overshoot is causing the problem. Using different filtering algorithm (that has better step response) should help to fix this. Critically damped filters have this property - all that is necessary is to calculate coefficients for biquad. Just adjusting Q for existing calculation may work.
Filter performance will be worse, but that ma be ok ..

martinbudden, 2018-01-03:
@ledvinap , filter overshoot is not the cause of the problem - filter overshoot just makes it manifest itself more often.

The problem is there is overlap in battery voltages, as shown in my table above. So a battery with a voltage of 17.0V could either be a highly charged 4S battery, or an almost discharged 5S battery. There is no actual overlap between 3S and 4S batteries, on the assumption that the max cell voltage is 4.3V, but that assumption might not hold as batteries get better.

The second problem is the current code determines the cell count by dividing the voltage by `vbatmaxcellvoltage` (and adding 1) - this is precisely where maximum ambiguity occurs because users start their aircraft with fully charged (or almost fully charged) batteries. A better result could be obtained by dividing the voltage by `vbatmincellvoltage` (and not adding 1).

My cascade is slightly better than dividing by `vbatmincellvoltage` because it allows some adjustments in the ranges to help avoid ambiguity (this takes effect for 5S batteries and above) - and so is more "future proof" - it is not uncommon for people to fly with 5S and 6S batteries.

ledvinap, 2018-01-04:
@martinbudden: The cell detection algorithm assumes that max cell voltage is pretty well defined and that is it never exceeded. Current code is effectively `cells = ceil(packVoltage/vbat_max_cell_voltage)`.
`vbatmincellvoltage` is used for battery critical detection, user can set it pretty high to improve battery life or pretty low to prevent false alarms. Using it for cell detection will cause problems

Problem with Butterworth filter is that is overshoots on step input (connecting battery), possibly by 8% (not sure is overshoot is cutoff dependent). With fully charged battery, 4.2*1.08 = 4.53V, so this will exceed vbat_max_cell_voltage with wide margin.

`isVoltageStable` complicates this even more - it will be set only when unfiltered value is within 200mV (VBAT_STABLE_MAX_DELTA) from filtered. This lets unfiltered measurement noise in - detection code will trigger on first `isVoltageStable` sample (with 5Hz sampling).
This will usually trigger detection with `filtered` ~ `unfiltered`-200mV

martinbudden, 2018-01-04:
@ledvinap , you are missing my point.

The point is that, because cellCount is determined when the battery is almost fully charged, it is better checking that the battery is above a minimum voltage threshold rather than checking it is below a maximum voltage threshold.

ie `voltage > 13.2` is a better way of checking for a 4S battery than `voltage < 17.2`. (Assuming we have  excluded 5S batteries and above).

And since we know the battery is near fully charged, we can raise this voltage threshold and further decrease the chance of mis-identification, ie use `voltage > 14.0` even.

martinbudden, 2018-01-04:
I've written my cascade statement the other way around, to try and make this clear:
```
if (voltageMeter.filtered > 320) {
    batteryCellCount = 8;
} else if (voltageMeter.filtered > 260) {
    batteryCellCount = 7;
} else if (voltageMeter.filtered > 220) {
    batteryCellCount = 6;
} else if (voltageMeter.filtered > 180) {
    batteryCellCount = 5;
} else if (voltageMeter.filtered > 135) {
    batteryCellCount = 4;
} else if (voltageMeter.filtered > 99) {
    batteryCellCount = 3;
} else if (voltageMeter.filtered > 66) {
    batteryCellCount = 2;
} else {
    batteryCellCount = 1;
}
```

ledvinap, 2018-01-04:
IMO you are describing current code - current code test that battery voltage is higher than maximum expected voltage of pack with one-less cell.

Assuming values from table in https://github.com/betaflight/betaflight/issues/4895#issuecomment-355084024, 21V (5*4.2) pack should be clasified as fully charged 5-cell instead of discharged 6-cell ...

ledvinap, 2018-01-04:
And pack with 22v (`voltageMeter.filtered > 220`) should be clasified as 6-cell (3.7V/cell), not 5-cell (4.4V/cell)

pl4hh, 2018-01-06:
Not sure if my problem comes from this, but i have a problem that somehow my single cell voltage shows 3,5 instead 4.3 when connecting a High Voltage 4S LiPo. Somehow the FC calculates that it is a 5S battery and hence it shows 3.5. I haven't found any quick solution, currently i let it sit there for a while and constantly reboot from the osd menu, after some tries it shows the right voltage for single cell, probably due to draining the power low enough to a normal full LiPo

jflyper, 2018-01-06:
@pl4hh It's related, but you have a solution. Increase "Maximum Cell Voltage" to 4.4.

pl4hh, 2018-01-06:
@jflyper thanks, was trying to play with settings next after i went to default just to make sure it's not some leftover from the stock firmware. been using this FC for the second day.

mikeller, 2018-01-09:
@jflyper, @martinbudden, @ledvinap: Not sure if there is a good way to do reliable cell count detection that will work with all possible battery / capacitor combination. I think the advantage of the current solution is that it is tuneable in a relatively intuitive way.
As a suggestion, considering that most users will set up and use their quads with only 1 type of battery (cell count wise), how about adding a `cell_count` parameter that allows the user to preset the expected cell count, and only use the cell count detection as a fall back if `cell_count` isn't set?

jflyper, 2018-01-09:
`cell_count` may be good. Btw, wasn't this done that way and moved to automatic detection? I'm not sure.

mikeller, 2018-01-09:
> Btw, wasn't this done that way and moved to automatic detection? I'm not sure

Potentially. Auto detection sounds tempting - but only if it's working reliably.

woutput, 2018-02-11:
I think having a parameter cell_count isn't very usable for all users. I started only a short time ago in this hobby and I bought two batteries: 2S 850 mAh for indoor use and 3S 450 mAh for outdoor use. In practice I'll probably use both batteries in both environments. Having to change this parameter with every battery change would be very user unfriendly.

I think the solution to this is quite simple:
1. First find the typical cell voltage. I disagree with the assumption that batteries are typically charged. The typical cell voltage should be the voltage that the cell provides when it's halfway used. Where halfway used is between fully charged and discharged to a level that will not have large negative impact on lifetime. I propose to assume LiPo type battery since I understand this chemistry is currently used most and LiHV voltages aren't very different.
* 3.7 V is mentioned as nominal voltage for LiPo (http://imaxrc.com/B610-PRO.html or https://cdn-shop.adafruit.com/product-files/2750/LP552035_350MAH_3.7V_20150906.pdf)
* 4.2 V is mentioned as charge voltage (http://www.imaxrc.com/images/pdf/B6AC%20PRO%20Manual.pdf or https://cdn-shop.adafruit.com/product-files/2750/LP552035_350MAH_3.7V_20150906.pdf) so that is maximum voltage
* 3.0 V is mentioned as discharge voltage (http://www.imaxrc.com/images/pdf/B6AC%20PRO%20Manual.pdf or https://cdn-shop.adafruit.com/product-files/2750/LP552035_350MAH_3.7V_20150906.pdf) so that is minimum voltage
2. Implement simple algorithm that uses the typical cell voltage as center point and find the cell count (called minimum Hamming distance in math).

mikeller, 2018-02-11:
@woutput:

> First find the typical cell voltage. I disagree with the assumption that batteries are typically charged. The typical cell voltage should be the voltage that the cell provides when it's halfway used. 

I think you are misunderstanding what is actually done. The configured maximum voltage is used to define the cut off when a battery is considered to have to have more than a given number of cells (i.e. 'if the measured voltage is larger than X * max_cell_voltage, then the number of cells must be larger than X'). This works relatively well, except that there is some overlap for high cell counts, causing partially charged batteries to be detected as one cell less than what they really are - there is nothing that can be done about this, since there is overlap in voltages.

Also, the idea is to have the preconfigured number of cells as an optional feature: The default would be to auto-detect, unless the user knows that they will only ever use one type of battery, and explicitly configure the number of cells for this battery type.

woutput, 2018-05-05:
>I think you are misunderstanding what is actually done.

I think I understand what was is done. My point was: why use this approach (which has issues as you already indicated) and not use the method I propose? (That is under the assumption that in this discussion we are searching for an algorithm to predict cell count from total battery voltage)

Copter38, 2018-05-31:
I am running BF 3.3.2 on a Matek F405-OSD FC. My RX is an R-XSR on FPort protocol using the uninverted hack.

I have also recently started to experience issues with wrong cell number detection.
As this started to appear shortly after I upgraded my setup to FPort RX protocol, I investigated in this direction using blackbox logging with debug set to BATTERY.

This is what I found when plugging in a fully charged, brand new 4S battery measured at 16.74 Volts:
1. When battery is plugged in with TX off: 
[image]
In this case the cell detection works as expected and detects a 4S battery.

2. When battery is plugged in with TX on:
[image]
Here the detection is wrong as 5 cells are detected. This results in a low battery alarm.

As can be seen in above graphs, when the TX is on and therefore when telemetry and control data is exchanged between the RX and the FC, a lot of jitter appears in the voltage ADC readings. 
This jitter can reach as much as 1.5V. This leads to VBat filtered readings being higher than in case 1 when battery cell detection kicks in and this results almost systematically in wrong cell count detection.

In the above examples, I had already reduced VBat scale to 109 (hence my VBat measurements are actually lower than reality).
I eventually managed to get around this issue by also raising VBat max cell voltage to 4.4V but this is really just a quick fix to get back in the air. I don't think that the ADC readings observed in case 2 are as they should be.

As a side issue that may be related to the above, I am also seeing some strange behavior in the unfiltered RC Command traces in blackbox where, every now and then, large stair steps appear that can last as much as 10 times the normal frame duration (9ms).
[image]

Has someone else using the new FPort protocol observed similar behaviors or is it something with my particular setup? Any feedback on this would be much appreciated.

mikeller, 2018-05-31:
@Copter38: I don't think this is related to your change to FPort. Could be related to drift in your charger, and it now charging the batteries to a slightly higher voltage. This is an often discussed problem with overcharged / HV batteries, and can be solved by increasing the value set for `max_cell_voltage`.

Copter38, 2018-06-01:
@mikeller: thank you for your feedback. I agree that the issue of wrong cell detection can be somehow circumvented by increasing max_cell_voltage which I did already. But still, there seems to be something strange happening on voltage ADC readings as soon as FPort is active and transmitting. A drift in the charger would clearly not explain this phenomenon.
Could it be that some interrupts relating to FPort management may impact the voltage ADC reading function? That could then also explain the second strange behavior noted on RC input signals reading.
Any thoughts on this?

mikeller, 2018-06-01:
@Copter38: Can you please confirm that this is indeed linked to FPort by switching back to whatever protocol and firmware you were using before FPort, and checking if that makes it go away?

Copter38, 2018-06-03:
@mikeller: Ok, I have switched over to 'regular' use of the R-XSR and connected both SBUS and Sport wires to the FC.
1. Connecting a regular 4S battery whilst TX is OFF:
[image]
Same result as before, the ADC samples show very little jitter.

2. Connecting same 4S battery with TX ON (SBUS and Telemetry active):
[image]
Jitter is back on ADC sampling with more than 1V between lows and highs.

Whilst I was at it I did a 3rd test, this time with only SBUS active (Telemetry disabled).

3. Connecting same 4S battery with TX ON (SBUS only):
[image]
Jitter on ADC sampling is similar to case 2.

So, I am not sure what conclusion to draw from this. Clearly the jitter observed on ADC sampling is not strictly related to FPort but it has visibly a relation with RX data management...

Lastly, I did re check the RX Command traces now acquired through regular SBUS and I got this result:
[image]
This test clearly shows that the random larger steps in RC inputs are still there but this time they never exceed 2 frames in length when using SBUS (max 18ms) where as I could observe much larger steps when on FPort.
Is this missing of RC frames something normal? As I am new to the Taranis/SBUS/Fport world I have no past experience on behavior of these protocols with older BF firmware revisions.

mikeller, 2018-06-03:
@Copter38: This almost makes me suspect that the jitter on the ADC sampling is caused by good old electrical interference. You could test this hypothesis by powering the RX from a second battery.

Re the larger steps on RC inputs, this looks like frames are being missed. I haven't seen this happening this often with either SBus or FPort, but again, if there is a problem with interference, then the serial connection could be affected as well.

Copter38, 2018-06-04:
@mikeller: I think you effectively identified the origin of my particular issue. 
After some further testing, it effectively appears to be related to electrical noise and this noise seems to be fed back by the R-XSR receiver into the FC. 
This noise is only there when I power my system on battery only. 
When I have the USB plugged in at the same time as the battery, the noise disappears. On the Matek F405, the SBUS receiver gets powered when the FC is connected to the USB. 
So now I need to find a way to suppress this noise (filtering with a capacitor or moving the receiver on another voltage pad) as it may actually also be the source of the missed SBUS frames.
Thanks for helping out to track this one down!

mikeller, 2018-06-05:
@Copter38: Happy to have helped. My bet would be on a noisy on-board voltage regulator. A capacitor on the 5V rail might be all that's needed to fix this.

Copter38, 2018-06-05:
@mikeller: I checked out the 4.5V voltage pad dedicated to the RX with an oscilloscope and , sure enough, as soon as the R-XSR starts to transmit SBUS frames, I get noise on this power supply line. Looking at the waveform of this noise, its shape is very recognizable and exactly that of an SBUS frame with periodicity of 9ms! The amplitude peak to peak is 0.4V (quite impressive). 
So thinking about it again, I am starting to think that this may be caused by cross talk between signal and power cables at the RX side... I am pretty sure that a cap will help sort this out.

vkolotov, 2018-07-02:
I can confirm, I'm having the same issue with a Bardwell F4 board. Voltage spikes when connecting a fully charged battery to around 17v, then goes to normal after 1-2 secs, incorrect number of cells detected.