betaflight/betaflight #4896: BetaflightF3 Motors and Voltage reading problems 3.3.0

gsimpson2g, 2018-01-03:
As of build 469, motors 2 and 3 no longer function on the BFF3 board with DSHOT600. If I flash back to build 468, they work perfectly fine. The voltage reading is also wacky since build 469. It bounces from 6v to 8v, even thought the battery is at 15v. Flashing back to build 468, the voltage appears to be correct. These issues are still in the latest 480 build.

AndersHoglund, 2018-01-03:
Confirmed. 
Seems PR #4852 breaks BFF3. New PR to fix this coming shortly. Stay tuned.

AndersHoglund, 2018-01-04:
Could you please try build 481 or later. Then close this issue, if you are happy with the fix.
https://ci.betaflight.tech/job/Betaflight/changes

gsimpson2g, 2018-01-05:
Works great now, thanks!

closed 2018-01-05 by gsimpson2g