betaflight/betaflight #4901: Could SOFTSERIAL be used to IRC TRAMP?

nyway, 2018-01-05:
Hi,I have a question.
The hardware serial were not enough use.
So, Could SOFTSERIAL be used to IRC TRAMP?

richard-scott, 2018-01-05:
This issue (https://github.com/betaflight/betaflight/issues/3692) seems to suggest that SOFTSERIAL will work.

jflyper, 2018-01-05:
@nyway Yes. Tramp can be used on software serial. For further assistance, please using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack: https://slack.betaflight.tech or other user support forums & groups (e.g. facebook).

closed 2018-01-05 by jflyper

nyway, 2018-01-05:
@jflyper @richard-scott 
Thanks for your help ,has been solved,now i can resource software serial  port to use IRCTRAMP,and communication successful.
Resource Software serial port must have timer