betaflight/betaflight #4903: min_check ranges from 0 - 2000

etracer65, 2018-01-05:
Shouldn't min_check be constrained to 1000 - 2000?

```
# get min_check
min_check = 1050
Allowed range: 0 - 2000
```

Looking at mixer.c it looks like bad things will happen if min_check is set below 1000.

`    rcCommandThrottleRange = PWM_RANGE_MAX - rxConfig()->mincheck;`

and

`        throttle = rcCommand[THROTTLE] - rxConfig()->mincheck;`

lastly, this would prevent motor_stop from working since rcData[THROTTLE] could never get below min_check.

```
        if (feature(FEATURE_MOTOR_STOP) && ARMING_FLAG(ARMED) && !feature(FEATURE_3D) && !isAirmodeActive()) {
            if (((rcData[THROTTLE]) < rxConfig()->mincheck)) {
                motorOutput = disarmMotorOutput;
            }
        }
```

bkeinert, 2018-01-06:
I run min_check = 1000 (I would be right at your proposed min setting with no headroom) and max_check = 2000, because it nicely matches the FrSky SBUS defaults [988, 2012]. I don't think this is really an issue.

AndersHoglund, 2018-01-06:
This is a non-issue. 
There are however a lot of Qs and support issues with THROTTLE arming prevention, caused by the BF default `min_check` value 1050 and RC radio systems with a default minimum throttle value above that, say 1100.  Not sure how to educate new users to adjust Tx range to 1000 ... 2000.  Maybe we should revert `min_check` to its old default value of 1150?

etracer65, 2018-01-06:
Ok, I'll close the issue.  I guess it won't cause a safety issue since the quad can't arm unless throttle can get below min_check.  I just don't see any reason why allowing it to be set to 0 is valid.  At least it should be constrained to the same minimum as rx_min_usec (750).

PR #4922 (2018-01-10): Adjusted CLI parameter value range for PWM input parameters to be the valid PWM input range.
Fixes #4903.
closed 2018-01-06 by etracer65