betaflight/betaflight #4904: Possible division by zero from setpoint transition.

JesseDFBlue, 2018-01-06:
In pid.c "relaxFactor = 1.0f / (pidProfile->setpointRelaxRatio / 100.0f);" a division by zero could happen (0 in pidProfile->setpointRelaxRatio).  

A value of 0 isn't catched in the code, the range of pidProfile->setpointRelaxRatio should be 1 - 100 instead of 0-100.

mikeller, 2018-01-09:
@JesseDFBlue: Can you please open a pull request to change the range for `pidProfile->setpointRelaxRatio` to [1, 100]?

McGiverGim, 2018-01-10:
Don't forget to change it in the CMS system too:

https://github.com/betaflight/betaflight/blob/940d85e20be621586f14092f8c6f952d390204b9/src/main/cms/cms_menu_imu.c#L284

richard-scott, 2018-01-13:
I was just going to report that when having D Setpoint Transition at 1, and D Setpoint Weight at zero (which are the defaults in 3.3) the quad is sluggish. If you set both to zero, the quad follows stick commands exactly. Maybe this is the reason.

PR #4961 (2018-01-14): Fixed range for `setpoint_relax_ratio` to avoid division by zero.
Fixes #4904.
closed 2018-01-14 by mikeller