betaflight/betaflight #4917: Matek VTX continuous power cycle with nightly build

voodoo614, 2018-01-09:
Matek VTX running version 1.5 firmware will continue to cycle every 10 seconds and cuts out video for a second.  This only happens when changing the power level to 600 of the VTX through BFCMS.  The Matek VTX max mW is 500.  Using BFCMS to change power level 25, 100, 200 or 400 is fine.  It seems like BFCMS is continuously sending a signal to the VTX causing the power to cycle.  Betaflight firmware nightly build #479.  This does not seems to happen on 3.2.3.  I have not tried it with other nightly build.

I tried two different FC CL Racing F4S and Matek F405 AIO, both have problem with nightly build.  Whether is was SBUS/S.Port or F.Port, or different UART, it does not matter, problem occurs.

```# diff

# version
# Betaflight / MATEKF405 (MKF4) 3.3.0 Jan  1 2018 / 09:46:56 (5a448ea9b) MSP API: 1.37

# name

# resources

# mixer

# servo

# servo mix

# feature

# beeper

# map

# serial
serial 1 0 115200 57600 0 115200
serial 3 64 115200 57600 0 115200
serial 4 8192 115200 57600 0 115200

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# vtx

# rxfail

# master
set serialrx_provider = FPORT
set serialrx_halfduplex = ON
set vtx_power = 4

# profile
profile 0

# rateprofile
rateprofile 0
```

Shadow-MasterTf, 2018-01-09:
Yep happens to me too. If you select 400mw from the LUA it actually sets the VTX to 500mw and works fine but setting 600mw via the LUA it power cycles over and over making you lose video

codecae, 2018-01-09:
Did you recently update the firmware on the VTX?

voodoo614, 2018-01-09:
Version 1.5 is the latest firmware. I have been in contact with Matek, they were able to reproduce.  They just told me this morning that they will fix this with the next firmware update. Prior to my knowledge this morning,  it looks like betaflight implementation of BFCMS to change VTX somehow changed from 3.2.3 to 3.3. I wasn't quite sure if this was a Matek or Betaflight firmware issue.

MATEKSYS, 2018-01-10:
with BF3.2.x,  25--25mw, 100/200 --200mw,  400/600--500mw
with BF3.3, Pls choose 25,200,400 in BFOSD to match 25,200,500mw of VTX-HV /FCHUB-VTX.

codecae, 2018-01-10:
The power options in CMS for the tramp protocol were implemented about a year ago (Jan 16, 2017)

https://github.com/betaflight/betaflight/blame/master/src/main/io/vtx_tramp.c

This appears to be in issue with the power index selection and not with what is displayed on the screen.

codecae, 2018-01-10:
The power output options in CMS have not changed between 3.2 an 3.3.  There were other changes that may affect index selection that may have had an effect.

Taking a look...

codecae, 2018-01-10:
According to the code, it appears that the power output value itself (25, 100, 200, 400, 600) is sent via the tramp protocol, rather than an index.  in 3.3, there is a system that enforces settings persistence to match the configuration in betaflight.  If the hardware does not send a response to correspond with the request, it will continue to attempt to set the value in order to keep the hardware in sync.

@MATEKSYS is it possible that the vtx is not sending a response to the 600mw request?

voodoo614, 2018-01-10:
@codecae. This sounds like the exact problem.

MATEKSYS, 2018-01-11:
@codecae BF send 400 or 600 to VTX, VTX return 400 to BF. VTX perform 500mw output.
BF send 100 or 200 to VTX, VTX return 200 to BF. VTX perform 200mw output

NFoxEZ, 2018-01-23:
The problem remained unsolved?

mikeller, 2018-01-23:
@codecae: Are you looking into this?

twidgit, 2018-02-19:
So the current workaround is to use the 400mw setting?

voodoo614, 2018-02-19:
@twidgit. Yes, select 400mw for 500mw.

zigzaggin, 2018-03-21:
I am experiencing one reboot after exactly 10 seconds following power up when set at 200mw.

Running BF 3.3 on two rigs, one has a JB F4 and the other has a Fortini F4 OSD, both exhibit the same behavior.

The reboots occur at random throughout flight, no particular regularity to it. On the bench it only reboots the single time. 

Flashing back to 3.2.5 removes the issue entirely. 

I disabled VTX low power disarm and it had no effect.

MATEKSYS, 2018-03-21:
@zigzaggin select 200 for 200mw, 400 for 500mW in BFOSD. don't select 100 & 600.

zigzaggin, 2018-03-21:
@MATEKSYS yeah, 200mw is selected. It reboots at 25mw as well. Something changed in BF 3.3 to cause it, don't think there is a particular issue directly on the VTX as flashing back to 3.2.5 removed the issue on both setups. Its pretty odd.

voodoo614, 2018-03-21:
What firmware do you have on the VTX? I have BF 3.3 and Matek latest firmware, and there is no issue with power cycle.

zigzaggin, 2018-03-21:
@voodoo614 most likely the factory default, as I have not flashed them. They were purchased this year, so unless the stock was particularly old, it may be up to date. If they are older though, version 1.4 may correct the issue. I will explore updating the firmware this evening.

BillyGriffiths, 2018-04-12:
Ran into the same problem with power cycle when an "unsupported" value is set, BF3.3.

This is however noted on Matek's website. 

[image]

When setting the power value via BF LUA, is the option index or the option value used? I modified the LUA to only include 25, 200  and 500 as options, but only 25 then sets properly. For 200 and 500 the LED changes, but keeps blinking and cycling.

atomiclama, 2018-05-10:
Has anybody tried the V1.9 from Matek?
http://www.mateksys.com/?portfolio=vtx-hv#tab-id-6
Does this fix the problem? Can't get at my vtx to verify.

krzaczek, 2018-05-10:
@atomiclama I did and it work perfectly with F405 AIO

closed 2018-05-10 by voodoo614