betaflight/betaflight #4919: How to set pin default on boot to Pull-Up

nikola-v, 2018-01-09:
Hello,

I am using PPM output on Turnigy 9X IA-8 receiver and problem is that receiver has bind port on the same pin as PPM output.

How can I set pin in Betaflight firmware to pull-up at initial boot inside firmware?

FC is Omnibus F4.

jflyper, 2018-01-09:
@nikola-v 
Do you have any document describing bind signal specification?

nikola-v, 2018-01-09:
Bind is pretty simple , on Bind/PPM pin put short between Signal and Ground and receiver will start flashing. 

I have identified part of the code which I have changed but no effect, still researching.

git\betaflight\src\main\drivers\rx\rx_pwm.c

```
void ppmRxInit(const ppmConfig_t *ppmConfig)
{
    ppmResetDevice();

    pwmInputPort_t *port = &pwmInputPorts[FIRST_PWM_PORT];

    const timerHardware_t *timer = timerGetByTag(ppmConfig->ioTag, TIM_USE_ANY);
    if (!timer) {
        /* TODO: fail here? */
        return;
    }

    ppmAvoidPWMTimerClash(timer->tim);

    port->mode = INPUT_MODE_PPM;
    port->timerHardware = timer;

    IO_t io = IOGetByTag(ppmConfig->ioTag);
    IOInit(io, OWNER_PPMINPUT, 0);
#ifdef STM32F1
    IOConfigGPIO(io, IOCFG_IPD);
#else
    IOConfigGPIOAF(io, IOCFG_AF_PP, timer->alternateFunction);
#endif

    timerConfigure(timer, (uint16_t)PPM_TIMER_PERIOD, PWM_TIMER_1MHZ);
    timerChCCHandlerInit(&port->edgeCb, ppmEdgeCallback);
    timerChOvrHandlerInit(&port->overflowCb, ppmOverflowCallback);
    timerChConfigCallbacks(timer, &port->edgeCb, &port->overflowCb);

#if defined(USE_HAL_DRIVER)
    pwmICConfig(timer->tim, timer->channel, TIM_ICPOLARITY_RISING);
#else
    pwmICConfig(timer->tim, timer->channel, TIM_ICPolarity_Rising);
#endif
}

```

I have changed 

```
#else
    IOConfigGPIOAF(io, IOCFG_AF_PP, timer->alternateFunction);
#endif

```
to 

```
#else
    IOConfigGPIOAF(io, IOCFG_AF_PP_UP, timer->alternateFunction);
#endif
```

jflyper, 2018-01-09:
Do you have a problem of the receiver always going into bind mode upon power-up?

nikola-v, 2018-01-09:
Yes exactly that, receiver is going into bind mode upon power-up.
Do you know any solution for this problem?

jflyper, 2018-01-10:
@nikola-v 
Hm... It may have something to do with timer gpio initialization.

Oh, and a word of caution.
Setting the pin to pull-up may solve your issue, but it may brake other cases. So the change must be limited by an CLI option or similar facility unless you can carry out an extensive regression test against all existing working PPM receivers. It is also better to have a firm basis for the change, which is the bind process specification for your receiver to get your mod into the repository quickly.

nikola-v, 2018-01-10:
@jflyper 
What would be bind process specification? Could you please explain meaning of this? Where would one get such specification?

_First thinking:_
I was thinking on testing how bind implementation works on the IA-8 receiver following way:
Assumption is that signal pin stays in input mode for some time to detect if pin is connected to ground.When this time expires mcu inside receiver does reconfigure of the pin to output pin and emits PPM on it.

So to find out what is the timing on this, one could connect FET on it and activate it with delay.
Through the process of delay increment it would give out information on how much time does it wait to detect short between signal and ground (BIND process).

_Second thinking:_
If pins on the controller are floating upon power-up, just small valued resistor for pull-up should solve the thing? Should it?

jflyper, 2018-01-11:
Good! I wasn't thinking of the hardware solution at all. That's why I was asking you for the specification to find out the valid time window that the line must be pulled up and kept high.

nikola-v, 2018-01-11:
I will try to test it and report back with info on that. Till then it seems that this solutions works. 
It is interesting that receiver design isn't guarding against that with pull-up or delayed pin reading just to be sure that outside circuit has stable state.