betaflight/betaflight #4921: Fc3 Motors high on low throttle

jerseyjay, 2018-01-10:
I hope this is the right spot for this. I just got a beta75,ran it through Betaflight and the motors would stay on as soon as I plugged in the battery.Messed around with some setttings and got that to stop but not when I throttle its all or nothing. 

# dump
# Betaflight / BETAFLIGHTF3 (BFF3) 3.2.0 Sep 28 2017 / 16:16:50 (ceb6f70) MSP API: 1.36
name Mamma
resource BEEPER 1 C15
resource MOTOR 1 A06
resource MOTOR 2 A07
resource MOTOR 3 B08
resource MOTOR 4 B09
resource MOTOR 5 B00
resource MOTOR 6 B01
resource MOTOR 7 A03
resource MOTOR 8 A02
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 B07
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 B01
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 B00
resource SERIAL_RX 12 NONE
resource LED 1 NONE
resource LED 2 NONE
resource LED 3 NONE
resource RX_BIND 1 NONE
resource RX_BIND_PLUG 1 NONE
resource SPI_SCK 1 B03
resource SPI_SCK 2 B13
resource SPI_SCK 3 NONE
resource SPI_MISO 1 B04
resource SPI_MISO 2 B14
resource SPI_MISO 3 NONE
resource SPI_MOSI 1 B05
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 NONE
[... 470 lines cut ...]
set p_alt = 50
set i_alt = 0
set d_alt = 0
set p_level = 50
set i_level = 50
set d_level = 75
set p_vel = 55
set i_vel = 55
set d_vel = 75
set level_limit = 55
set horizon_tilt_effect = 75
set horizon_tilt_expert_mode = OFF
rateprofile 0

set rc_rate = 100
set rc_rate_yaw = 100
set rc_expo = 0
set rc_expo_yaw = 0
set thr_mid = 10
set thr_expo = 100
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 10
set tpa_breakpoint = 1640

mikeller, 2018-01-10:
@jerseyjay: Looks like the board in this craft is not properly supported in Betaflight. For a board to be supported in Betaflight, the board manufacturer has to either develop and submit a target definition for their board, or else at least seek a dialogue with the Betaflight community, and provide schematics and sample boards, so that somebody from the community can contribute a target definition for this board. Please contact the manufacturer and / or your supplier and ask them to do any of the above so that support for your board can be added to Betaflight.

closed 2018-01-10 by mikeller

richard-scott, 2018-01-10:
Their website says it doesn't run a BFF3 based board but a SPRACINGF3EVO based FC.

https://betafpv.com/products/beta75-bnf-tiny-whoop-quadcopter

mikeller, 2018-01-10:
Well spotted, @richard-scott.

richard-scott, 2018-01-10:
@mikeller - I had time to kill on the bus this morning on the way into work :-)