betaflight/betaflight #4928: Standalone Configurator does not work under Vista

BadRaven, 2018-01-10:
Just had an issue getting comms to a SPRacingF3, cured by changing to Standalone. However, along the way found that my Vista OS laptop would not run the Standalone version reporting a missing .dll.

Win7 laptop runs it perfectly.

Can someone confirm if Standalone is OK on Vista and its my PC or if its general?

McGiverGim, 2018-01-10:
Could you paste the error?

This issue belongs to the configurator so maybe it's better to put it there.

BadRaven, 2018-01-10:
The error message is:

The procedure entry point QueryUnbiasedInterruptTime could not be located in the dynamic link library KERNEL32.dll

McGiverGim, 2018-01-10:
It seems that windows XP and Vista support was drop from chromium long time ago. The only solution is build a custom version of the standalone using NW LTS version https://dl.nwjs.io/v0.14.7

But I don't know if this will work....

mikeller, 2018-01-10:
Duplicate of betaflight/betaflight-configurator#813.

@McGiverGim: I had a look into this a while ago, and it looks like the latest NW.js version to support XP or Vista is from before support for serial ports was added, so it's definitely a no-go for the configurator.

closed 2018-01-10 by mikeller