betaflight/betaflight #4931: Feature request:  Use the Garmin "LIDAR-Lite v3" for vertical hold instead of baro/sonar?

joeostrander, 2018-01-10:
Sorry if this isn't the correct place to put this... but I'm curious if it would be possible to add this as a feature.

These things are only about $130 and have WAY better range (40 meters) than an ultrasonic sensor.  I think it would be very interesting to try to use one for vertical hold.  I don't really use baro/sonar myself, but I think it might be useful to others.

LIDAR-Lite v3
https://buy.garmin.com/en-US/US/p/557294/pn/010-01722-00#

Thanks!

jflyper, 2018-01-11:
@joeostrander I have the device itself, so I'll move the project up in the stack.

dave-patterson-bristol, 2018-02-09:
@joeostrander I've got a Benewake TFMini lidar that's a really good alternative to a sonar if you have a spare UART. This thing is super light and takes up no more room than a sonar. The code to support it will be included in the BF 3.3 release.  

But beware: altitude hold is still experimental in BF. My experience with sonar suggests that you may get solid sensor readings but that doesn't translate into a stable hover. It's not getting much attention from developers. I'm new to the code but I'm keen to get it working.

INav is another cleanflight fork that's more focused on autonomous flight. It focuses on GPS and barometer as position sensors and I'm told that altitude hold support isn't working for rangefinders either.

joeostrander, 2018-02-09:
@dave-patterson-bristol ,
Thanks, I actually have one of those too.  I agree it's a great little device.  
I'm using both of these for unrelated projects but thought maybe someone could get some use out of the Garmin one since it has a much higher range (40m vs 12m).  

Even though I don't really need it, I did install the TFMini on a basher quad just for fun.  I compiled up the 3.3 code and got it working in the betaflight sensor tab.  I couldn't get it to do altitude hold for some reason however.  I didn't really need it though, so I gave up quickly after.  

This being said... I'm not opposed to closing this issue if there is no interest for betaflight.  I think you're right in that it's probably more of an INav thing.

dave-patterson-bristol, 2018-02-09:
@joeostrander I think there's enough interest and existing support to keep alt hold in betaflight as well as INav. My hope is that we can get the betaflight alt-hold to work and use that to help fix the more complicated INav alt-hold code.

I'm not surprised to hear you didn't get a hover. I had the same experience with sonar. I've yet to try any real testing of vertical PID settings but I gather the defaults are not good. @DietmarM has been looking at the problem and experimenting with a custom alt-hold pid controller.

dave-patterson-bristol, 2018-02-09:
@joeostrander Did you have to preconfigure the benewake before connecting it to your quad? I just tried plugging into a fresh BF build and the sonar sensor didn't light up.

dave-patterson-bristol, 2018-02-11:
Got it working. I needed to update the target.h for my FC (SPRACINGF3) to include #define USE_RANGEFINDER_TF and rebuild the hex. Then the following CLI commands enabled the TFMini on UART 2: 
set rangefinder_hardware = TFMINI
serial 1 32768 115200 57600 0 115200

joeostrander, 2018-02-12:
@dave-patterson-bristol Sorry I missed your message.  That's exactly what I did (even the same hardware).  
I think I also tried enabling the sonar feature to get the alt-hold mode option on an aux switch.  
It didn't seem like it was even attempting to hold for me.  Sensors tab showed the height data fine, but I'm sure I was doing something else wrong.  Maybe I should have tried configuring aux mode via CLI instead of checking sonar.

joeostrander, 2018-02-12:
I just looked at the diff export I made when I was testing, I had these:
feature RANGEFINDER
set rangefinder_hardware = TFMINI
serial 1 32768 115200 57600 0 115200
aux 3 22 1 1300 2100 0

Peeking in the code, 22 appears to be RANGEFINDER, so I think I had it set OK... not sure if I missed something or what.

dave-patterson-bristol, 2018-02-12:
Turns out you also need to define USE_ALT_HOLD on the F3 targets. I'm going to try a build tonight.

joeostrander, 2018-02-12:
I'm curious, does FLASH_SIZE need to be set in the F3 target?  

Just poking around and noticed this:
common_fc_pre.h
#if (FLASH_SIZE > 256)
#define USE_ALT_HOLD

PDF says the F3 has 256kB flash

dave-patterson-bristol, 2018-02-12:
I rolled the dice and defined USE_ALT_HOLD in the target.h for the SPRACINGF3.
I can now enable alt hold.
It's doing something but I'm not sure what.
It responds to throttle movements but it doesn't appear to settle at a fixed height if I leave the stick.
Occasionally it gives a little jump.
Not sure if there's a dead band setting I can make wider.
INav has a setting for throttle hover. Not sure if that's important in BF. (My packs dip pretty quick so it wouldn't stay steady for long.)

joeostrander, 2018-02-12:
Great, sounds like a good start.

I thought the throttle was basically only going to adjust speed once alt-hold switch was enabled, no?  Like if you want to change height, you have to disable the switch, move up/down and then enable it again.

P.S. did you notice that part of common_fc_pre.h:
`
#if (FLASH_SIZE > 256)
#define USE_ALT_HOLD
`

The SPF3 only has 256kB, do you think this is related (since it's not _greater_ than 256)?  I guess I'm not even sure if FLASH_SIZE is defined in the target.

mikeller, 2018-02-12:
@joeostrander: `FLASH_SIZE` is defined dependent on the MCU type used, and is 256 for all F3s.

dave-patterson-bristol, 2018-02-13:
@joeostrander I think the idea is that you can climb or drop but there's a deadband in the middle of the throttle where alt-hold applies. See "set alt_hold_deadband" in the CLI. 40 may be too narrow. 

In this mode "adjusting speed" should be a function of pitch/roll attitude. If you want to speed up you lean into it and the alt-hold controller should drive the throttle to maintain height as the craft accelerates. 

FYI we've set up an altitude_hold topic on the betaflight developers slack group: https://betaflightgroup.slack.com/messages/C96RA289H

@podracer has done some interesting work on alt-hold using a barometer on his big . He's been helping me understand debug techniques.

joeostrander, 2018-08-07:
I believe rangefinder support has been removed in BF 3.4

closed 2018-08-07 by joeostrander

markandkymward, 2019-03-16:
How did this feature request get so far off track?

Firehammer047, 2021-01-20:
I’ve been wanting to do this for years now. I bought an original Lidar lite 3 years ago and was trying to adapt the sonar code in Cleanflight for my Lidar.

Does anyone know the current status of altitude hold using LIDAR?

haslinghuis, 2021-01-20:
I have the Garmin v3hp. It should not be to hard to implement and maybe we could also use it for landing in case of fail-safe.

Firehammer047, 2021-01-21:
That was my thought also, but my original idea was to be able to fly at constant altitude (close to the ground without crashing).

I have some time to work on this, I just need pointed in the right direction. I’ve been reviewing the rangefinder code, and I can start tomorrow.

mikeller, 2021-01-21:
@Firehammer047: The answer to your question re altitude hold is: It does not exist in Betaflight. The original code that had been forked from cleanflight (and probably copied from baseflight into cleanflight before this) had never worked properly, resulting in violent up / down shaking whenever used. For this reason it was removed.

Not sure if using LIDAR as the only data source for altitude hold would be a viable option - this will obviously fail in any situation where the LIDAR goes out of range or over a surface that does not give it a useful readout (long grass). In these cases, unless the pilot is standing by to immediately take over altitude control manually, altitude hold will likely result in a crash / runaway into the sky.

As for using proper sensor fusion of GPS / barometer / LIDAR for altitude hold, this is certainly feasible, but outside of the project goals of Betaflight. If you want this or other 'autonomous flight' related features, have a look into iNav.

dkustec, 2021-01-21:
What @mikeller said. And this is why is in the iNav LIDAR+baro (at least) mandatory setup for ALT-handling modes.

Firehammer047, 2021-01-21:
@mikeller I was confused because in the rangefinder code, there is stuff still there--so it's not completely removed. I can see how LIDAR/SONAR data alone would make altitude hold difficult, but it's something I've been wanting to experiment with for a few years now and I finally have time.

Yes, there are many limitations, as you have mentioned, yet I feel (maybe naively) that this data will be more useful than most think.

I'm looking at INAV now, but I have a hunch it would be easier to incorporate LIDAR into betaflight, rather than rewrite INAV to work without GPS or baro.

Anyways, any help, suggestions or direction is appreciated.

mikeller, 2021-01-25:
@Firehammer047: Can you please point out where in the rangefinder code there is code for altitude hold?